asrob-uc3m/impresoras-asrob #1: Actualizar versión software de control

kleiren, 2016-06-06:
Las versiones de Repetier y de los generadores de G-Code en los PC de control de las impresoras son bastante antiguas, ¿hay alguna razón para eso? ¿Podrían ser actualizados?.
La cuestión de seguir teniendo Windows XP en esos PC también es discutible.

jgvictores, 2016-06-07:
Hi @kleiren,
En la actualidad estamos mejorando la documentación de lo que existe y funciona: [Manual de usuario de los i3](https://www.gitbook.com/book/asrobuc3m/impresoras-user-manual-i3/details), [Manual de mantenimiento de los i3](https://www.gitbook.com/book/asrobuc3m/impresoras-maintenance-manual-i3/details).
Para llevar a cabo una tarea de actualización, hará falta tener respuesta a las siguientes preguntas:
- [ ] ¿Qué software debemos instalar (OS, generador G-Code...)?
- [ ] ¿Quién lo instala y se encarga de que funcione?
- [ ] ¿Quién lo documenta?

David-Estevez, 2016-06-07:
Dado que los PCs no son exactamente de última generación, yo optaría por GNU/Linux con un gestor de ventanas ligero como [LXDE](http://lxde.org/).

Para el Slicer, lo más user-friendly (y rápido) que conozco es [Cura](https://ultimaker.com/en/products/cura-software), para el que se pueden crear perfiles de configuración de nuestras impresoras y compartirlos, en caso de que alguien quiera generar el gcode en su PC y usar los PCs de la nave sólo como host.

Rodriwp, 2016-06-08:
Me parece perfecto actualizar las impresoras aunque antes haria una copia de seguridad de las configuraciones que subiria al repositorio por si acaso.

Siotma, 2017-02-08:
Propongo actualizar las versiones del Marlin de ambas impresoras. Me ofrezco a configurarlo.

También propongo utilizar Repetier + Slic3r ya que es lo que uso a nivel particular y tengo gran experiencia. También me ofrezco a configurar los perfiles de impresión fileteo de Slic3r.

Documentar su instalación es fácil. Pero a hacer manuales de uso no estoy dispuesto, ya que hay mucha información en las páginas oficiales. Creo que alguien podría hacer estos manuales a modo de ejercicio de iniciación en el mundo de las impresoras.

Lanzo la propuesta de una reunión de gente de la asociación que esté interesada en impresoras en  #12. Pretendo que cuantos más mejor sepamos hace estas cosas.

Siotma, 2017-03-02:
Acabo de commitear el firmware de Hija Resurrection ([Marlin 1.0.2-2](https://github.com/MarlinFirmware/Marlin/releases/tag/1.0.2-2)) dentro de nuestro repositorio para que esté siempre disponible y se pueda trabajar sobre él en la medida de lo necesario hasta que quede perfectamente afinado. La idea más importante es no perder todos los parámetros de configuración únicos que están almacenados en el fichero Configuration.h, pero creo que es interesante guardar una copia de todo por las interrelaciones que pueda tener con otros ficheros y que puedan cambiar en futuras verisones.

Además, he de destacar que tal y como se ha reflejado en #2 , el software instalado en el ordenador de Blacky ha sido Repetier ([RepetierHost 1.6.2](https://www.repetier.com/download-now/)). La idea es usarlo junto a Slic3r pero también tiene integración con Cura. Ahora mismo es perfectamente funcional desde la cuenta de invitado de Lubuntu.

Próximamente trabajaremos en la configuración de los perfiles de fileteo para obtener mejores resultados de impresión.

David-Estevez, 2017-03-02:
Hombre, igual lo más interesante hubiese sido hacer un fork del repositorio. Para tener una versión concreta es suficiente con hacer checkout en el commit que nos interese.

En teoría con tener los Configuration.h de las impresoras y apuntar el commit de esa release deberíamos poder restaurarlo todo a como estaba en caso de fallo catastrófico.

Siotma, 2017-03-05:
Lo he estado valorando y tienes toda la razón.

En breve haré los cambios necesarios para dejarlo como tú dices.

Siotma, 2017-03-06:
Hecho.

Siotma, 2017-10-17:
En el día de hoy, se ha probado la funcionalidad del nuevo ordenador de Hija Resurrection desde la sesión de invitado ( #3 ). Así pues, considero que esta issue está totalmente cubierta:
- Ambos PCs cuentan con Lubuntu 14.4, y está documentada su [instalación](https://asrob-uc3m.gitbooks.io/impresoras-maintenance-manual-i3/content/anexo_1_sistema_operativo_y_sofware.html), [backup](https://asrob-uc3m.gitbooks.io/impresoras-maintenance-manual-i3/content/anexo_4_proceso_de_backup_del_s.o_configuraciones_y_software.html) y de [reinstalación](https://asrob-uc3m.gitbooks.io/impresoras-maintenance-manual-i3/content/anexo_3_reinstalacion_de_sistema_operativo_software_y_configuraciones_desde_backup.html)).
- Ambos PCs cuentan con Repetier 1.6.2, y está documentada su [insatalación](https://asrob-uc3m.gitbooks.io/impresoras-maintenance-manual-i3/content/anexo_1_sistema_operativo_y_sofware.html) y su [utilización](https://asrob-uc3m.gitbooks.io/impresoras-user-manual-i3/content/primeros_pasos.html). En este punto, podría ser interesante actualizar las capturas de pantalla para la nueva versión, pero es un detalle menor porque aunque el aspecto visual es distinto, los controles siguen siendo idénticos. He puesto una issue específica [aquí](https://github.com/asrob-uc3m/impresoras-user-manual-i3/issues/1) para que esto no se olvide.
- Ambos PCs pueden imprimir desde la sesión de invitado, y está documentada su [configuración](https://asrob-uc3m.gitbooks.io/impresoras-maintenance-manual-i3/content/anexo_1_sistema_operativo_y_sofware.html).
- Ambas impresoras cuentan con una versión de Marlin actualizada en marzo de este año, y los ficheros de configuración están en el repositorio.

Dada la antigüedad de esta issue, en caso de que en un futuro se quieran realizar nuevas actualizaciones, se debería abrir una nueva.

closed 2017-10-17 by Siotma