asrob-uc3m/impresoras-asrob #2: Configuración PC nuevo de Blacky

jgvictores, 2016-06-30:
Se están configurando los PCs nuevos de Blacky e Hija Resurrection (véase [1](https://github.com/asrob-uc3m/impresoras-maintenance-manual-i3/blob/master/anexo_1_archivos_de_configuracion.md) y [2](https://github.com/asrob-uc3m/impresoras-maintenance-manual-i3/blob/master/anexo_2_sistema_operativo_y_sofware.md)).
Tareas pendientes:
- [ ] Mejorar resolución de pantalla
- [ ] Instalar Slic3r para Repetier

Siotma, 2017-02-19:
Este PC tiene Lubuntu, tal y como se explica en el [manual de mantenimiento](https://github.com/asrob-uc3m/impresoras-maintenance-manual-i3/blob/master/anexo_2_sistema_operativo_y_sofware.md). En el mismo manual también se especifica que tiene instalado el siguiente software:

- Cura. Efectivamente, aparece entre los programas instalados, pero no fui capaz ni siquiera de conectar con la impresora.
- Repetier. O yo soy un negado, o no fui capaz de encontrar el Repetier. Intenté instalarlo el jueves pasado con ayuda de @David-Estevez pero el resultado no fue positivo. 

En definitiva, solicito ayuda de alguien experimentado con este sistema operativo que nos pueda ayudar a dejar los programas funcionando sobre este sistema operativo que, por cierto, creo que **es un gran acierto** ya que la sesión de invitado se autodestruye al salir, lo que va en la línea de su utilidad.

David-Estevez, 2017-02-19:
Me dio la sensación de que Repetier se instaló, pero no generó iconos de acceso directo. Porque cuando lo probamos arrancaba.

Siotma, 2017-02-19:
Perdón, no he sido todo lo explicativo que debería:
Efectivamente, corrimos el Script de instalación de Repetier, y llegó a iniciarse. Incluso conectó con la impresora una vez. Sin embargo, poco después de irte, de repente crasheó y se cerró solo. A partir de ese momento, Repetier no duró abierto más de 5 segundos antes de cerrarse súbitamente... 
De ello desprendo que la instalación no fue correcta...

David-Estevez, 2017-02-19:
Vale, en ese caso sí que habría que echarle un ojo...

Siotma, 2017-02-19:
Me imagino que la manera más rápida de echarlo a funcionar sería conectar el ordenador a internet y que descargue todos los paquetes que vaya necesitando, el script de la instalación decía que había algo desactualizado en el sistema. No debe ser muy difícil, pero de verdad que yo soy un negado y no quiero hacerlo mal... 
Poner en marcha ese ordenador implicaría tener una impresora totalmente operativa, ya que Hija Resurrection podría donar su espejo hasta que reparemos su calentador.

David-Estevez, 2017-02-19:
Para conectarlo se me ocurren dos formas:
* Un cable de red muuuuy largo.
* Mover la torre temporarmente cerca del switch de red más cercano en la nave.

PD: Mono son las librerías que permiten ejecutar programas .NET (cosas de Windows).

Siotma, 2017-02-19:
Yo había pensado en algún pincho WiFi y conectarlo a la red abierta de la universidad.

Vale, pues si necesita esas librerías para funcionar, y las que tiene no le valen, normal que crashee.

Igual que antes, dejo por aquí algo de info que he encontrado: http://www.mono-project.com/docs/getting-started/install/linux/

David-Estevez, 2017-02-19:
Tiene pinta de ser eso, le hacen falta las Mono y se las intenta descargar, pero como no tiene red no puede, y por eso crashea.

Lo del pincho es una buena idea, pero si no lo pilla porque le faltan los drivers no hay interné para bajárselos :sweat_smile: Como es sencilo de probar, sería cosa de intentarlo como primera opción y si no funciona probar con las que he puesto antes.

Siotma, 2017-02-23:
Lamento no haberlo comunicado antes, el trabajo realizado con el PC de Blacky realizado el día 21 de febrero se puede encontrar en #11 , concretamente [aquí](https://github.com/asrob-uc3m/impresoras-asrob/issues/11#issuecomment-281401624).

Siotma, 2017-03-02:
Hoy se ha completado la instalación de Repetier. Ahora mismo se puede lanzar tanto desde la sesión de ASROB como desde la de invitado. Sin embargo, aún no es funcional en la cuenta de invitado. 
Tal y como se ha mencionado en #11, el sistema operativo no permite a la sesión de invitado acceder a los puertos serie. Este es el último paso necesario para que la impresora se pueda utilizar por externos.
@jgvictores habla de "crear un fichero rules de udev (por no meter un chmod cutre)"

David-Estevez, 2017-03-02:
¿Y no se podrá resolver añadiendo al usuario invitado al grupo `dialout`? (como hay que hacer con el IDE de Arduino para poder programar las placas sin sudo)

jgvictores, 2017-03-02:
Metí un `/etc/udev/rules.d/80.guest-printer.rules`:
```
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", GROUP="dialout", MODE="0666"
```

Y ya conecta desde `guest`!

Siotma, 2017-03-02:
En relación al comentario de @David-Estevez , resaltar el detalle de que la cuenta de invitado se crea una distinta cada vez que se inicia sesión, con distinto nombre de usuario. Por ello no hay ningún usuario al que añadir al grupo, sino que debe ser hecho en cada inicio de sesión. Así pues la solución de @jgvictores es maravillosa.

David-Estevez, 2017-03-02:
no tenía ni idea, gracias por la aclaración!

Siotma, 2017-03-03:
Problemas encontrados con el PC durante la mañana de hoy:

1. La sesión de invitado no puede ver los perfiles de Slic3r creados por la sesión de ASROB, por tanto no se puede imprimir con ellos como invitado. Especifico el problema:
	a. La sesión de invitado sí puede lanzar Repetier.
	b. La sesión de invitado sí puede conectar Repetier con la impresora.
	c. La sesión de invitado sí pude lanzar Slic3r desde Repetier.
	d. La sesión de invitado NO puede utilizar los perfiles creados previamente por el usuario ASROB para Slic3r.

Idealmente, la sesión de invitado debería poder VER y UTILIZAR los perfiles de Slic3r creados y afinados para Blacky, es decir, permisos de lectura, pero no MODIFICARLOS, es decir, de escritura.

2. El PC de Blacky se puede utilizar sin problemas para imprimir usando la sesión de ASROB pero durante la impresión, la sesión se bloquea automáticamente pasado un tiempo. Esto no afecta a la impresión, que continúa aunque la sesión esté bloqueada.
3. Al desbloquear la sesión, el PC de Blacky se ha quedado pillado con la pantalla en blanco permanente, aunque la impresión no se ha detenido.

Así pues, propongo lo siguiente:

- [ ] Eliminar el bloqueo de sesión por tiempo de la configuración de las cuentas de usuario.
- [ ] Permitir de alguna manera que la sesión de invitado pueda acceder a los ajustes de Slic3r. Tal vez con un fichero .ini de por medio, o recurriendo a las rules mágicas de @jgvictores otra vez...

Siotma, 2017-03-03:
Documentación gráfica para el post anterior:

Pieza impresa con la sesión de ASROB (hubo un problema de adhesión a la cama, pero ninguno relacionado con el PC):
[image]

Pieza impresa con la sesion de invitado:
[image]

Vídeo del ordenador de Blacky frito con luces de discoteca (perdón por el ZIP, pero GitHub no me deja adjuntar el vídeo tal cual):
[PC_Blacky_frito.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/816538/PC_Blacky_frito.zip)

Siotma, 2017-03-09:
Hoy, @David-Estevez y yo hemos estado intentando enlazar las sesiones de ASROB e invitado con éxito parcial.

Hemos descubierto que existe un script (no recuerdo ahora mismo la ruta ni el nombre) que se ejecuta con cada inicio de sesión de la sesión de invitado. Este script se puede modificar con permisos de sudo y añadir los pasos necesarios para enlazar la configuración de Repetier y Slic3r con la sesión de invitado mediante archivos .ini
Además, también se puede utilizar este script para confeccionar el escritorio de la sesión de invitado de manera que aparezcan por defecto algunas cosas como el acceso directo a Repetier o una carpeta con piezas de ejemplo.

Poco a poco, vamos avanzando.

Siotma, 2017-04-04:
Hoy tenemos reunión de Printer One ( #12 ) y mi idea es retomar lo dicho en el último comenario. Si podéis pasaros @David-Estevez o @jgvictores durante la tarde para repasar esto sería genial.

David-Estevez, 2017-04-04:
Como ya hemos quedado esta tarde en la nave, os puedo echar un cable cuando terminemos la reunión con Javi

Siotma, 2017-04-04:
Creemos que el tema está en copiar los archivos values.xml

[image]

Se puede añadir código en el script guest-session-auto.sh que está dentro de la carpeta usr/lib/lightdm para realizar las copias siempre que se inicie sesión.

David-Estevez, 2017-04-04:
y no se podría copiar directamente el registry del repetier de un user a otro? y así dejamos de tener que controlar tanto archivo suelto...

Siotma, 2017-04-05:
No sé lo que es el registry... pero cualquier prueba es bienvenida! :D 

Yo creo que todos tenemos claro el objetivo:
Que la sesión de invitado inicie Repetier con todo listo para darle al botón de Slic3r y al de imprimir y funcione.

La forma de conseguirlo, es lo que estamos buscando y, por cierto, el primero que lo consiga, se llevará una gran admiración general 🤣

David-Estevez, 2017-04-05:
Seguramente no me expliqué bien. Quería decir la carpeta  `repetier` de dentro de la carpeta `registry` de mono (que según la imagen que has puesto parece que contiene todos los archivos `values.xml`.

Imagino que esta carpeta lo que hace es emular el registro de Windows, que no existe en GNU/Linux.

Siotma, 2017-04-05:
Ah, comprendo, sí sí, totalmente de acuerdo. Yo incidí en los archivos xml porque son lo realmente importante pero obviamente una sola copia de la carpeta repetier cumple con todos los requisitos.

Siotma, 2017-06-07:
Hoy, he conseguido darle vida a la sesión de usuario, tras echarle unas cuantas horas.

Voy a describir lo que he hecho para que no se me olvide. La intención es que se pueda programar un script que lo haga todo automáticamente, pero para eso hoy no tengo tiempo ni conocimiento. Ahí va:

1. La sesión de usuario necesita 3 carpetas: 

- `/home/asrob/.Slic3r` Es donde se guardan los perfiles de Slic3r: carpetas print, printer y filament.
- `/home/asrob/.mono/registry/CurrentUser/software/repetier` Es donde se guarda la configuración de Repetier, para que según se abra, todas las opciones aparezcan preseleccionadas.
-  `/home/asrob/.local/share/RepetierHost` Es donde se guarda la configuración de Cura, entre otras cosas. No hace falta para que funcione con Slic3r.

2. Dichas carpetas deben copiarse a una intermedia a la que tenga acceso la cuenta de invitado. En nuestro caso estamos usando `/usr/share/asrob` .

- Para evitar conflictos con las rutas, deben eliminarse los siguientes campos de los siguientes archivos:

| Nombre del campo | Nombre del archivo | Ruta del archivo |
|---|---|---|
| `lastLoadGCodeFolder` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier` |
| `lastLoadModelFolder` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier` |
| `workdir` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier` |
| `soundTemperature` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier/window` |
| `soundSoundCommand` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier/window` |
| `fileHistory` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier/window` |
| `soundError` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier/window` |
| `soundPrintPaused` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier/window` |
| `soundPrintFinished` | `values.xml` | `~/.mono/registry/CurrentUser/software/repetier/window` |

- Todos estos valores tienen en común que son rutas a las que no tiene acceso la sesión de invitado. Repetier al detectar que faltan, los regenera con valores correctos en la primera ejecución.

- Todas las tareas descritas en el punto 2 deben automatizarse con un script que se tendría que ejecutar con privilegios de administrador (para realizar las copias y modificaciones de los archivos).

3. Dichas carpetas deben copiarse a la carpeta local del invitado, mediante las siguientes instrucciones al final del script `guest-session-auto.sh`, que es el que crea la sesión de usuario y está localizado en `/usr/lib/lightdm` ; mediante el código:

 `# cp -r /usr/share/asrob/RepetierHost ~/.local/share/ #en principio solo para cura`
 `mkdir -p ~/.mono/registry/CurrentUser/software`
 `cp -r /usr/share/asrob/repetier ~/.mono/registry/CurrentUser/software`
 `cp -r /usr/share/asrob/.Slic3r ~/`

Ahora mismo la sesión de invitado es funcional para imprimir! 🎉

Siotma, 2017-06-08:
Tal y como comenta @pamarinp en https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-306917681 , la sesión de invitado tiene activado el auto log-out. Esto hace que sea técnicamente imposible volver a la sesión ya que si se inicia de nuevo como invitado creará otra cuenta distinta. Aunque la impresión sigue funcionando, el Repetier desaparece de la pantalla por lo que no es posible realizar tareas básicas como cancelar o parada de emergencia.

Habría que buscar la línea de comandos necesaria para desactivar el auto logout y añadirla al script de creación de la sesión de invitado `guest-session-auto.sh`.

Siotma, 2017-09-28:
En la reunión de Printer One de la semana pasada, se añadió una línea adicional al script para desactivar al auto-logout; `export TMOUT=0`, tal y como aparece con una búsqueda rápida en Google.

Sin embargo, hoy, @davidvelascogarcia y yo hemos descubierto tras varias pruebas, que esta línea no cancela los temporizadores, que mantienen su valor por defecto (ver foto), por lo que no es válida por si sola.
[image]

Hemos localizado el comando que controla los temporizadores en el fichero `tmp/guest-xxxxx/.config/autostart/screensaver-settings.desktop` (ver foto) y hemos añadido una nueva línea que los cancela `xset s 0 dpms 0 0 0` en el script de creación de la sesión de invitado `guest-session-auto.sh`. Con esta nueva línea, los temporizadores se desactivan correctamente y el auto-logout ya no salta nunca. 

[image]

En esta issue hay actualmente toda la información necesaria para configurar el PC por lo que ya se puede comenzar con la configuración del PC de Hija Resurrection (ver #3 ).

Únicamente quedan dos mejoras que hacer al script, desde mi punto de vista:
- Programar la copia o creación en el escritorio de la sesión de usuario de un acceso directo a Repetier.
- Programar la copia o creaciónen el escritorio de la sesión de usuario de una carpeta con piezas de muestra y de calibración.

Además, quedaría automatizar la exportación de los perfiles desde la sesión de administrador hasta la sesión de invitado, mediante un script tal y como indiqué en este mensaje: https://github.com/asrob-uc3m/impresoras-asrob/issues/2#issuecomment-306881958

David-Estevez, 2017-10-01:
No sé si esto está hecho ya o no, pero creo que sería bueno tener una copia del script `guest-session-auto.sh` y de todos los archivos de configuración que se hayan modificado en este repo por si las moscas.

davidvelascogarcia, 2017-10-03:
@David-Estevez
El Script guest-session-auto.sh, ya está disponible en el repositorio en el apartado de Blacky, con un añadido además en el readme.txt, comentando la funcionalidad del Script.
El Script se encuentra en el ordenador que controla la impresora Blacky, pero a su vez, dado que se ha clonado el disco duro de este ordenador al nuevo ordenador que controlará Hija Resurrection, también se encuentra incorporado en ese.

davidvelascogarcia, 2017-10-03:
Tras haberlo consultado con @Siotma , podría ser una buena incorporación de seguridad, realizar un backup del sistema operativo con todos los programas instalados y configuraciones en el estado actual, en formato ISO.
En caso de suceder cualquier tipo de problema con alguno de los ordenadores, dado que son clones, se podría re instalar rápidamente para que volvieran a ser funcionales en el menor tiempo posible.

David-Estevez, 2017-10-03:
Ocupa mucho la ISO del PC?

jgvictores, 2017-10-03:
En todo caso, recomiendo fsarchiver (fsa), pese a que en RobotDevastation tiremos por dd. :smile:

davidvelascogarcia, 2017-10-04:
@David-Estevez , creo recordar que con el S.O, datos y configuraciones ocupaba al rededor de 8 GB. Podría servir para luego hacer re-instalaciones desde un USB booteable, o desde un DVD de doble capa.
@jgvictores , sí, además con fsarchiver, no habría ningún problema de licencias al ser open source, y con el añadido de que el resultado lo da en un fichero comprimido. Lo único, como puede dar problemas al hacer un backup desde una unidad montada, habría que usar otro PC de apoyo como con el clonado de disco de Blacky a Hija Resurrection, pero no debería de durar más de 30 minutos el proceso,

David-Estevez, 2017-10-04:
Siendo 8GB, si se podría tener el backup y además tenerlo subido a algún sitio online. (Incluso, a unas malas, subirlo al drive de ASROB en UC3M que tiene espacio infinito).

Gran trabajo, @davidvelascogarcia !! :clap: :clap:

jgvictores, 2017-10-04:
> como puede dar problemas al hacer un backup desde una unidad montada, habría que usar otro PC de apoyo como con el clonado de disco de Blacky a Hija Resurrection, pero no debería de durar más de 30 minutos el proceso

Lo habitual es arrancar de una distro Live (cualquier USB instalador de Ubuntu Xenial o anterior), y hacer el fsa/dd/etc desde ahí. Sobre todo, para no tener que estar conectando/desconectando discos duros. :-)

davidvelascogarcia, 2017-10-04:
@David-Estevez
Si, pienso lo mismo, en un disco duro, DVD o USB, podría acabar perdiéndose. Había pensado que junto al Drive, subirlo al github, pero desconozco la capacidad ahora mismo. Por tenerlo en varios sitios.
@jgvictores
Totalmente de acuerdo, y así hay menos complicaciones.
Ayer tuvimos problemas al bootear desde USB, aunque la placa lo permite. De todos modos acabo de acordarme de que tengo varias distritos de Ubuntu live en DVD RW, por lo que podría llevarlas y añadir alguna versión particular para estar seguros de poder hacerlo a la primera.

David-Estevez, 2017-10-04:
Quizá https://git-lfs.github.com/ ? O el SVN de ASROB, @jgvictores ?

jgvictores, 2017-10-04:
My preferences, de más a menos:
1. https://git-lfs.github.com/ suena bien, a ver si lo podemos usar.
1. Directamente como fichero (sin SVN tan siquiera) en https://sourceforge.net/projects/asrob/files/?source=navbar
1. Ojo que he metido ficheros enormes sin problema en SVN, así que en SVN de https://sourceforge.net/projects/asrob
1. Y ya lo peor, tirar de algo del server que a fecha de hoy aloja asrob.uc3m.es (del que me gustaría dejar de depender).

davidvelascogarcia, 2017-10-09:
@Siotma 
En la sesión de hoy, se ha comenzado el proceso de backup del sistema operativo, aplicaciones y configuraciones con el fin de generar una ISO con la cual restaurar los ordenadores, aprovechando la ventaja de ser clones, en caso de que surgiera cualquier tipo de problema.
En un inicio, se comenzó con la carga de un DVD Live, distro Ubuntu 14.04 i686 x32 bit, así como añadió el repositorio de manera temporal con el fin de instalar el fsarchiver del mismo modo. Sin embargo, al no encontrar el packcage, se procedió con el método secundario, la creación de copia desde las unidades de disco, encontrando a su vez el error de ficheros.
Por último, se procedió con un método funcional desde el terminal, la generación de una copia con el comando dd. El proceso de copiado, tuvo que detenerse, dado que en ese método se desconocía el porcentaje de progreso, el cual al finalizar se descubrió que era de aproximadamente el 50%, a costa de un largo tiempo.
Aprovechando la sesión del día de mañana para organización y limpieza, dado que los equipos estarán disponibles, y no se requerirá de supervisión de los mismos, se procederá con el backup en segundo plano.
Para evitar inconvenientes, se ha buscado un sustituto también en CD Live, también Open Source, concretamente, Clonezilla, que permitirá realizar el backup en formato ISO, y así será más sencilla su reinstalación.
Tras haber consultado los manuales del software, en principio mañana no deberían de surgir problemas adicionales, y podría guardarse el backup en los servidores.

Morgloz, 2017-10-09:
Hay un par de opciones para monitorizar el comando dd, nunca recuerdo cuál, en la próxima reunión las explico

jgvictores, 2017-10-09:
De memoria, diría que `ddrescue` con alguna opción verbose `-v`, aunque también había hacks de consola sobre `dd`.
`fsarchiver`, pese a que Clonezilla no está mal, me sigue pareciendo la mejor opción. En trusty (14.04) simplemente hay que tener activados loa repositorios Universe, ¿podéis comprobarlo?

David-Estevez, 2017-10-09:
A `dd` le puedes mandar una señal SIGUSER1 para que te muestre el progreso.

Conociendo

David-Estevez, 2017-10-09:
Fallo mío, confundí "close" con "cancel" :sweat_smile:

davidvelascogarcia, 2017-10-11:
@jgvictores @Siotma @David-Estevez 
En la sesión extraordinaria de ayer, en paralelo con la limpieza y organización, se continuó con el proceso de backup, esta vez de forma exitosa. Dado que ayer aún no entendía del todo bien, como emplear fsarchiver, con vistas a una medida temporal, hasta entender como emplearlo procedí con el backup desde Clonezilla.
Se realizó el backup del PC de Hija Resurrectión, posteriormente, tras cambiar el lector del DVD del ordenador antiguo de Hija Resurrection al de Blacky dado que este lector estaba dañado, se procedió con el backup de Blacky.
Tras comprobar que en PC de Blacky funcionaba con ambas impresoras correctamente, y con el fin de probar que funcionan correctamente los backups, y aprovechando que son clones, se restauró el PC de Hija Resurection con el backup de Blacky, sin dar ningún problema. 
Actualmente ambos ordenadores controlan sin problemas ambas impresoras en modo administrador, por otro lado en modo invitado Blacky controla ambas, pero Hija Resurrection, solo la impresora de Blaky y no la de Hija Resurrection.
En la próxima sesión se podría realizar el backup adicionalmente con fsarchiver, que había sido la primera idea, dado que he conseguido entender un poco mejor el funcionamiento.
De todos modos, por seguridad podría guardarse además el backup hecho con Clonezilla que solo ocupa 2.1 GB.
He intentado subirlo a sourgeforge, pero no tengo usuario ASROB en esa plataforma, por otro lado, en  https://git-lfs.github.com/ , no entiendo muy bien como hacerlo.
De momento he optado por subirlo al drive de alumno de la universidad, por si hubiera un drive de ASROB, pasarlo directamente o compartirlo.
Por cierto, he documentado el ANEXO 3 y el ANEXO 4 explicando con detalle el proceso de backup y de restauración en el manual de mantenimiento, y en la carpeta de backup de impresoras hecho un índice para redirigir a la descarga cuando este del backup y los anexos.
Quedo a la espera de vuestra respuesta. 
Un saludo.

Siotma, 2017-10-13:
Añadir que la imagen ya se encuentra en el Drive de ASROB (dominio UC3M) y se ha enlazado desde el repo vía 4679c574c4e6e30baf7de19aec925fd5b912a3cf .

Siotma, 2017-10-17:
A mi ver, lo único que queda para tener los ordenadores perfectamente configurados es que la sesión de invitado se inicie con un par de cosas en el escritorio:
- Un accceso directo a Repetier, con logo y todo, para que se vea bien.
- Una carpeta con piezas de prueba. Pueden ser piezas típicas de calibración y como mucho alguna figura pequeña.

jgvictores, 2017-10-17:
Re @davidvelascogarcia 
> He intentado subirlo a sourgeforge, pero no tengo usuario ASROB en esa plataforma, por otro lado, en https://git-lfs.github.com/ , no entiendo muy bien como hacerlo.

Si te creas un usuario en sf.net, te puedo agregar al proyecto.

Re @Siotma 
> Un accceso directo a Repetier, con logo y todo, para que se vea bien.

En robotDevastation, usamos un `.desktop` en `~/Desktop`: véase [robotDevastation.desktop](https://github.com/asrob-uc3m/robotDevastation/blob/2646b3079ac4d84904a2a8a3d8bd992293fb3718/scripts/gnome/robotDevastation.desktop) que a su vez enlaza con el [icono 64x64](rd-icon-64x64.png) (que debería instalarse en otro lado, pero así nos es más cómodo por cómo está montado).

Siotma, 2017-10-18:
Sí, el .desktop de Repetier ya existe, lo crea automáticamente con la instalación. La cosa es sacar un ratillo para configurar el script del inicio de sesión para que se copie en el escritorio. Gracias aún así por el apunte!

davidvelascogarcia, 2017-10-18:
@jgvictores 
He creado el usuario en Sourceforge. El nombre de usuario es davidvelascogar.
Link:
https://sourceforge.net/u/davidvelascogar/profile/

Un saludo y gracias.

jgvictores, 2017-10-18:
Añadido!

davidvelascogarcia, 2017-10-18:
@jgvictores 
Perfecto, muchas gracias.

jgvictores, 2017-10-28:
Respecto a [mis propuestas para almacenar ficheros grandes](https://github.com/roboticslab-uc3m/tiago-lfs), resulta que https://git-lfs.github.com/ no es excesivamente difícil de usar (se instala, se define mediante una línea de comando la extensión del tipo de fichero que va a ser grande, y luego es git vía consola normal), pero tiene una **limitación de 1-2 GB por fichero** (más comentarios míos [aquí](https://github.com/roboticslab-uc3m/best-practices/issues/13)). No nos valdría.
Por tanto, https://sourceforge.net/projects/asrob/ sigue teniendo sentido de momento.

davidvelascogarcia, 2017-10-29:
@jgvictores he creado una carpeta en Sourceforge, con nombre Backup_OS_3D_Printers_PC, para subir ahí el backup.
Los ficheros de configuración de la copia ya están subidos, sin embargo da limitación con el fichero grande de 2 GB, lo he dividido en 3 partes de 700 MB y una de 27 MB, ya está en proceso de subida.
Solo habría que descomprimir para unir esos 4 ficheros.
[https://sourceforge.net/projects/asrob/files/Backup_OS_3D_Printers_PCs/](url)

jgvictores, 2017-10-29:
¿Estás a tiempo de detener la subida? La limitación (que yo sepa/recuerde) es en GitHub, no en Sourceforge... 😅

davidvelascogarcia, 2017-10-29:
Está aproximadamente, como al 60%. 

[image]

Lo subo dónde prefiráis. :)

jgvictores, 2017-10-29:
Bueno, está bien. No conocía esa limitación de la subida vía interfaz web. De todas formas, dará menos problemas a la hora de descargárselo.

davidvelascogarcia, 2017-10-29:
Me ha dado problemas con la subida, respuesta del servidor con código 0.

[image]

jgvictores, 2017-10-29:
Ahora mismo veo un poco de caos en https://sourceforge.net/projects/asrob/files/Backup_OS_3D_Printers_PCs/
¿Podrías instalarte el Cliente de FTP de Filezilla (https://filezilla-project.org/download.php?type=client) y tratar de gestionar todo desde ahí?

De https://sourceforge.net/p/forge/documentation/SFTP/ entiendo que el servidor al que te conectas es **frs.sourceforge.net** y las credenciales las tuyas.

davidvelascogarcia, 2017-10-29:
He hecho limpieza en la carpeta del backup, he borrado todos los ficheros.
Lo estoy subiendo todo en un único Zip como el de Drive,  desde el cliente Filezilla.
Tardará un poco, tengo poca velocidad de subida.
@jgvictores , tienes razón, desde subida FTP, no hay limitaciones de tamaño, al menos no me ha notificado nada al subirlo.
He tardado un poco, porque no me aparecía directamente la carpeta del proyecto.
La ruta era /home/pfs/project/asrob/...
:).

jgvictores, 2017-10-29:
Cierto! Una vez conectado, hay que dar un par de rodeos en la estructura de carpetas del servidor hasta encontrar el proyecto en cuestión... ¡Bien visto!

[10 more replies]

closed 2018-09-19 by dmarcosm