asrob-uc3m/impresoras-asrob #3: Configuración PC nuevo de Hija Resurrection

jgvictores, 2016-06-30:
Se están configurando los PCs nuevos de Blacky e Hija Resurrection (véase [1](https://github.com/asrob-uc3m/impresoras-maintenance-manual-i3/blob/master/anexo_1_archivos_de_configuracion.md) y [2](https://github.com/asrob-uc3m/impresoras-maintenance-manual-i3/blob/master/anexo_2_sistema_operativo_y_sofware.md)).
Tareas pendientes:
- [ ] Montar equipo
- [ ] Instalar sistema operativo
- [ ] Instalar Cura
- [ ] Mejorar resolución de pantalla
- [ ] Instalar Slic3r para Repetier

Siotma, 2017-02-19:
Si con "replicar lo hecho hasta ahora en Blacky", se pretende instalar Lubuntu y quitar Windows, me parece bien, ya que el sistema de sesiones de invitado es una gran ventaja.
Aún así, creo que es mejor primero completar la configuración del PC de Blacky y documentarlo bien, ya que al menos este PC ahora funciona.

jgvictores, 2017-02-19:
@Siotma Sí, se pretendía instalar Lubuntu y quitar Windows. Totalmente de acuerdo, primero completar la configuración del PC de Blacky y documentarlo bien.

Siotma, 2017-06-24:
Ahora que ya sabemos configurar Lubuntu ( #2 ), resucito este hilo para hacer lo propio con el PC de Hija Resurrection.

jgvictores, 2017-06-24:
Ok, en uno de los armarios debería estar un PC similar al que ahora mismo tiene Lubuntu, para replicar el proceso (y dar de baja el equipo que ahora mismo lleva Windows).

David-Estevez, 2017-06-26:
El PC del armario tiene RAM? No se la llevó el ex-tesorero al recibir la donación?

jgvictores, 2017-06-26:
Mmm creo que había ocurrido eso, y entre Fernando el técnico y yo lo habíamos tratado de solucionar. Por favor revisadlo, y si no, volvemos a comentar con Fernando.

Siotma, 2017-09-28:
Esto hay que resucitarlo urgentemente. Ahora que tenemos gente en Printer One hay muchas manos para pocas cosas que hacer, podríamos encargarnos de esto en la próxima reunión.

Si hay que utilizar un nuevo equipo, agradecería que se nos indicara cual es, y también dónde localizar el disco o los medios de instalación del sistema operativo Lubuntu.

Siotma, 2017-09-30:
@David-Estevez y @jgvictores . Necesitaríamos de vuestra colaboración!

jgvictores, 2017-09-30:
Hi!
Lo comentado arriba: en uno de los armarios debería estar un PC similar al que ahora mismo tiene Lubuntu. Sólo hay que revisar si tiene RAM. En cuanto a cómo instalar el sistema operativo, aquí enlace: https://asrob-uc3m.gitbooks.io/impresoras-maintenance-manual-i3/content/anexo_1_sistema_operativo_y_sofware.html

daavm, 2017-10-03:
Holi,
Tras clonar los discos y conseguir que Lubuntu funcionase en la nueva torre, y tras ver que Repetier no conectaba con la impresora, hemos optado por instalarlo de nuevo (Repetier) y aprovechar para poner una versión más actualizada (2.0.5). El problema después ha sido el hacerlo funcionar en la sesión de invitado. Al haberlo instalado usando el AppImage (fallo por mi parte, porque necesita permisos para ejecutar), ha sido un dolor de cabeza que no he sabido resolver. 
Sin embargo, creo tener la solución y espero que el próximo dia podamos solucionarlo. Con esto, y si no me equivoco, creo que el PC estaría listo para su uso y disfrute.

Siotma, 2017-10-05:
El contenido de este mensaje se sitúa cronológicamente antes que el de @daavm.

En la [última reunión de Printer One](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora#3_de_octubre_de_2017) decidimos sacar el nuevo PC de Hija Resurrection y empezar a configurarlo.

En primer lugar lo abrimos para comprobar que es exactamente igual que el de Blacky:
- Torre Acer con 1GB de RAM DDR2 (2 ranuras, queda una libre); y procesador Pentium D de doble núcleo a 3GHz.
- Venía con WinXP, procedente de los laboratorios de los autómatas industriales. Se ha intentado instalar [Lubuntu](https://help.ubuntu.com/community/Lubuntu/GetLubuntu) mediante USB bootable pero no ha sido posible al no poder ejecutarlo durante el arranque.
- finalmente, se ha optado por clonar el Disco Duro que tiene el PC de Blacky, dado que ambas torres son idénticas. Para ello se ha utilizado un tercer PC con winXP y Easeus TODO Backup. Satisfactoriamente.
- La versión clonada de Lubuntu es totalmente funcional salvo porque al intentar conectar con Hija Resurrection esto no ha sido posible. Por ello se ha optado por sustituir la antigua versión de Repetier por una más moderna, y más facil de manejar, en formato AppImage. Actualmente estamos trabajando en la puesta a punto de esta, tal y como explica @daavm en el anterior comentario.

Actualmente, ambas torres se encuentran sobre la mea de Hija Resurrection.

[image]

Siotma, 2017-10-13:
Dicho por @daavm : 

> En la sesión del lunes 9, y dado que usando el appimage de Repetier no estaba dando resultados, nos decidimos por reinstalarlo con la versión habitual. Se consiguió poner un ".sh" en el escritorio de la cuenta guest pero el acceso directo no acababa de funcionar como queríamos, así que el próximo día se pretende o terminar de apañar el sh o buscar otra solución.
El martes 10, dado que al final restauramos todo el sistema del pc de blacky, la cuenta guest ya tiene repetier totalmente funcional. Nos falta darle acceso a la cuenta al usb de la impresora hija a través del archivo rules, lo cual haremos en la próxima reunión.

Ambos PCs vuelven a ser clones ya que se usó Clonezilla para restaurar la imagen obtenida del PC de Blacky. Hija Resurrection conecta perfectamente desde la sesión de administrador, pero no desde la sesión de invitado. Enseguida nos hemos dado cuente de que se trata de un problema de permisos y nos hemos acordado del fichero rules.

Hemos utilizado como inspiración [este maravilloso comentario](https://github.com/asrob-uc3m/impresoras-asrob/issues/2#issuecomment-283737116) de @jgvictores .

El problema que hemos encontrado es que, mientras que la placa Arduino de Blacky se reconoce como un dispositivo `dev/tty/USB0`, la placa Arduino de Hija Resurrection aparece con un nombre mucho más largo, algo así como `serial/usb/by-id/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX`. Hemos actualizado los números de idProducto y de idVendor, pero no ha sido suficiente. Yo imagino que en vez de poner `SUBSYSTEM=="tty"` habrá que poner otra cosa, pero no hemos sido capaces de acertar con lo que poner en el fichero rules para que se active al conectar la impresora. 

¿Ideas @David-Estevez @jgvictores @Morgloz ?

jgvictores, 2017-10-15:
¿Habéis probado con simplemente omitir `SUBSYSTEM=="tty", `?

Siotma, 2017-10-15:
🤣 No, eso no lo probamos. Gracias por la idea 😊

daavm, 2017-10-17:
Dicho por @jgvictores:

>¿Habéis probado con simplemente omitir SUBSYSTEM=="tty", (además de actualizar los números de idProduct y idVendor)?

Efectivamente, hemos quitado el atributo SUBSYSTEM y la cuanta guest ya ha reconocido y conectado a la impresora. Así mismo, se ha probado a mandar una impresión y la ha empezado. La impresora hija quedaría totalmente funcional pendiente del cambio/arreglo de la fuente.

[image]

Siotma, 2017-10-17:
Además de lo dicho en https://github.com/asrob-uc3m/impresoras-asrob/issues/2#issuecomment-337302357 , en este PC queda también por hacer:
- Actualizar los perfiles de impresión de Slic3r por los que hemos recuperado de su antiguo PC. Creo que al final nos hemos ido sin hacerlo @daavm corrígeme si me equivoco.

davidvelascogarcia, 2017-10-18:
@Siotma 
No sabía muy bien donde comentarlo, dado que no hay una issue como tal para ello.
Respecto a lo que estuvimos hablando el último día respecto de la nueva fuente para Hija Resurrection, la fuente de alimentación que era igual a la de Blacky.
Durante la sesión, se comprobó que la alimentación, era la correcta en todos los canales 3.3, 5, 12 V.
Tras separar el ventilador de la fuente, se comprobó que el fallo y el ruido era electrónico, y que al volver a soldar el ventilador, se incrementaba.
El ruido, me recordaba al de la bobinas a altas frecuencias, sin embargo, el fallo típico suele ser por condensadores, aunque a simple vista no estaban hinchados ni dañados.
Se me ha ocurrido, si os parece bien, para la próxima reunión.

1.  Aprovechando el osciloscopio de la nave, medir los condensadores y ver si hay picos en la señal y el condensador esta dañado y no filtra correctamente.
2. En caso de no ser el condensador, y ser las bobinas, intentar identificar el origen del ruido y calzar el solenoide en cuestión para disminuir el ruido.

¿Que os parece?.

jgvictores, 2017-10-18:
En el armario hay más fuentes, preferiría que se usasen frente al peligro de uno `refurbished`. ¿Serviría alguno del armario?

Siotma, 2017-10-19:
Como dice @jgvictores , la opción de utilizar la fuente tras haberla manipulado, a mi tampoco me convence. Las impresoras manejan mucha potencia y no queremos que constituyan un peligro. Las fuentes seleccionadas son las que mayor potencia nominal tienen en el canal de 12V, y por tanto si funcionan correctamente son las que menor peligro representan ya que creo que son las únicas que cubren totalmente las necesidades de las impresoras.

He leído en foros, que las fuentes de alimentación a veces producen sonidos electrónicos cuando trabajan sin carga porque les cuesta estabilizarse. Aprovechando las resistencias que ha traído @JorFru en https://github.com/asrob-uc3m/impresoras-asrob/issues/33#issuecomment-337376829 , podemos comprobar este hecho, y si la fuente de alimentación sigue sonando seguramente acabemos buscando otra aunque no tenga tanta potencia.

JorFru, 2017-10-19:
> <https://github.com/jorfru> en #33 (comment)

En la siguiente reunión instalaremos las resistencias y comprobaremos con
osciloscopio si la salida de la fuente está bien regulada.
Las fuentes funcionan oscilando y algunos componentes pueden vibrar. En
principio estás vibraciones están fuera de rango audible, generalmente por
encima de los 300KHz.
Es posible que lo que estemos escuchando sea un subharmónico de la
frecuencia de oscilación. Lo comprobaremos próximamente.

davidvelascogarcia, 2017-10-19:
Perfecto, se puede probar a conectar con carga. Quizá lo que propuso @Siotma del lector de DVD también pueda servir para esta prueba.
Por mí parte @JorFru , si no te importa me uno a las comprobaciones y en lo que pueda te echo una mano. Así aprovecho, aprendo y aplicó desde un plano más práctico.

Siotma, 2017-10-30:
Respecto a lo que se ha venido comentando en los últimos comentarios, mejor seguir en #35 .

Vengo a contar que hoy, después de un buen rato, he conseguido perfeccionar el [script de la sesión de invitado de Hija Resurrection](https://github.com/asrob-uc3m/impresoras-asrob/blob/master/Hija Resurrection/guest-session-auto.sh).

Ahora, el script copia en el escritorio una carpeta con piezas de calibración de ejemplo, y también un acceso directo funcional a Repetier. Además, he perfeccionado el mensaje de aviso de que se trata de una sesión temporal.

[image]

La única cosa fea que tiene la sesión de invitado es que nada más crearse, envía un mensaje de error diciendo que no se encuentra un proceso con un PID determinado (cada vez uno distinto). 

[image]

No he sido capaz de hacer que este mensaje no aparezca, pero no da problemas, simplementa se da en Ok y la sesión es totalmente funcional.

Si os gusta cómo lo he dejado, el próximo día se puede replicar en Blacky en mucho menos tiempo.

Lo único que le queda a este PC es hacerle un backup en Clonezilla 😊

davidvelascogarcia, 2017-11-16:
@Siotma 
El nuevo backup de los ordenadores de ASROB, ya se encuentra disponible en Sourceforge. A su vez, se ha modificado el link del readme del repositorio de impresoras de Github para la descarga y sustituir la descarga de drive por la del repositorio de Sourceforge.
He aprovechado, y he pasado los anexos 3 y 4 del manual, de como realizar el backup o restaurar, a ficheros PDF, intentando dejarlos algo más preparados y subido al repositorio de Sourceforge junto al backup. A su vez,también hay una copia en el repo de Github de impresoras.

[image]

Figura 1. Ficheros Sourceforge

Adjunto los links:

Backup en Sourceforge:
[https://sourceforge.net/projects/asrob/files/Backup_OS_3D_Printers_PCs/?upload_just_completed=true](url)

Fichero PDF de proceso de Backup
[https://github.com/asrob-uc3m/impresoras-asrob/blob/master/Backup%20S.O%20configuraciones%20y%20aplicaciones%20de%20equipos%20de%20impresion/Proceso%20de%20backup.pdf](url)

Fichero PDF de restauración de Backup
[https://github.com/asrob-uc3m/impresoras-asrob/blob/master/Backup%20S.O%20configuraciones%20y%20aplicaciones%20de%20equipos%20de%20impresion/Reinstalacion%20de%20sistema%20operativo.pdf](url)

davidvelascogarcia, 2017-12-06:
@jgvictores El README.md esta actualizado. He borrado la carpeta, y he dejado solamente el README.md con los ficheros de backup, adjunto [link](https://sourceforge.net/projects/asrob/files/Backup_OS_3D_Printers_PCs/).

jgvictores, 2017-12-07:
> El README.md esta actualizado. He borrado la carpeta, y he dejado solamente el README.md con los ficheros de backup, adjunto [link](https://sourceforge.net/projects/asrob/files/Backup_OS_3D_Printers_PCs/).

@davidvelascogarcia Está realmente genial. Gracias!

davidvelascogarcia, 2017-12-07:
Me alegro. Con tal de ayudar.

dmarcosm, 2018-09-19:
Para evitar tener issues duplicadas, cierro esta issue y continuamos en https://github.com/asrob-uc3m/impresoras-asrob/issues/78, donde seguiremos con la configuración y documentación del PC que tenemos actualmente.

closed 2018-09-19 by dmarcosm