asrob-uc3m/impresoras-asrob #4: HIJA RESURRECTION problema con pasos por mm

jgvictores, 2016-11-30:
Nota: Issue puesta hace 28 días por @jorgerc21 migrada desde https://github.com/asrob-uc3m/impresoras-maintenance-manual-i3/issues/2

Today @Rodriwp and me have been testing Hija Resurrection trying to fix it, these are the problems we have found:

- Excesive amount of material extruded: To solve it we have decreased the extrusion speed (From 250 to 150 -recommended value- and then to 50) It improved the printing quality, but not enough.

- Second layer is printed "inside" the first one, the extrusor does not lift to keep printing.

- Noisy motors

- Printer miss-recognizes dimensions

@Siotma suggested changing some config in Configuración/Configurar Firmware EEPROM in the fields Pasos por mm form 80-80-400-775 to 160-160-8000-315 and it worked perfectly, but these changes although being saved, are dismissed when the printer is reconnected.

Additionally, we have replaced the init files in the PC by the ones in the guide before knowing this info.

Pending to be done: 

- [ ] Make EEPROM changes persistent 

- [ ] Get further information of why this problem happen and document it

jgvictores, 2016-11-30:
Esto es, la solución actual es: cambiar `Configuración/Configurar Firmware EEPROM` en los campos `Pasos por mm` de `80-80-4000-775` a `160-160-8000-315`. Esto debe hacerse cada vez que se reinicia el PC.

David-Estevez, 2016-11-30:
Curiosamente, la configuración actual es de `165-165-4200-800`, pero sigue sin poder cambiarse... ¿Ha cambiado alguien dicha configuración?

UAVs --> @andrespy 
City of ... --> @Rodriwp

Rodriwp, 2016-11-30:
City of Robots no ha cambiado configuraciones en la impresora en su horario de reunion.

David-Estevez, 2016-11-30:
:+1: 

Es sobretodo para que nos diga cómo diantres lo ha conseguido, que estamos intrigados :joy:

Dani191995, 2016-12-01:
Estuve un par de dias intentando arregrarlas lo que hice fue meterles el marlin que hay en kitprinter, y usar pronterface y cura, con respecto al marlin la impresora extruia mucho por lo que le cambie los valores que no podeis cambiar, lo que me fije es que el eje z no subia lo necesario cambiase lo que cambiase, el eje x e y parecen ir bien, el tema de que no lo podais cambiar me imagio que es usando repetier o pronter si no me equivoco, entonces a lo mejor no os deja cambiarlo  por algo en el codigo, no se porque no os deja cambiar los valores.

Siotma, 2017-02-08:
Hoy he probado Hija Resurrection y no parece que siga teniendo este problema. Aún así no lo he comprobado del todo. Sugiero que alguien lo compruebe y cierre esta Issue.

David-Estevez, 2017-02-08:
Mencionaremos a nuestro actual responsable ( @jorgerc21 ) para que se pase por allí cuando pueda y lo compruebe.

Siotma, 2017-02-16:
Efectivamente, la configuración actual es 165, 165, 4200. Los ejes X e Y imprimen bien aunque un pelín grandes (yo creo que lo suyo es 160), y el eje Z imprime aproximadamente la mitad, por lo que yo apostaría por 8000.

@Dani191995 entonces, ¿tú cogiste y pusiste un firmware que pillaste por ahí o como? ¿Tienes una copia del firmware que pusiste?

Dani191995, 2017-02-16:
Cogí el firmware de la página kitprinter, porque iban mal los pasos y
después como iba igual cambie los valores.

Siotma, 2017-02-16:
De acuerdo, gracias por aclararlo. Esto enlaza con la issue #1. Desconozco si los firmware de las impresoras están documentados en algún sitio. En caso negativo, en el grupo de trabajo ( #12 ) lo tenemos como uno de los objetivos principales.

Siotma, 2017-03-02:
Hoy, hemos flasheado este código a Hija Resurrection.

No hemos podido realizar pruebas de impresión aún, pero animamos a todos los usuarios a que las hagan.
[Marlin-1.0.2-2_Hija_Resurrection_2_Marzo.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/814002/Marlin-1.0.2-2_Hija_Resurrection_2_Marzo.zip)

Siotma, 2017-03-02:
Las pruebas de impresión realizadas por @JorFru y reflejadas en #11 demuestran que el problema finalmente se ha subsanado al flashear el nuevo firmware.

closed 2017-03-02 by Siotma