asrob-uc3m/impresoras-asrob #6: BLACKY no se sabe lo que le pasa

jgvictores, 2016-11-30:
En fin, que no se sabe lo que le pasa.

Siotma, 2017-02-08:
Ya se sabe lo que le pasa. Consultar #11. Cierro.

JorFru, 2017-03-07:
Blacky dió problemas las dos veces que fui a imprimir con ella.
La primera capa la imprime bien. Al pasar a la segunda, parece que no sube lo suficiente en el eje Z. Atropella lo que había impreso antes, y cuando intenta hacer la tercera capa todavía peor. Parada de emergencia antes de romper algo.
Se probó cambiando la altura de capa a casi el doble y nada. Se revisó la configuración de pasos/mm y no aparecían valores en la pantalla. Se puso en pasos/mm del eje Z el mismo valor que tiene Hija y tampoco hemos conseguido que suba de altura para hacer la segunda y sucesivas capas.

Actualmente inoperativa a no ser que quieras imprimir algo de una capa.

andrespy, 2017-03-07:
Los pasos por mm estan bien puestos en el Marlin? Podria ser una causa

Siotma, 2017-03-07:
La cuestión es que el viernes pasado estuve imprimiendo sin errores,
entonces es un fallo que ha aparecido entre el viernes por la mañana y el
lunes.
Este jueves investigaremos y probablemente le flashearemos un firmware
nuevo para tener las dos impresoras con la misma versión.

Morgloz, 2017-03-09:
Era un problema con Slicer, estaba configurado con un offset de -4. Ya está corregido
[image]

closed 2017-03-09 by Morgloz

JorFru, 2017-03-09:
Ayer probé a imprimir con un perfil de Slic3r que hice a posta y llamé
JorFru-prueba. Ajusté el parámetro Zoffset a 0 y aunque mejoró un poco la
cosa, seguía arrastrando mucho el extrusor por la pieza. Parada de
emergencia a la tercera capa por riesgo de rotura.

Si decís que está bien cojonudo. Igual que el problema llegó, se puede ir:
fugazmente.

Qué desajustes más raros le ocurren a esa máquina...