asrob-uc3m/impresoras-asrob #7: Mejoras en página de "Gestión de Impresoras" de asrob.uc3m.es

jgvictores, 2016-11-30:
Estamos mejorando http://asrob.uc3m.es/index.php/Gestión_Impresoras

- [ ] Implementando algo como el diagrama de flujo de la figura.
- [ ] Implementando scripts que automatiza este proceso.

[image]

andrespy, 2016-12-01:
Ya es hora de contribuya un pelin
Aqui esta el diagrama:
[image]

Si alguien quiere dejarlo más bonito, tengo el archivo del programa dia, que es software gratuito.

andrespy, 2016-12-01:
Ya es hora de que contribuya un pelin
Aqui dejo el diagrama:
[image]

Esta hecho con el programa dia que es software gratuito asi que si alguien quiere mejorar el diagrama cuando sepa donde poner el archivo lo subo para que lo haga.

jgvictores, 2016-12-05:
Enlazado con grupo de telegram en wiki: http://asrob.uc3m.es/index.php/Gesti%C3%B3n_Impresoras

Siotma, 2017-12-09:
@David-Estevez , @jgvictores . Esta issue es muy antigua y no está en el repo que debe. Sin embargo creo que a día de hoy, ya está más que resuelta.

David-Estevez, 2017-12-10:
Yo movería los "enlaces útiles" al manual de usuario, pondría el cartel enlazando a la nueva página y con eso daría esta issue por cerrada.

jgvictores, 2017-12-10:
1. Teníamos la idea de sólo poder reservar si está logueado un operario. ¿Esto está hecho? 

2. Nunca hice nada parecido, pero en GitBook puedes acceder con una cuenta de GitHub (aparte de Google, etc). Si lo de (1) no está hecho, sería una opción, no?

Siotma, 2017-12-10:
> Teníamos la idea de sólo poder reservar si está logueado un operario. ¿Esto está hecho?

Actualmente, está hecho que únicamente personas identificadas con una cuenta de correo UC3M en Google pueden reservar.

> Nunca hice nada parecido, pero en GitBook puedes acceder con una cuenta de GitHub (aparte de Google, etc). Si lo de (1) no está hecho, sería una opción, no?

No entiendo cuál es el objetivo de utilizar cuentas de GitBook. Sin embargo, creo que las propuestas para perfeccionar el sistema de reservas deberían hacerse [aquí](https://github.com/asrob-uc3m/printers-status-webpage/issues/4).

jgvictores, 2017-12-10:
> No entiendo cuál es el objetivo de utilizar cuentas de GitBook. Sin embargo, creo que las propuestas para perfeccionar el sistema de reservas deberían hacerse [aquí](https://github.com/asrob-uc3m/printers-status-webpage/issues/4).

Sí, me referí a GitBook pero el del Issue mover es igual, como propones [en la issue correspondiente](https://github.com/asrob-uc3m/printers-status-webpage/issues/4#issuecomment-350535856). Totalmente de acuerdo con seguir los desarrollos ahí.

@Siotma Cierra esta issue sin problema, ¡gracias!

Siotma, 2017-12-10:
Wiki actualizada: http://asrob.uc3m.es/index.php/Gesti%C3%B3n_Impresoras

Manual de usuario actualiado: https://asrob-uc3m.gitbooks.io/impresoras-user-manual-i3/content/enlaces_utiles.html

closed 2017-12-10 by Siotma