asrob-uc3m/impresoras-asrob #8: Software de automatización de estado de impresoras

David-Estevez, 2016-11-30:
Se desea desarrollar un software que corra en el servidor de ASROB y que realice la laboriosa tarea de automatizar la página de impresoras de la wiki (ver issue #7 ).

Para ello, propongo desarrollar un bot que viva en el servidor de ASROB. Su funcionamiento será muy sencillo:

 

1. Cuando se estropee una impresora, el usuario deberá poner una issue en este repositorio, con las etiquetas "no funciona" y la etiqueta de la impresora.
2. Github notificará con un webhook a nuestro bot que se ha creado una issue nueva.
3. El bot usa la API pública de Github para encontrar los datos de la issue, y los escribe en dos páginas plantilla (template) de la wiki, una por cada impresora actual. En esa template se escribirá un número:
    * Si la impresora funciona correctamente, -1
    * Si la impresora no funciona, el número de la issue en la que se detalla el error (Siempre >0)
4. La wiki usa esa plantilla para generar automáticamente una página según lo descrito en #7

David-Estevez, 2017-04-11:
A modo de primera aproximación, he creado https://github.com/asrob-uc3m/printers-status-webpage

Es una web estática, lo que significa que habría que hacer un cron job periódico para mantenerla actualizada. Como sólo implica una petición a la api de github, no creo que haya problema por actualizar la página 3/4 veces al día.

En el futuro se podría integrar con los webhooks de GitHub para que GitHub informase de nuevas issues o issues resueltas y la página se actualizase al nuevo estado.

Si os mola la página, me coordino con @jgvictores para integrarla lo antes posible en la web de ASROB.

David-Estevez, 2017-04-12:
Ya lo tenemos funcionando! http://asrob.uc3m.es/printers/

Se debería actualizar cada hora de las 8h a 22h (por si hay alguna incidencia durante el día). No es instantáneo pero debería servir por ahora.

Siotma, 2017-04-15:
Pequeños comentarios al trabajo realizado:

1. La web es una pasada total. Me encanta el aspecto, lo bien que se maneja con el móvil... todo!
2. El texto correspondiente al apartado "manual de operador" no es el adecuado, ya que es el mismo que el de "incidencias".
3. La hoja de excel a la que dirigen los botones de reserva es en ambos casos la misma, esto puede permitir que la gente realice reservas en una impresora no disponible.
4. La hoja de excel está desactualizada en fecha y en disponibilidad de impresoras.

5. Pese a que el sistema de la hoja de cálculo ha demostrado históricamente cumplir con su cometido, se trata de algo totalmente arcaico, y ya que hemos dado un lavado de cara a la página de gestión, propongo ir más allá y mejorar esa parte también.

Ideas:

- Programar una página con un formulario (por ejemplo, en [html](https://www.w3schools.com/html/default.asp)) en el que sea necesario introducir un correo de dominio uc3m. Se enviará un correo de confirmación a esa dirección para confirmar la reserva. De esta manera llevaremos un control de quién y cuánto uso se hace de las impresoras. En caso de que alguien sin cuenta uc3m quiera utilizar las impresoras, el usuario deberá incluir un texto explicando el motivo del uso y el sistema además de enviar el correo de confirmación, remitirá otro correo a algún miembro de ASROB para obtener autorización.
- Crear una base de datos que será atacada desde los formularios para realizar las reservas. En esta base de datos estará almacenada la información histórica de las reservas desde el momento de su creación. Cada semana, se programará la creación de nuevas entradas en la base de datos para dar cabida a las reservas de la siguiente semana.
- Esta base de datos será leida cada vez que la página del formulario se cargue, y mostrará en una tabla similar al documento excel los huecos libres para cada impresora. Sería genial que la forma de reservar fuera pulsando sobre las distintas celdas de la tabla, al igual que se seleccionan las fechas de una estancia en un hotel, o un viaje...

Logrando esto, habremos conseguido realmente automatizar la gestión de las impresoras y la cerveza que nos tomemos sabrá mucho mejor.

David-Estevez, 2017-04-15:
Muchas gracias por los halagos, @Siotma , realmente animan a seguir trabajando en ello :smile: 

Sobre los fallos que comentas (2, 3 y 4), voy a crear issues correspondientes en el [repositorio de la web de impresoras](https://github.com/asrob-uc3m/printers-status-webpage) para tenerlas centralizadas.

Sobre el punto 5, también pondré una issue, pero seguramente no se lleve a cabo de forma inmediata. Te explico, en mi opinión, por qué no es tan fácil de implementar:

* Al depender del correo, se introduce una dependencia en una persona de la asociación, que se tiene que encargar de ello. Si ya nos cuesta llevar al día el correo y las redes sociales, o incluso mantener al día la hoja de reservas, introducir una nueva responsabilidad creo que va a ser contraproducente.
* La base de datos ya existe (la hoja de reservas antigua) y no me consta que se haya consultado ninguna vez para un uso a posteriori (análisis, investigación de incidencias...).
* La página actual es una página estática. Se genera automáticamente cada hora a partir de las issues de GitHub. Las mejoras que tú dices estarían muy bien, pero tendríamos que pasar a una página dinámica, con un backend y una web. Habría que buscar o pensar en una buena interfaz para la reserva. Que sea intuitiva y que ya esté hecha, a ser posible. En definitiva, no es un "hotfix", es algo que llevaría bastante curro.

Por todo ello, creo que el sistema que propones molaría, pero no es algo que vaya a estar listo a corto plazo. Crear una página nueva de forma semanal es algo que lleva muy poco tiempo (y que se podría incluso automatizar). Como solución intermedia, si deseamos un mayor control de las reservas, mi propuesta es crear la hoja de reservas en la cuenta de la uni de ASROB, lo cual permite que sólo personas con cuenta UC3M puedan acceder al documento de reserva. O que se pueda ver de forma pública, pero sólo editar por gente de la UC3M. 

¿Qué te parece esta propuesta?

Siotma, 2017-04-18:
@David-Estevez , no era consciente de que la página tuviera su propio repositorio, a futuro realizaré los comentarios en las issues correspondientes.

Suscribo totalmente los comentarios que haces. Yo hacía las propuestas con vistas a futuro, y obviamente sujetas a debate previo, pensando en que en la asociación hay gente no solo de electrónica y automática, sino también de informática y otras disciplinas para los que realizar algo como lo descrito puede requerir mucho menos esfuerzo.

En un futuro optimista, con un grupo de trabajo activo en impresoras y un sistema de gestión más optimizado, las impresoras tendrán mucho más uso del que tienen ahora y por tanto, limitar el uso a personas de la UC3M tendría valor de cara a poder usar un seguimiento de las reservas e identificar los usuarios más activos y los menos activos. Utilizando las cuentas de correo UC3M se evita que un usuario cree múltiples cuentas para reservar un número de horas mayor, y el sistema de correos de confirmación refuerza este filtro de forma activa (que tal vez se pudiera automatizar) limitando las horas de reserva y comprobando que el usuario que pretende reservar es un operador o no. Fuera de la ventaja de evitar duplicación de cuentas, no tiene mayor sentido limitar el acceso a la hoja de reservas, máxime cuando no hemos tenido problemas de este tipo todavía. Como ya he explicado, esta propuesta hace frente a problemas que aún no han surgido, pero que si conseguimos atraer más gente a usar las impresoras, puede que surjan. Por ello no veo necesario restringir el acceso a la hoja en este momento.

Por otro lado, sí existe una clara ventaja en sacar estadísticas de uso de impresoras, ya que podría valer para justificar los gastos relacionados con ellas, pedir subvenciones, obtener financiación desde el Departamento, optimizar su rendimiento (detectar problemas más frecuentes, escuchar sugerencias de usuarios), e incluso plantear mejoras funcionales o ampliación del número de impresoras disponibles.

En mi línea de coordinador de grupo de trabajo de Printer One, quiero aprovechar este nuevo tirón de gente para implementar cualquier mejora que pueda facilitar el acceso a las impresoras y reducir las tareas de mantenimiento para intentar garantizar buena salud para las mismas durante mucho tiempo.

David-Estevez, 2017-04-18:
No problem! Yo me ofrezco para todas estas cosillas informáticas, pero son cosas que se irán haciendo muuuuy poco a poco (en los ratos libres que tenga, básicamente :sweat_smile: ).

¿Damos pues por cerrada esta issue?

Siotma, 2017-04-18:
Cierra sí 👍 ya duplico yo lo que he escrito en su sitio, y que esta Issue descanse en paz.

David-Estevez, 2017-04-18:
Guay! Cerrando issue entonces...

closed 2017-04-18 by David-Estevez