asrob-uc3m/impresoras-asrob #10: Cambiar el cable de alimentación del extrusor

Morgloz, 2017-02-02:
Tras tener que desatornillar el extrusor para desatascarlo llegamos a la conclusión de que habríamos tardado la mitad si el cable que va del circuito al extrusor acabara en un conector, en vez de ir directamente al extrusor.

David-Estevez, 2017-02-04:
¿¿¿El cable estaba soldado a la placa???

Morgloz, 2017-02-07:
No, estaba conectado a una clema, pero el cable atravesaba el armazón y estaba sujeto con bridas en varios puntos. Sacarlo nos costó más de 20 minutos, y ponerlo de vuelta más o menos lo mismo.

David-Estevez, 2017-02-07:
Ah, pues entonces la clema hace de conector :wink: 

Desmontar el extrusor es algo que no se hace muy a menudo (o no debería hacerse a menudo en condiciones normales), por lo que 20 minutos no me parece demasiado tiempo. Ahora que sabéis cómo va conectado seguramente tardéis menos incluso. 

El problema de añadir un conector es que se añade también otra forma en la que puede fallar la impresora, ya sea de tanto conectar/desconectar o en las uniones del cable con el conector. Me da la impresión que a la larga daría más problemas de los que solucionaría.

Siotma, 2017-02-08:
Imagino que esta Issue es para Hija Resurrection.

En vez de cambiar el cable, hoy he estado haciendo pruebas y he comprobado que el cable está dañado, puesto que dependiendo de su curvatura el calentador calienta o no. Esa es la causa de los atascos: durante la impresión al desplazarse el eje X, el cable se mueve y la temperatura oscila. Tengo una foto que lo demuestra. 
[image]

Sugiero hacer pruebas con el extrusor desmontado y un polímetro para medir resistencia. En caso de que sea muy difícil localizar o reparar el problema sugiero comprar un nuevo calentador.

Siotma, 2017-02-19:
Tal y como comenté en #11 , hemos confirmado que el calentador del extrusor está dañado, por lo que se debe proceder a su sustitución.

JorFru, 2017-02-21:
He hecho uso de un multímetro y he hallado que el problema estaba en los cables. @Siotma me ha ayudado a encontrarlo.

He saneado los cables, retirando la parte que daba problemas (roto el cable por dentro) y poniendo un fragmento nuevo.

He probado durante diez minutos con la impresora en movimiento y parece que ya no da problemas.

[image]

Ahora bien, he puesto una clema para sujetar la cinta de kapton vieja y que a su vez sujeta el termistor. Habría que ponerle cinta nueva antes de que mi solución temporal salga ardiendo.

[image]

Siotma, 2017-02-21:
Gracias, @JorFru.

closed 2017-02-21 by Siotma

jgvictores, 2017-02-22:
@JorFru Thanks!!!