asrob-uc3m/impresoras-asrob #11: Revisión minuciosa de impresoras

Siotma, 2017-02-08:
En la mañana de hoy realicé lo siguiente:

Pasos realizados con Hija Resurrection.

	1. Introducción del filamento en Hija Resurrection
	2. Recolocación del Endstop Y de Hija Resurrection para que no estorbe al cable de alimentación de la base caliente.
	3. Recolocación de los cables del termistor de Hija Resurrection para que no toquen con la cama durante la impresión.
	3. Prueba de impresión de Hija Resurrection: fallida por atasco.
	4. Observo que el filamento se extruye con dificultad. Subo la temperatura de impresión a 200 ºC.
	4. Advierto que la temperatura del extrusor de Hija Resurrection cae a pesar de tener la consigna de calentar.
	5. Tras varias pruebas, confirmo que el calentador del extrusor de Hija Resurrection está dañado y dependiendo de la posición de los cables funciona o no: ¿Cable seccionado?
	6. Tras asegurar el cable procedo a una nueva prueba de impresión.
	7. Prueba de impresión satisfactoria.
	El diagnóstico de Hija Resurrection es:
		- [ ] 1. Cable del calentador del extrusor dañado: recomiendo desmontarlo y evaluar daños y/o cambiarlo por uno nuevo.
		- [ ] 2. Temperatura de impresión demasiado baja: recomiendo usar 200ºC
		- [ ] 3. También recomiendo fijar el termistor del extrusor con Kapton ya que se mueve mucho en su agujero y eso puede falsear las lecturas de temperatura. Pese a ello estoy seguro de que el cable del Extrusor está dañado (he usado los dedos para asegurarme).

Pasos realizados con Blacky.

	1. Introducción de filamento: fallido: extrusor atascado.
	2. Desatasco del extrusor con el siguiente proceso.
		1. Retirar el ventilador del disipador.
		2. Aumentar la temperatura del extrusor.
		3. Introducción repetida de un alambre fino por la parte superior para eliminar el atasco.
		4. Pruebo a introducir de nuevo el filamento: éxito. Saco el filamento ya que el extrusor está muy caliente.
		5. Recoloco el ventilador y procedo a enfriar el extrusor antes de meter de nuevo el filamento.
	3. Con el extrusor frío introduzco de nuevo el filamento: éxito.
	4. Prueba de impresión utilizando la siguiente configuración:
		a. Mi propio ordenador portátil, ya que el suyo no se puede utilizar porque el monitor no tiene fuente de alimentación.
		b. El cristal de la base de Hija Resurrection porque Blacky no tenía cuando llegué.
		c. El soporte de filamento de Hija Resurrection porque Blacky no tenía cuando llegué.
	5. Éxito de la prueba de impresión.
	El diagnóstico de Hija Resurrection es:
		- [ ] 1. Le falta el cristal de la base.
		- [ ] 2. Le falta un soporte para el filamento.
		- [ ] 3. No se puede utilizar su ordenador al carecer el monitor de fuente de alimentación.
		- [ ] 4. La impresora estaba atascada, la he desatascado.
		- [ ] 5. La prueba de impresión ha demostrado que la base de Blacky y/o el eje X están desnivelados

David-Estevez, 2017-02-08:
Simplemente decir que esta issue es de lo mejorcito en calidad que hemos tenido en ASROB en mucho tiempo. :clap: :clap: :clap: 

Muchas gracias por acercarte a echarnos un cable con las impresoras, @Siotma

Siotma, 2017-02-16:
Esto es lo que hemos hecho hoy con Hija Resurrection:

1. Yo, he donado una bobina de filamento ABS verde a HIJA RESURRECTION. Imprimir a 220ºC extrusor y 100ºC base.
2. Prueba del extrusor de HIJA RESURRECTION con polímetro. No parece estar dañado. Hemos mejorado las conexiones en la placa: cable saneado y estañado.
3. Prueba de impresión de HIJA RESURRECTION: éxito parcial
	Los pasos por mm del eje Z siguen estando mal, se deben cambiar a 8400 pasos por mm, ver #4 
	La base está desnivelada. La he nivelado.
4. Hemos sujetado bien el termistor utilizando Kapton que he llevado yo.
5. Prueba de impresión de HIJA RESURRECTION: éxito.
6. Tras sucesivas pruebas, el error del calentador del extrusor ha vuelto a aparecer, por lo que vamos a pedir uno nuevo. Jorge se encargará de realizar el pedido.
7. Hemos afianzado el ventilador del extrusor.

Y con Blacky:
1. Colocación del filamento ABS amarillo de ASROB a Blacky, y del soporte de madera.
2. Se han realizado pruebas positivas de impresión con Blacky utilizando el ABS amarillo, por lo que lo hemos dejado puesto.
3. Aún no se puede utilizar porque carece de cristal en la base.
4. Hemos conectado la fuente de alimentación (12V, 3.3A) al monitor previa consulta a Rodrigo, quien asegura que es la suya. No he sido capaz de encontrar ninguna referencia al monitor por internet.
5. El PC carece de software de impresión (o al menos no lo hemos encontrado), por tanto hay que hacer caso a #2

jgvictores, 2017-02-16:
Re: Blacky (4):
- Buscar buscar buscar.
- https://www.alibaba.com/product-detail/17-Inch-LCD-Monitor_10801015.html
- http://www.freewtc.com/products/17-inch-lcd-monitor-528134-61531.htm
- Y no pensar en hacer cosas como: http://www.forosdeelectronica.com/f27/monitor-necesita-transformador-12v-puedo-ponerle-uno-19v-45988/

Siotma, 2017-02-21:
Hoy, de 13:00 a 15:00, hemos estado trabajando con Blacky:

1. La base donada por @jgvictores queda preciosa en Blacky. Le hemos quitado el Kapton y limpiado con limpiacristales. Ya le hemos echado bien de laca y funciona a la perfección. Cuidémoslo entre todos para que dure!
2. He donado 3 pinzas para sujetar el cristal a la base, pues solo había puesta en Blacky cuando llegamos.
3. Hemos cambiado de lado el endstop del eje Y, y alargado el cable utilizando un alargador de servos para evitar enganches durante la impresión.

Además, hemos trabajado con su ordenador:

1. Hemos descargado las librerías Mono utilizando [este tuto](http://www.mono-project.com/docs/getting-started/install/linux/#debian-ubuntu-and-derivatives).
2. Luego las hemos instalado `apt install mono-complete`.
3. Tras su instalación, el instalador de Repetier aún detectaba un problema con la versión, por lo que hemos limpiado instalaciones anteriores.
4. Hemos realizado sucesivas pruebas con Repetier y crasheaba siempre. Lo hemos lanzado con la consola y parecía ser un problema con las librerías X (XRead...).
5. Hemos añadido el usuario asrob al grupo dialout con `sudo usermod -a -G dialout asrob`
6. Hemos vuelto a lanzar Repetier desde la consola y no ha habido problema: en su directorio creo que hay que ejecutar `mono RepetierHost.exe`.
7. Comportamiento absolutamente estable del programa y prueba de impresión positiva utilizando Cura por defecto.
8. Durante la impresión hemos descargado Slic3r y lo hemos dejado listo para empezar a calibrar la impresora y a configurar los perfiles de impresión.

Hay que destacar, que no estamos seguros de qué era lo que fallaba, ni si volverá a hacerlo.

Créditos a @Morgloz y a @Honru por su ayuda e ideas durante este proceso.

Siotma, 2017-02-21:
No quito la etiqueta de Blacky porque, aunque ya funciona, aún no debe ser utilizada por externos, ya que no se encuentra configurada. De esta forma no se permitirá realizar reservas.

JorFru, 2017-03-02:
En la reunió de hoy se ha flasheado [este](https://github.com/asrob-uc3m/impresoras-asrob/issues/4#issuecomment-283662186) firmware a Hija Resurrection.
Se han nivelado las camas de ambas, Hija y Blacky.
El ajuste de pasos por mm de Hija se ha resuelto y ya imprime con las dimensiones correctas.
Y se han hecho algunas pruebas de impresión y se añaden las fotos a continuación.
Respecto al software del ordenador de blacky, se sigue trabajando en ello, falta por terminar de configurar Slicer para trabajar en el usuario invitado.
Se ha descubierto un error en Hija que consiste en que salta error de temperatura de la cama por salirse fuera del rango de trabajo (ahora puesto a 100°). Pensamos que es por el uso de un relé para controlar la resistencia, por eso la temperatura cambia de forma abrupta y por eso detecta error.
La calidad de la impresión en mejorable. Y por eso estamos aquí!

[image]
[image]
[image]

JorFru, 2017-03-02:
Las piezas están saliendo bien en cuanto a medidas.
El problema de error en la temperatura creemos que está ya resuelto. Hemos cambiado el tiempo de refresco de la lectura de temperatura de la cama (de 5000 a 1000ms) y hemos activado la función de histéresis para que el control sea mejor.
Damos por cerrada la issue, ya que esto funciona y abriremos otra para añadir mejoras y pijadas a las máquinas.

closed 2017-03-02 by JorFru