asrob-uc3m/impresoras-asrob #12: Reunión de gente interesada en impresoras.

Siotma, 2017-02-08:
Creo que las impresoras nos gustan a todos los miembros de ASROB, pero me temo que por miedo o desconocimiento no disfrutamos de ellas todo lo que se podría.

Propongo una reunión de gente interesada en este asunto, conmigo como instructor y resolvedor de dudas. En caso de tener tirón se podría crear grupo de trabajo activo y creo que sería justificable que tuviéramos pleno acceso a las impresoras para perfeccionarlas y trastear todo lo que queramos... que van súper lentas XD.

Mi disponibilidad es de cualquier día (preferiblemente miércoles) por la mañana. Dentro de dos meses también podré por la tarde pero hasta entonces no...

Contestad con vuestra opinión y/o disponibilidad. En caso necesario crearé un Doodle para acordar una fecha.

Saludos.

Morgloz, 2017-02-08:
Yo puedo por las mañanas también

Siotma, 2017-02-10:
Hola, @Morgloz. Gracias por tu respuesta!

En vista de que la issue no ha tenido mucho tirón he creado un Doodle donde podemos concretar la primera reunión. De esta manera espero que pueda llegar a más gente.

Utiliza el siguiente enlace para especificar tu disponibilidad.

http://doodle.com/poll/nt87w6ppyrppas5m

Siotma, 2017-02-13:
El doodle se ha cerrado ya.

La fecha seleccionada es el próximo jueves día 16 de febrero.

Yo estaré en la nave desde las 9:00 horas. De esta manera, podré dedicar un buen tiempo tanto a Jorge como a Alejandro explicando cosas. De 13:00 a 14:00 horas será la "reunión de grupo" para tratar temas como los intereses, líneas de investigación y disponibilidad de los miembros.

¡Animo a todo el mundo a pasarse por la nave durante la mañana!

Se informará por esta issue de los resultados.

Siotma, 2017-02-16:
Tras la reunión de hoy, hemos llegado a la conclusión de que hay gente en ASROB que está interesada en:

1. Aprender más sobre impresoras.
2. Mejorar las impresoras de la asociación.
3. Llevar un seguimiento de su estado, y ser capaces de realizar un mantenimiento.
4. Documentar todo lo anterior.

En concreto, entre otras cosas hemos hablado de:

1. Documentar y actualizar el firmware de las impresoras.
2. Calibrar los pasos por mm de los ejes de las impresoras.
3. Realizar mejoras estructurales como la colocación de ventiladores

Hemos propuesto una segunda reunión para el próximo martes a las 13:00, aunque voy a dejar un Doodle por aquí: http://doodle.com/poll/2idnnvxa5dbhb95s

Siotma, 2017-02-20:
El doodle se ha cerrado. La reunión tendrá lugar mañana de 13:00 a 15:00 en la nave 1.0.B06.
Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.
Será útil que traigáis vuestros portátiles, en la medida de lo posible.

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí.

Siotma, 2017-02-21:
En la reunión de hoy se ha trabajado con Blacky (ver #11).

Hemos trabajado en la estructura de la impresora y también hemos estado jugando con Linux para instalar las librerías y echar a andar Repetier.

Además, @JorFru ha estado trabajando con el calentador del extrusor de Blacky.

Habrá una reunión la semana que viene. Doodle: http://doodle.com/poll/cgp9r294tnbk23f8

Siotma, 2017-02-27:
El doodle se ha cerrado. La reunión tendrá lugar el jueves día 2 de marzo de 13:00 a 15:00 en la nave 1.0.B06.
Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.
Será útil que traigáis vuestros portátiles, en la medida de lo posible.

Trabajaremos en la calibración de ambas impresoras con el objetivo de dejarlas operativas:

- Aprenderemos a configurar los parámetros de Repetier y Slic3r para obtener buenos resultados en las impresiones.
- Aprenderemos a configurar el firmware de las impresoras (Marlin) y flashearemos la última versión para corregir los pasos por mm de Hija Resurrection.

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí.

Siotma, 2017-03-02:
En la reunión de hoy se ha trabajado con ambas impresoras (ver #11 )
@Morgloz , @jorgercobo , @LuciaHonrubia y @JorFru (menciono para que se suscriban automáticamente al hilo) hemos estado nivelando las impresoras y mejorando la instalación de Repetier en Linux. Además hemos aprendido a configurar el firmware de Hija Resurrection para arreglar el problema de pasos por mm, aunque ha aparecido otro nuevo relacionado con un sistema de seguridad que se activa de manera demasiado estricta ante cambios de temperatura en la base de impresión. Seguimos trabajando en ello.

Para ello he hecho un nuevo **Doodle** para la semana que viene: http://doodle.com/poll/7u782whypyuacmp8

Adicionalmente, y a nivel personal, mañana estaré durante la mañana en la nave haciendo pruebas con las impresoras, aproximadamente de 10 a 13 horas. Todo el que pueda acercarse será bienvenido :)

David-Estevez, 2017-03-02:
Llamadme pesado, pero... http://goo.gl/forms/movrqVPe6k :see_no_evil:

Siotma, 2017-03-05:
El doodle se ha cerrado. La reunión tendrá lugar el jueves día 2 de marzo de 13:00 a 15:00 en la nave 1.0.B06.
Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.
Será útil que traigáis vuestros portátiles, en la medida de lo posible.

Trabajaremos en la calibración de ambas impresoras con el objetivo de mejorar su funcionamiento:

1. Aprenderemos a configurar los parámetros de Repetier y Slic3r para obtener buenos resultados en las impresiones.
2. Debatiremos las posibilidades de formalizar el grupo de trabajo activo #16 

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí.

Siotma, 2017-03-09:
En la reunión de hoy hemos hecho lo siguiente:

- Nos hemos constituido como grupo de trabajo activo, tras valorar si teníamos la motivación y cumplíamos los estatutos (ver #16 ).

- Realizado una prueba de impresión en Hija Resurrection. El resultado ha sido satisfactorio salvo porque la impresión se ha interrumpido debido a un nudo en el filamento. Ya ha sido resuelto, y esperamos que no vuelva a pasar.

- Investigado los motivos por los que a Blacky no se sabía lo que le pasaba ( #6 ). El problema ha resultado ser el offset de Z en los perfiles de slic3r. Slic3r generaba el gcode para que la impresión empezara en la coordenada Z=-4, sin embargo la impresora está configurada para nunca alcanzar una posición de Z menor que 0, lo cual es muy lógico y necesario si tenemos en cuenta que existe un límite físico llamado endstop.
Así pues, cuando el gcode envía la impresora a una posición menor que 0, esta se limita a no moverse. De esta manera, hemos calculado que las primeras 15 capas aproximadamente se imprimirían al mismo nivel, hasta que la coordenada Z del gcode fuera superior a 0 y la impresora comenzara a subir.
Lo desconcertante es que este parámetro ya había sido cambiado en reuniones anteriores, por lo que debemos hacer es descubrir por qué ha vuelto a ponerse mal.
Tras solucionar el problema, se ha realizado una prueba de impresión satisfactoria.
- Además, se ha añadido a la base de impresión de Blacky, una pieza de cartón por debajo, a modo de aislante, de manera que se acelere el proceso de calentamiento de la base antes de una impresión.
- Por último, hemos estado repasando algunos conceptos importantes de los perfiles de Slic3r que seguiremos comentando en la próxima reunión.

En la próxima reunión:

- Hablaremos mucho de Slic3r por lo que recomiendo leer acerca de este programa en el siguiente enlace: http://manual.slic3r.org/ , o ver algún vídeo en youtube como https://www.youtube.com/watch?v=AVf6GUcxiJU.
Incluso podéis descargar el programa en http://slic3r.org/ , instalarlo e investigar en los distintos menús para familiarizaros con su aspecto. La interfaz es bastante autoexplicativa.

- Además, flashearemos la última versión de Marlin en Blacky, dados los problemas encontrados a la hora de obtener sus ajustes.

He creado un nuevo Doodle, al que se puede acceder en dos formatos:

- Tabla: http://doodle.com/poll/kq2rzkfu3ykmm7ys#table
- Calendario: http://doodle.com/poll/kq2rzkfu3ykmm7ys#calendar

Animo a todos los interesados a participar en la votación.

Siotma, 2017-03-13:
El doodle se ha cerrado. La reunión tendrá lugar el martes día 14 de marzo de 13:00 a 15:00 en la nave 1.0.B06. Además, he seleccionado el día de hoy y el jueves en el mismo horario como días adicionales para que otros miembros del grupo de trabajo puedan acercarse a trastear con Slic3r.

Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra. Si no os es posible acceder en los horarios marcados, pedid ayuda por el grupo horizontal para que alguien pueda abriros.

Será útil que traigáis vuestros portátiles, en la medida de lo posible, sobre todo el martes.

- Trabajaremos en la configuración de los parámetros de Repetier y Slic3r para obtener buenos resultados en las impresiones.

- El martes, también, flashearemos un nuevo Marlin en Blacky

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí, y en el [Cuaderno de Bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora).

jgvictores, 2017-03-13:
@Siotma Enhorabuena por la cantidad de curro que le estáis echando, y sobre todo... ¡Muchas Muchas GRACIAS y ENHORABUENA por estar documentando con tanto detalle!

cc: @Morgloz @jorgercobo @LuciaHonrubia @JorFru et al. [at] Printer One.

David-Estevez, 2017-03-13:
Su cuaderno de bitácora tiene muchísima mejor pinta que el nuestro... habrá que aprender de ellos!!

Siotma, 2017-03-14:
En la reunión de hoy, hemos trabajado con ambas impresoras:
En Blacky:
- Hemos flasheado la última versión de Marlin.
- Hemos realizado una prueba de impresión satisfactoria.

En Hija Resurrection:
- Hemos realizado dos pruebas de impresión satisfactorias (ver imagen)
- Hemos colocado el cartón por debajo de la base para reducir la pérdida de energía.
- Hemos retirado la brida y hemos puesto un trozo nuevo de kapton en el extrusor para sujetar el termistor.

[image]

En la próxima reunión veremos lo siguiente:

- Mejorar los finales de carrera.
- Aprender a editar la Wiki.
- Dejar unos perfiles de Slic3r definitivos, haciendo hincapié en la impresión de puentes.

El link del doodle es:
- Tabla: http://doodle.com/poll/kwbv2gz54ewt5vym#table
- Calendario: http://doodle.com/poll/kwbv2gz54ewt5vym#calendar

Todos a votar.

Siotma, 2017-03-19:
El doodle se ha cerrado. La reunión tendrá lugar el viernes día 24 de marzo de 13:00 a 15:00 en la nave 1.0.B06. 

Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.

Será útil que traigáis vuestros portátiles, en la medida de lo posible.

    Trabajaremos en la configuración de los parámetros de Repetier y Slic3r para obtener buenos resultados en las impresiones. También realizaremos trabajos estructurales en cuanto a la ubicación de los finales de carrera, y aprenderemos a editar la Wiki.

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí, y en el [Cuaderno de Bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora).

Siotma, 2017-03-24:
En la reunión de hoy, hemos trabajado con ambas impresoras:
En Blacky:

1. Hoy hemos nivelado el final de carrera de Z.
2. Impresión de pieza de calibración para comprobar que las dimensiones de impresión son correctas y los ángulos entre el eje X y Y están a 90º.
3. Se comprueba que los ejes están a 90º y las dimensiones tienen un error menor del 0.7%. (ver imagen)

[image]

En Hija de Resurreción:

1. Se ha visto que el Kapton recientemente adquirido se ha quemado a temperaturas a las que esto no debería ocurrir (220 grados). (ver imagen)
[image]

2. Impresión de pieza de calibración para comprobar que las dimensiones de impresion son correctas y los ángulos entre el eje X y Y están a 90º.
3. Se comprueba que los ejes están a 90º y las dimensiones tienen un error menor del 0.6%. (ver imagen)

[image]

Además, hemos buscado ideas en Thingiverse para mejorar los finales de carrera. Próximamente se intentarán implementar estas mejoras.

Siotma, 2017-03-27:
He creado un Doodle para la presente semana.
Como ya comenté, esta semana me será imposible acudir a la reunión, aun así ya acordamos que yo no soy tan importante y que podéis tirar sin mi  :stuck_out_tongue_winking_eye:

He propuesto dos objetivos para que sigamos avanzando, aunque al final la última palabra la tendréis los que vayáis:

- Mejorar los textos de la Wiki y adaptarla al formato de otros grupos.
- Realizar nuevas pruebas de impresión y mejorar los perfiles de Slic3r.

El link del nuevo Doodle es: http://doodle.com/poll/72dbbxw5ssk2wzgq

Siotma, 2017-03-30:
El doodle se ha cerrado. La reunión tendrá lugar mañana viernes día 31 de marzo de 13:00 a 16:00 en la nave 1.0.B06. 

Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.

Será útil que traigáis vuestros portátiles, en la medida de lo posible.

Trabajaremos en la configuración de los parámetros de Repetier y Slic3r para obtener buenos resultados en las impresiones. También se trabajará en mejorar la información proporcionada por la página de la Wiki del grupo de trabajo: http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí, y en el [Cuaderno de Bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora).

jorgercobo, 2017-03-31:
En la reunión de hoy, hemos trabajado en:

- En la wiki:
 Hemos puesto a Printer One y a G.A.M.E en portada como grupos activos
- En Hija de Resurreción:
1. Prueba de impresión con soportes generados por Slicer fallida. 
    Cambio de configuración en Slicer: Interface layers: de 3 a 0.
2. Segunda prueba de impresión con soportes generados por Slicer. Soportes impresos correctamente, aunque debido a un fallo ha habido que pararla.
    **Problema**: Solo la última capa del soporte se hace maciza. Al imprimirse la primera capa apoyada, esta capa de soporte se suelta.
    **Ñapa/solución**: poner un punto de pegamento termofusible para que se imprima sobre este.
    **Solución fetén**: que las x últimas capas de soporte sean macizas para que no se suelten. 
    **PENDIENTE DE IMPLEMENTACIÓN**

[image]
[image]

David-Estevez, 2017-03-31:
> Ñapa/solución: poner un punto de pegamento termofusible para que se imprima sobre este - @JorFru

Como habréis descubierto empíricamente, esa solución no funciona muy bien. Dado que estáis imprimiendo en ABS, hay un par de soluciones más eficaces y extendidas entre la comunidad:
1. Usar la famosa "pasta de acetona". Disolver un trozo de ABS en acetona (lo más pura posible) y aplicar, o bien a la base antes de imprimir, o bien a la pieza en la zona que se está despegando. Al secarse se funde con el ABS y se pega que da gusto.
2. Configurar el slicer para hacer la pieza con "Raft" o "Brim". "Raft" es una base extensa de ABS que cubre todo el inferior de la pieza y ayuda a que se pegue cuando la pieza tiene poca superficie de contacto (como vuestro pinguino de la imagen). "Brim" son perímetros extra en la primera capa, que aumentan la superficie de contacto y evitan el warping, ya que lo que warpea es el "Brim" y no la pieza.

Siotma, 2017-04-01:
Mis experiencias con ABS confirman que el Brim es una muy buena opción para mejorar la adhesión a la base de impresión.

He creado un Doodle para semana que viene.
Esta semana mi disponibilidad cambia y tengo algunas tardes libres, lo que hará que podamos realizar reuniones más extensas.

Como objetivos podemos tener los siguientes:

- Mejorar los textos de la Wiki y adaptarla al formato de otros grupos.
- Realizar nuevas pruebas de impresión y mejorar los perfiles de Slic3r. Probar la opción de Brim.
- Completar la instalación de Repetier en Linux y su integración con la sesión de invitado.

El link del nuevo Doodle es: http://doodle.com/poll/mhnrae4u2cgdpg42

Siotma, 2017-04-01:
Asimismo, puntualizo que al haber sustituido el enlace de la portada, la antigua página de impresoras se ha quedado huérfana (http://asrob.uc3m.es/index.php/Impresora-3D_Open_Source).

Además de utilizar la información que contiene para crear la nuestra, habría que enlazarla desde la sección antiguos proyectos...

Siotma, 2017-04-04:
El doodle se ha cerrado. La reunión tendrá lugar hoy martes 4 de abril de 13:00 a 19:00 en la nave 1.0.B06.

Puntualizar que no es necesario venir durante todo el horario, que cada uno se acerque según vaya pudiendo. Yo, por ejemplo, iré a las 14:00.

Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.

Será útil que traigáis vuestros portátiles, en la medida de lo posible.

Trabajaremos en la configuración de los parámetros de Repetier y Slic3r para obtener buenos resultados en las impresiones. También se trabajará en mejorar la información proporcionada por la página de la Wiki del grupo de trabajo: http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story
Además, intentaremos completar la instalación de Repetier en Linux y su integración con la sesión de invitado 

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí, y en el [Cuaderno de Bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora).

jorgercobo, 2017-04-05:
En la reunión de hoy (14:00 a 21:00 horas), hemos trabajado en:
- En Blacky:
1. Se ha avanzado en la configuración de su PC. #2 
2. Se ha intentado imprimir un Hairy Lion: el tiempo de fileteado tardaba demasiado, así que hemos optado por no seguir. Posible tema de reunión: Modelos 3D y tiempo de fileteado.
3. Prueba de impresión satisfactoria: tornillo y tuerca moleteados.
[image]

- En Hija Resurrection: Mejora en el endstop del eje Z
1. Comprobamos que la escala de las piezas era la correcta imprimiendo las primeras capas de cada pieza y tomando medidas.
2. Hemos impreso las piezas para el montaje.
[image]

3. Se han desatornillado de los puntos necesarios: el motor izquierdo del eje Z y 2 puntos del motor del eje Y. Una de las uniones del primero estaba desgastada y sujeta por una vara de metrica tres. Se ha repuesto la tapa superior de este motor con un recambio de Andres, que muy amablemente nos ha cedido esa parte y un tornillo que era necesario. Tambien se ha soltado el primer motor para trabajar con él.
4. Hemos hecho que los tornillos necesarios para la regulación se ajustasen a la pieza: se escoje un tornillo de metrica superior al agujero y se calienta con el soldador y se atornilla --force. Se ha realizado tanto para la zapata como para el soporte superior.
5. Hemos colocado y atornillado las piezas y el motor.
[image]

6. Hemos recalibrado manualmente el eje Z y hemos impreso dos cubos de calibrado con resultado satisfactorio.

- Tambien hemos ayudado en la limpieza del armario de la asociación.

jorgercobo, 2017-04-17:
El doodle se ha cerrado. La reunión tendrá lugar el miercoles 19 de abril a partir de las 16:00 en la nave 1.0.B06.

Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.

Será útil que traigáis vuestros portátiles, en la medida de lo posible.

Trabajaremos en la configuración de los parámetros de Repetier y Slic3r para obtener buenos resultados en las impresiones. También se trabajará en ajustar el movimiento y los finales de carrera de la cama para conseguir la máxima área de impresión e indicarle a la máquina lo que realmente mide el espejo para que no pueda imprimir fuera. Además, intentaremos completar la instalación de Repetier en Linux y su integración con la sesión de invitado

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí, y en el [Cuaderno de Bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora).

Siotma, 2017-04-24:
En la última reunión (miércoles 19 de 17:00 a 21:00) trabajamos en lo siguiente:

En Blacky:

- Se ha querido imprimir una pierna de Zowi: el resultado ha sido erróneo dos veces:
1. La primera vez el ordenador cerró sesión por tiempo de inactividad y al recuperarla el proceso Repetier crasheó. Se han deshabilitado todos los temporizadores de inactividad en este PC para que esto no vuelva a ocurrir. Antes de fallar, además, esta impresión se estaba deformando por no usar Brim.
2. La segunda vez, la impresora perdió pasos (ver siguiente foto).

[image]

En Hija Resurrection:

- Se imprimió un Hairy Lion, con resultado satisfactorio salvo por una pérdida de pasos. en la mitad de la impresión (ver siguientes fotos)

[image]
[image]

Ambas pérdidas de pasos pueden relacionarse en mi opinión con una nula o insuficiente ventilación de la electrónica de la impresora (Pololus muy calientes pierden precisión).

He creado un Doodle para la presente semana.

Como objetivos podemos tener los siguientes:

- Poner y fijar ventiladores a la electrónica de ambas impresoras.
- Modificar los perfiles de Slic3r para que todos tengan Brim.
- Ajustar el movimiento y los finales de carrera de la cama para conseguir la máxima área de impresión. Indicarle a la máquina lo que realmente mide el espejo para que no pueda imprimir fuera.
- Investigar el error de thermal runaway y en caso necesario desactivarlo. 

El link del nuevo Doodle es: http://doodle.com/poll/m9372rahkarexukr#table

Siotma, 2017-04-26:
El doodle se ha cerrado. La reunión tendrá lugar el viernes 28 de abril de 15:00 a 17:00 en la nave 1.0.B06.

Puntualizar que no es necesario venir durante todo el horario, que cada uno se acerque según vaya pudiendo. Si alguien va a llegar entre las 13:00 y las 15:00 que avise para que se le abra. Yo calculo que me quedaré hasta las 19.

Recuerdo que en caso de que lleguéis y esté cerrado, os asoméis por la barandilla del primer piso para comprobar si estamos dentro y avisarnos para que se os abra.

Será útil que traigáis vuestros portátiles, en la medida de lo posible.

Como objetivos podemos tener los siguientes:
- Poner y fijar ventiladores a la electrónica de ambas impresoras.
- Ajustar el movimiento y los finales de carrera de la cama para conseguir la máxima área de impresión. Indicarle a la máquina lo que realmente mide el espejo para que no pueda imprimir fuera.
- Investigar el error de thermal runaway y en caso necesario desactivarlo. 

Como siempre, todo el que quiera venir está invitado!

Los resultados se postearán aquí, y en el [Cuaderno de Bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora).

Siotma, 2017-04-28:
En la reunión de hoy (de 15:00 a 19:00) trabajamos en lo siguiente:

En Blacky:

- Se han querido imprimir soportes para ventilador de electrónica:
1. Se ha impreso [este](https://www.thingiverse.com/thing:770533) modelo para usarlo con el ventilador de 80mm de Hija Resurrection, pero es un diseño para 60mm por lo que no servirá. 
2. Luego se ha impreso [este otro](https://www.thingiverse.com/thing:704418/#files) que sí es para 80mm y tiene buena pinta. Pendiente ponerlo en la siguiente reunión.

En Hija Resurrection:

- Se ha modificado el Marlin:
1. Se ha aumentado la tolerancia de la detección de Thermal Runaway a 10 grados durante 60 segundos, por lo que ya no debería volver a saltar fuera de contexto.
2. Se han ajustado las dimensiones de impresión, perfeccionando también la posición de los finales de carrera. Actualmente puede imprimir en XY = 200x180mm. Se podría aumentar a 200x200 pero esto implicaría cambiar el carro del extrusor.

- Se han realizado dos pruebas de impresión:
1. Una cadeneta portacables. El resultado ha sido impresionante, si no fuera porque el ABS utilizado (bobina negra puesta recientemente en #13 ) parece plastilina. Las paredes finas se han deshecho y la pieza se ha roto por completo.
2. Un carro de eje X para @andrespy . Próximamente se actualizará el post con el resultado.

Asrobot ha creado un Doodle para la semana que viene: http://doodle.com/poll/svq5wypfp7bp2pb6#table

Como objetivos podemos tener los siguientes:

1. Terminar de imprimir las piezas de sujección y montar los ventiladores en la electrónica. Se proponen estas piezas:

        Blacky: Dos ventiladores de 40mm usando: http://www.thingiverse.com/thing:1052196

        Hija Resurrection: Un ventilador de 80mm que ya está conectado: https://www.thingiverse.com/thing:704418/#files

2. Replicar en Blacky lo hecho en Hija Resurrection para el Thermal Runaway y las dimensiones de impresión.

El link del nuevo Doodle es: http://doodle.com/poll/svq5wypfp7bp2pb6#table

Siotma, 2017-05-03:
En la reunión de hoy (de 17:00 a 19:00) trabajamos en lo siguiente:

En Blacky:

- Se han terminado de imprimir los soportes para ventilador de electrónica:

1. Se ha impreso [este](http://www.thingiverse.com/thing:1052196) que soporta dos ventiladores de 40mm y tiene buena pinta. Pendiente de colocar de momento.

- Se ha modificado el Marlin:
1. Se ha aumentado la tolerancia de la detección de Thermal Runaway a 10 grados durante 60 segundos, por lo que ya no debería volver a saltar fuera de contexto.
2. Se han ajustado las dimensiones de impresión, perfeccionando también la posición de los finales de carrera. Actualmente puede imprimir en XY = 188x158mm. Se podría aumentar a 200x200 pero esto implicaría cambiar el carro del extrusor y redistribuir la estructura de la impresora.

En Hija Resurrection:

- Se ha colocado el ventilador de la electrónica con las [piezas impresas el último día en versión 80x80mm](https://www.thingiverse.com/thing:704418/#files):

[image]

- Se ha aumentado la temperatura de impresión del ABS negro a 240º en vez de 220.
- Se ha realizado una prueba de impresión (pendiente de ver resultado) de una pieza diseñada por @andrespy para hacer concéntrico el rollo de filamento con el eje de sujección y facilitar el giro:

[image]

Siotma, 2017-05-04:
Asrobot ha creado un Doodle para la semana que viene: http://doodle.com/poll/vvu9witzzwbe4mre#table

Como objetivos podemos tener los siguientes:

1. Buscar el mejor lugar para soldar, soldar y colocar los ventiladores de la electrónica de Blacky.

2. Imprimir un final de carrera de Z  como [este ](http://www.thingiverse.com/thing:1776429) para Blacky. Igual que el que le pusimos a Hija Resurrection. Y colocarlo y calibrarlo.

3. Imprimir alguna pieza de Zowi: https://github.com/asrob-uc3m/zowi/issues/3

El link del nuevo Doodle es: http://doodle.com/poll/vvu9witzzwbe4mre#table

jgvictores, 2017-05-08:
¿Sería mucho pedir que entre los objetivos se fuese sumando el mejorar/actualizar los manuales? :smile:
- https://www.gitbook.com/book/asrob-uc3m/impresoras-user-manual-i3
- https://www.gitbook.com/book/asrob-uc3m/impresoras-maintenance-manual-i3

jorgercobo, 2017-05-09:
Como se nos van acumulando objetivos poco a poco, he creado un proyecto en este repositorio en el que organizar los objetivos: [Printer One: A Clone Wars Story](https://github.com/asrob-uc3m/impresoras-asrob/projects/1)

David-Estevez, 2017-05-09:
Mola!! Pro tip: puedes importar también directamente issues que ya estén creadas al tablón, y te muestra el estado (abierta o cerrada) de la issue, y te pone el link automáticamente :wink:

jorgercobo, 2017-05-09:
Muchas gracias, queda mucho mejor así

Siotma, 2017-05-09:
Me encanta :')

Es justo lo que nos hacía falta!

Se podría programar a ASROBOT para que haga los Doodles y ponga el link al proyecto en la descripción del Doodle, por ejemplo:

> Elegir fecha para la próxima reunión de Printer One
> Consulta y propón objetivos para la reunión en: https://github.com/asrob-uc3m/impresoras-asrob/projects/1

Sería una pasada

David-Estevez, 2017-05-09:
No lo he entendido bien... queréis que ponga el link del doodle en el proyecto? en una tarjeta nueva?

Siotma, 2017-05-09:
No, justo lo contrario.

Ahora mismo, ASROBot es capaz de hacer Doodles como [este](http://doodle.com/poll/vvu9witzzwbe4mre#table)

Imagino que ASROBot es quien decide la descripción del Doodle. Actualmente dicha descripción es:
> Elegir fecha para la próxima reunión de Printer One

Y yo sugiero, que aumentemos esa descripción con otra línea, quedando así:
> Elegir fecha para la próxima reunión de Printer One
> Consulta y propón objetivos para la reunión en: https://github.com/asrob-uc3m/impresoras-asrob/projects/1

David-Estevez, 2017-05-09:
Ahora mismo el comando que se lanza para crear un doodle tiene muy pocos argumentos, entre ellos el nombre del grupo. Digamos que es una plantilla que pone "Elegir fecha para la próxima reunión de X".

Pero si os hace muuucha muuucha ilusión se podría modificar para que, además, se pudiese añadir una descripción personalizada.

Siotma, 2017-05-09:
Ah, no, no es tan importante. Lo proponía solo si era cuestión de añadir el texto...

Siotma, 2017-05-10:
En la reunión de hoy (de 17:00 a 19:00) trabajamos en lo siguiente:

En Blacky:

- Se ha colocado un nuevo soporte para el endstop Z, pero se han encontrado numerosas complicaciones por lo que aún no está operativo, y la impresora está fuera de servicio ( #22 )

En Hija Resurrection:

- Se ha probado la pieza diseñada por @andrespy para mejorar el giro del rollo de filamento, y se han detectado posibles mejoras para su diseño.

Además, se ha ordenado la caja de tornillos M3 que estaba desastrosa.

Siotma, 2017-05-18:
En la reunión de hoy (de 18:00 a 19:30) trabajamos en lo siguiente:

En Blacky:

- Se ha terminado de colocar el soporte de Z, y se ha calibrado y nivelado la cama.
- Se ha ajustado mejor el anclaje del extrusor con la pieza del carro de X para evitar que oscile. Estas oscilaciones eran la causa de movimientos extraños durante la impresión, que empeoraban preocupantemente el acabado de las piezas. 
 Para el ajuste se ha extraído el extrusor de su alojamiento retirando los dos tornillos Allen, y se ha introducido un suplemento de cinta de doble cara sobre el extrusor de manera que al volver a colocarlo, éste hace mayor presión contra la pieza que lo aloja y queda fijo.

En Hija Resurrection:

- Se ha perfeccionado también el mecanismo de ajuste del eje Z, y se ha nivelado.

@pamarinp se ha quedado tras nuestra reunión haciendo pruebas de impresión. Esperamos fotos con los resultados.

pamarinp, 2017-05-18:
[image]

La impresión de las 4 ruedas ha quedado finísima. Enhorabuena por esa calibración.

Con la otra impresora ha dado un problema de thermal runaway con cuatro ruedas pero con una si ha funcionado.
[image]

Siotma, 2017-05-25:
En la reunión de hoy (de 11:00 a 13:00) trabajamos en lo siguiente:

En Blacky:

- Se han valorado las dificultades que conllevaría instalar los ventiladores de la electrónica (https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-298969157). Necesitamos localizar dos conectores en los que podamos crimpar los cables que actualmente provienen de la fuente de alimentación del ventilador del extrusor, y añadir los nuevos cables para los ventiladores de la electrónica. Una vez crimpados, se podrán meter cómodamente en la clema de conexión de la RAMPS. ( #24 )

- Se ha colocado un soporte de motor Z ( https://github.com/asrob-uc3m/impresoras-asrob/issues/21#issuecomment-303991030 ). Adicionalmente se ha colocado en el otro motor una impresión a medias de otro soporte. Queda pendiente imprimir otro completo.

En Hija Resurrection:

- Se ha cambiado el Kapton que había alrededor del extrusor ( https://github.com/asrob-uc3m/impresoras-asrob/issues/23#issuecomment-303990201 ) ya que el que había anteriormente se había chamuscado por completo.

[image]

- Se ha impreso un soporte de motor Z para blacky.

[image]

Además, hemos trabajado con la impresora de @andrespy , para perfeccionar el funcionamiento del eje X. Actualmente está pendiente de imprimir una pieza, y esperamos verla funcionando pronto.

Siotma, 2017-06-07:
En la reunión de hoy (de 11:00 a 13:00) trabajamos en lo siguiente:

En Blacky:

- Se ha colocado el segundo soporte de motor Z y se ha cerrado la issue correspondiente ( https://github.com/asrob-uc3m/impresoras-asrob/issues/21 ). El resultado puede encontrarse en la siguiente imagen:

[image]

- Además se ha proseguido con la configuración de la sesión de usuario ( #2 ), llegando por fin a una válida (documentación is coming en #2 ).

En Hija Resurrection:

- Se ha impreso la base de Blacky antes mencionada.

Además, hemos trabajado con la impresora de @andrespy .

pamarinp, 2017-06-07:
Una impresión muy buena y rápida de blacky. Resultado muy correcto. Esta vez no se ha movido el espejo de la base. He imprimido con dos capas de raft y luego una de las capas no se despega de las piezas así que no se si es buena idea. He sido el primero en imprimir con la cuenta de invitado. Cuidado con que os salte el login de nuevo, no entréis otra vez a la sesión que se para la impresión. Sería buena idea evitar que saltara el bloqueo de pantalla o el auto log out. Aqui están los resultados.
[image]

Siotma, 2017-09-16:
Dada la buena acogida del grupo de trabajo Printer One: A Clone Wars Story, y para evitar seguir duplicando la información ya existente en [el cuaderno de bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora), se da por cerrada esta issue cuyo único objetivo era la promoción de reuniones de impresoras y la captación de gente interesada.

closed 2017-09-16 by Siotma