asrob-uc3m/impresoras-asrob #13: Faltan filamentos

Siotma, 2017-02-08:
Esta Issue puede valer para informar de cuándo falta filamento en alguna impresora.

Siotma, 2017-02-08:
En este momento, no hay filamento ni en Blacky ni en Hija Resurrection.

David-Estevez, 2017-02-08:
Hay ABS amarillo, pero va a ser un infierno conseguir imprimir con eso, creo yo. Se debería comprar algo de PLA, que es muchísimo más sencillo de usar para newbies.

Siotma, 2017-02-10:
Estoy totalmente de acuerdo con David, mi experiencia con PLA supera de lejos al ABS.

En mi impresora propia utilizo PLA de bq, que creo tiene una excelente relación calidad precio, y mucha variedad de colores. Además lo imprimo a 160ºC, lo que reduce el consumo energético y el esfuerzo de la electrónica de potencia para alimentar el extrusor.

Propongo que se realice un pedido a bq con cargo a la asociación de una bobina para cada impresora.

Siotma, 2017-02-16:
A este respecto, debo decir que esta mañana hemos dotado de filamento a las dos impresoras:

- Blacky está utilizando el rollo de ABS amarillo de ASROB, con el soporte de madera, queda pendiente hacer pruebas para calibrar las temperaturas óptimas de impresión.
- HIJA RESURRECTION está utilizando un rollo de ABS verde que he donado a la asociación. Imprimir a 220ºC extrusor y 100ºC base.

Tenemos la intención de utilizar estos filamentos al menos de forma provisional hasta que ambas impresoras funcionen correctamente. Una vez logrado, procederemos a realizar un pedido de PLA previa consulta para que se autorice la compra.

Siotma, 2017-02-16:
Cierro hasta que vuelva a faltar filamento.

Siotma, 2017-04-22:
El filamento que doné a Hija Resurrection está en sus últimos metros. Cómo podemos conseguir más?

David-Estevez, 2017-04-22:
Asigno a Juan para que nos diga cómo proceder.

Por cierto, cuidado con las etiquetas de las issues, porque con un "NO FUNCIONA" se desactivan las impresoras en la web :wink:

jgvictores, 2017-04-22:
1. Entiendo que hablamos de rollo de ABS verde (mencionado para HIJA RESURRECTION arriba).
1. Serviría el de [la foto](https://cloud.githubusercontent.com/assets/4448588/24302566/64402598-10b3-11e7-8e00-5488756174ad.jpg) de https://github.com/asrob-uc3m/actas/issues/69?

Siotma, 2017-04-23:
¡¡Claro que serviría!!

Había olvidado que existía. El primero que acabe con el rollo actual que ponga ese.

Gracias Juan, qué velocidad para dar soluciones!

jgvictores, 2017-04-24:
Cerrar issue? 😄

JorFru, 2017-04-24:
Me nombro responsable de gastar hasta el último centímetro!
Y también de reponerlo. Esta tarde cae!

JorFru, 2017-04-24:
Prueba de que el filamento antiguo se ha gastado
[image]
Prueba de que el filamento nuevo se ha instalado.
[image]

closed 2017-04-24 by JorFru