asrob-uc3m/impresoras-asrob #15: Uso externo de las impresoras

kleiren, 2017-02-17:
Buenas,
un compañero del departamento de materiales me pregunta si podría utilizar las impresoras de la asociación. Necesita una pieza para su tesis y la quiere imprimir en polipropileno, se traería el filamento él.
¿Sería posible?

Siotma, 2017-02-17:
Ahora mismo ninguna impresora es funcional, estamos intentando ponerlas en marcha pero estaban muy dejadas.
Si es algo urgente me temo que va a tener que buscar otra impresora, en caso de que pueda esperar, es posible que en una semana o 2 se pueda usar una de ellas.

Distinto es el tema de las condiciones de uso de nuestras impresoras que no estoy seguro de cuáles son.

jgvictores, 2017-02-17:
Cierro, migrando issue a actas para debate general: https://github.com/asrob-uc3m/actas/issues/61

closed 2017-02-17 by jgvictores