asrob-uc3m/impresoras-asrob #16: Formalización de grupo de trabajo activo

jgvictores, 2017-02-24:
Reuniones periódicas, documentación... ¡este grupo lo está haciendo muy bien!

Propongo que:
- [ ] Se rellene [el formulario correspondiente](http://goo.gl/forms/movrqVPe6k).
- [ ] Si el grupo cumple [los criterios de los Estatutos](https://asrob-uc3m.gitbooks.io/estatutos/content/definiciones.html#grupo-de-trabajo-activo), suba a tal categoría en [la wiki](http://asrob.uc3m.es/index.php/Main_Page).

Siotma, 2017-02-24:
Hola Juan!
Te agradezco la propuesta, actualmente es algo que valoramos. Lo comentaremos en la siguiente reunión.

David-Estevez, 2017-02-28:
Yo también os animo a que os constituyáis como grupo activo, tenéis mucho potencial y estáis haciendo las cosas realmente bien, realmente os merecéis todas las ventajas de ser un grupo activo.

De todas formas, y aunque no seáis grupo activo todavía, os he incluído en la portada de la wiki de ASROB junto a G.A.M.E para que tengáis la visibilidad que os merecéis: http://asrob.uc3m.es/index.php/Main_Page

(He puesto la fecha de la próxima reunión y la página antigua del grupo de impresoras, si no os mola podéis actualizarla :smile: )

Siotma, 2017-03-02:
Con retraso, te agradezco mucho el detalle David. De momento estamos documentando con Issues porque es más rápido y no tenemos mucho tiempo. Por ese mismo motivo, hoy no hemos podido comentar este tema. Próximamente tendréis más noticias nuestras.

Luciahonrubia, 2017-03-09:
En la reunión de hoy, hemos comentado la idea de formalizar un grupo de trabajo. Todos hemos estado de acuerdo; el responsable de impresoras será Jorge y el del grupo será Manuel. 
Se espera buena cooperación entre nosotros y progreso en el grupo.

closed 2017-03-09 by Luciahonrubia

JorFru, 2017-03-09:
No estuve en la reunión pero apoyo la idea! MAGA

David-Estevez, 2017-03-09:
Genial!! :clap: :clap:  para finalizar el proceso, seguid los pasos que se explican en #68