asrob-uc3m/impresoras-asrob #17: El bote de laca ha desaparecido

Siotma, 2017-03-03:
Esto es un serio problema, ya que es de los pocos recursos de la asociación que se dejan fuera del armario...

Hasta hoy no recuerdo que hubiera pasado nunca, pero ayer ya no estaba, y no lo encuentro en ningún armario.

David-Estevez, 2017-03-03:
El bote se lo dejé yo el otro día a los del departamento, que habían comprado laca que no era Nelly y la suya no pegaba nada... Pero se nos olvidó bajarla después de usarla :sweaty_smile:

Estoy gestionando si devolución.

Siotma, 2017-03-03:
Sin problema, David, lo importante es saber dónde está. Mi preocupación era que alguien la hubiera mangado pero ya veo que no :)

La laca ha sido devuelta, la he dejado junto a las impresoras.

Issue cerrada en tiempo récord.

closed 2017-03-03 by Siotma