asrob-uc3m/impresoras-asrob #18: Logo para wiki y otras cosas

Rodriwp, 2017-03-16:
He estado jugueteando con gimp, y tengo dos propuesta de logo para la wiki ? Que os parecen ?

[image]

[image]

Hacemos un doodle para votar cual poner en la wiki de asrob? 1, 2 o ninguno ?

David-Estevez, 2017-03-16:
:open_mouth:  Molan!! 

El segundo se lee un poco mal, ¿no? Yo probaría a cambiar el color de las letras de "printer one" o difuminar un poco el fondo a ver si se lee mejor :smile:

Siotma, 2017-03-16:
Creo que la primera está genial por ahora.

Contemplaría la posibilidad de hacer algo del estilo de la segunda pero creo que para eso deberíamos desanclar las impresoras y ponerlas con un fondo más uniforme. Se me ocurre imprimir alguna nave o soldado y ponerla también en forma decorativa.

David-Estevez, 2017-03-16:
Recordad subir los archivos fuente / vectorizados del logo molón a [recursos digitales](https://github.com/asrob-uc3m/recursos-digitales) para que se puedan usar en carteles, stickers y memorias cuando se necesite.

Siotma, 2017-04-01:
He añadido el logo a la página de la wiki.

http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story

Siotma, 2017-04-15:
Me gusta mucho la idea del segundo logo, que también contenga las fotos de las impresoras, pero habría que hacerlo con un fondo uniforme que no despiste tanto.

Siotma, 2018-04-21:
Creo que el primer logo es perfecto tal y como está. Por el tiempo que lleva en uso, yo lo consideraría definitivo y en todo caso si se quiere cambiar abriría una nueva issue.

Lo he subido a https://github.com/asrob-uc3m/recursos-digitales/tree/develop/logos , pero creo que la he liado, porque lo de "develop" no me suena muy bien :sweat_smile: 

@David-Estevez @jgvictores ayuda en esto último por favor! En el momento en que quede resuelto se podrá cerrar la issue.

David-Estevez, 2018-04-21:
Arreglado, siéntete libre de cerrar la issue :smile:

closed 2018-04-21 by Siotma

jgvictores, 2018-04-22:
> lo de "develop" no me suena muy bien

@Siotma Lo has hecho genial. Siguiendo [estas guías](https://github.com/roboticslab-uc3m/developer-manual/blob/22d4b931270712b12196f81bd32ec4233ce7c6ed/version-control.md) solemos seguir la metodología "gitflow" con un `develop` que luego va pasando a `master`.

En la misma guía, decimos que para respositorios que no son de código pueden ir con sólo `master`. Aprovechando que @David-Estevez hizo el merge de `develop` a `master`, hice el paso. En este repo, nos quedamos con `master` exclusivamente (aunque por supuesto, se pueden abrir branches para desarrollos específicos).

Thanks!