asrob-uc3m/impresoras-asrob #19: Sensor capacitivo

David-Estevez, 2017-03-25:
Ayer @Siotma me comentó que estábais interesados en ponerle un sensor capacitivo como el de la M-Prime (y muchas otras) a las impresoras para implementar un auto-nivelado.

He estado hablando con Diego Trapero, que fue el que me vendió la impresora, y que ha dicho que ya no le quedan en stock. Pero me ha dado el número de referencia del sensor: LJ12A3-4-Z/BY

Siotma, 2017-03-27:
Por lo que he leído, este modelo es de tipo inductivo:
- https://www.amazon.com/LJ12A3-4-Z-DC6-36V-Inductive-Proximity-Detection/dp/B01LWOS1VA
- http://www.marchantdice.com/worldofcnc/pdf/homeswitch_wiring.pdf
- http://www.ebay.es/itm/LJ12A3-4-Z-BY-Inductive-Proximity-Sensor-Detection-Switch-PNP-DC6-36V-/290671992278

Esto no tendría que ser un problema en sí mismo, ya que de dar buen resultado se puede comprar. He mirado capacitivos y creo que deben ser casi hermanos, aunque estos son más caros:
- http://www.ebay.es/itm/LJC18A3-H-Z-BX-1-10mm-Capacitance-Proximity-Sensor-Switch-NPN-NO-DC-6-36V-300mA-/142166146397?hash=item2119c33d5d:g:Cb4AAOSwB09YF0U6

Si finalmente nos decidimos por alguno, en 3DEspana tienen ambos tipos. Es una tienda que está aquí al lado y siempre da muy buena atención: www.3despana.com/74-sensores-autonivel

Tienen el Inductivo a 4€ y el capacitivo a 15€

David-Estevez, 2017-03-27:
Eso explicaría por qué la cama de la M-Prime es metálica... :sweat_smile: 

Creo que con la configuración actual que tenemos el inductivo no funcionaría, pero la verdad es que me pierdo un poco con estos sensores.

Siotma, 2017-03-27:
Jajajajaja claro, con una cama metálica seguro que funciona a las mil maravillas. No obstante, el metal que llevan los espejos o las pistas de la propia cama caliente es posible que pudieran ser detectadas.

Yo creo que lo más sensato es hacer pruebas con algún sensor que pudiéramos pillar prestado y en función de los resultados programar la compra del que mejor nos venga.

Siotma, 2017-04-05:
De momento estamos mejorando los finales de carrera mecánicos que tenemos ( ver [este comentario](https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-291723846), con muy buenos resultados.
A futuro, cuando otras mejoras más urgentes estén completas, se acometerá la investigación de este tipo de sensores.

Siotma, 2017-04-15:
Dejo este enlace por aquí ya que parece una buena opción. Al ser mecánico nos olvidamos de los problemas derivados del material de la cama. 

**http://www.3despana.com/sensores-autonivel/494-sensor-autonivel-3dtouch-bltouch-servo-switch.html**

David-Estevez, 2017-04-15:
Hombre, si queréis algo mecánico y más artesanal, hay servos de 9g en el armario, y finales de carrera. Y thingiverse tiene diseños de autolevel para ellos: 
    - http://www.thingiverse.com/thing:330162
    - http://www.thingiverse.com/thing:1703665
Lo digo sobretodo por el valor didáctico de diseñar/montar tu propio sensor :wink:

daavm, 2018-03-08:
Estaría bien retomar esto como tarea para la próxima reunión. Si tenemos finales de carrera y 9g en el armario... Me lo pido!

Siotma, 2018-03-10:
Hay muchos servos pequeños en el armario, y finales de carrera seguro que también!

Todo tuyo :wink:

Siotma, 2018-04-29:
He encontrado dos modelos de sensor inductivo de un proveedor de confianza de eBay:

- [Uno](https://www.ebay.es/itm/M12-Sensor-Proximidad-Inductivo-PNP-DC-6-36V-LJ12A3-4-Z-BY-3D-Impresora-S0036-/201916693496)
- [Dos](https://www.ebay.es/itm/M18-Sensor-Proximidad-Inductivo-8mm-NPN-NO-DC-6-36V-LJ18A3-8-Z-BX-Arduino-S0042-/201993387311)

dmarcosm, 2018-05-06:
Estoy pensando que estos sensores funcionan por... detección del metal? distancia? Lo digo sobre todo por la tronxy, que la base es de acrílico

Siotma, 2018-05-06:
Los sensores inductivos como esos son sensibles a la presencia de otros metales.

La base de la tronXY es de plástico, pero debajo la cama tiene pistas de metal, y además creo recordar que estaba recubierta de una superficie lisa de aluminio. Yo apuesto por que sería perfectamente compatible.

David-Estevez, 2018-05-06:
También existen sensores capacitivos, que no requieren un material metálico para funcionar. Ejemplo: http://store.thingibox.com/es/electricidad/224-sensor_capacitivo.html

Siotma, 2018-05-11:
Recibí el paquete el miércoles, y el jueves llevé [el sensor](https://www.ebay.es/itm/M12-Sensor-Proximidad-Inductivo-PNP-DC-6-36V-LJ12A3-4-Z-BY-3D-Impresora-S0036-/201916693496) a la nave. Está disponible dentro de una caja en la balda de Printer One.

Siotma, 2018-10-11:
Teniendo en cuenta la antigüedad de la Issue, y teniendo en cuenta la [nueva política de issues](https://github.com/asrob-uc3m/impresoras-asrob/issues/89#issuecomment-422891053), doy la compra por finalizada. Para la instalación del sensor en las impresoras se deberá abrir una nueva.

closed 2018-10-11 by Siotma