asrob-uc3m/impresoras-asrob #20: reajuste de pid de la cama caliente

andrespy, 2017-04-08:
La mejora de ponerle un material aislante en la parte inferior de la impresora me parece muy buena idea para disminuir la disipación por la parte inferior y llegar a la temperatura "meta" más rápido.

Ayer imprimiendo, algunas piezas se despegaban y diría que es debido a que al no ser igual la disipacion por la parte superior de la impresora que por debajo el NTC mide una temperatura que ya no es comparable a la superior y por tanto propongo:

* aumentar la temperatura "meta", para ello estudiaría la temperatura superior con respecto de la inferior con un sensor aparte.

* cambiaría la temperatura en marlin para que no salte el error thermal runaway al imprimir

* volveria a correr el comando M303 E0 S200 C8 (la cama prueba su dinamica y saca unos valores de kp ki kd) http://reprap.org/wiki/PID_Tuning
Y los cargaría en el marlin

Siotma, 2017-04-15:
Lo que comentas de comprobar las temperaturas de superficie y bajo la cama caliente me parece muy interesante, se podría hacer con otro termistor.

Respecto al Thermal Runaway, si ha vuelto a saltar, se debería seguir ampliando la tolerancia sí.

Pero en cuanto al PID de la cama, me temo que es imposible puesto que va con relé y por ese motivo el PID está desactivado. Activar el PID de la cama convertiría la impresora en un timbre y el relé se estropearía rápidamente. Me temo que esta mejora no va a poder implementarse.

Las dos primeras, son grandes ideas para nuestras próximas reuniones. Muchas gracias Andrés! 👍

Siotma, 2017-05-28:
Esta issue se está volviendo un poco antigua. De las tres propuestas, dos ya se han resuelto:

- [x] Ya se amplió la tolerancia del thermal runaway en https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-298057269 y en https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-298969157

- [x] El PID de la cama caliente se encuentra desactivado y no se debe activar para evitar un rápido desgaste del relé.

- [ ] Faltaría hacer la prueba de temperatura de la cama caliente con un segundo termistor.

Yo tengo un termistor en casa con cables y todo que podría servir para esto. Voy a añadir la tarea al proyecto.
Añadido al proyecto.

Siotma, 2018-04-21:
Los problemas descritos en esta issue ya no se dan. Además, en caso de querer incrementar la temperatura superior de la cama, siempre se puede incrementar la temperatura objetivo. La misión del termistor no es, en mi opinión, tener exactitud en la medida, sino control.

Cerrado por antigüedad.

closed 2018-04-21 by Siotma