asrob-uc3m/impresoras-asrob #21: Desaparición de soporte de motor de Blacky

jorgercobo, 2017-05-09:
Como el titulo explica, una de las piezas que sujetan los motores de Blacky para evitar que vibren ha desaparecido. Creo que el diseño era de @Siotma 

- [ ] Imprimir otro soporte de motor

Siotma, 2017-05-09:
En efecto, el diseño inicial es mío, ya que yo diseñe [estas piezas](https://github.com/Siotma/developrints/tree/master/Soporte%20motor%20Z) para mi impresora.

Sin embargo, no ha desaparecido nada, puesto que únicamente se imprimió una pieza, no dos.

El motivo de que solo se imprimiera una es que por aquel entonces, Hija Resurrection tenía el problema de pasos por mm ( #4 ), y de hecho las medidas de la pieza que hay son erróneas, y la pieza es más grande de lo debido.

Por otra parte, para imprimir la pieza para Blacky, hay que escalarla en Z, ya que los motores están mucho más cerca de la mesa que en el caso de mi impresora.

Propongo como ejercicio, rediseñar la pieza en FreeCAD ahora que sabemos :P y hacerla a medida de Blacky.

jorgercobo, 2017-05-09:
Pues creo que está cambiada de sitio y al verla vibrar me he hecho un lio 😅

Siotma, 2017-05-18:
Hoy, he roto el único soporte que había, ya que ha demostrado ser demasiado grueso y desequilibrar a Blacky, actualizo la tarea a "Diseñar a medida e imprimir soportes de motores de Blacky".

Siotma, 2017-05-25:
Hoy hemos medido la distancia entre los motores y la mesa: 2mm aproximadamente, y he diseñado esta pieza: [Soporte motor Z Blacky.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/1028841/Soporte.motor.Z.Blacky.zip)

Hemos impreso uno con Hija Resurrection (para que sea negro y haga juego con el resto de las piezas de Blacky):

[image]

Quedaría por imprimir otro idéntico. El archivo está en el ordenador de Hija Resurrection dentro de la carpeta Printer One, que está en el Escritorio:

[image]

Planeado para la siguiente sesión.

Siotma, 2017-06-07:
Hoy, se ha impreso la base que faltaba: https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-306869765

closed 2017-06-07 by Siotma