asrob-uc3m/impresoras-asrob #22: Recalibrar Blacky

jorgercobo, 2017-05-09:
La cama caliente de Blacky está ligeramente desnivelada y es dificil recalibrarla ya que en la posición correcta los tornillos que la regulan:

- El delantero izquierdo necesita estar muy suelto por lo que la cama puede vibrar.
- El delantero derecho necesita estar muy apretado por lo que no deja libertad para ajustarlo.

Ademas el final de carrera cada día está más suelto y es necesario reposicionarlo en cada impresión.
Creo que es necesario aumentar la prioridad de mejorar el final de carrera del eje Z de Blacky y solucionar el tema de los tornillos.

Siotma, 2017-05-09:
Si mal no recuerdo, @andrespy ya imprimió las piezas para equiparar el final de carrera de Z de Blacky al de Hija Resurrection.

Como bien dices, primero final de carrera, y segundo nivelado de la impresora

Siotma, 2017-05-10:
Hoy he montado las piezas que imprimió @andrespy para el final de carrera, sin embargo, presentan dificultades que han impedido dejarlo funcionando.

Ahora mismo Blacky está sin rollo de filamento por seguridad (está en el armario). En la próxima reunión lo vemos más detenidamente.

Siotma, 2017-05-18:
En la reunión exprés de hoy (18:00 - 19:30), @andrespy y yo hemos terminado de colocar el final de carrera y nivelado correctamente la impresora. Ver https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-302496980

closed 2017-05-18 by Siotma