asrob-uc3m/impresoras-asrob #23: Thermal runaway

pamarinp, 2017-05-18:
Mientras imprimia 4 ruedas del protobot se ha detenido la impresión a los 20 minutos aproximadamente en la segunda da capa. El fallo es debido a una caída de temperatura por parte del termistor del extrusor. 
[image]
[image]
[image]

He intentado continuar la impresión sin éxito puesto que el programa se queda en calentando la impresión y no responder de ni a control manual de los motores.

Siotma, 2017-05-18:
Es un fallo que ha sido recurrente últimamente, ya que detecta variaciones grandes en las temperaturas y realiza una parada de emergencia para evitar que la impresora se autodestruya.

Aunque las últimas veces se había activado la protección de forma demasiado rigurosa ( #20 , https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-298057269 , https://github.com/asrob-uc3m/impresoras-asrob/issues/12#issuecomment-298969157 ); en este caso yo creo que se ha activado correctamente, a la vista de la gráfica de temperatura 🎉 

Sospecho que el Kapton que sujeta el termistor del extrusor está cada vez más deteriorado y el termistor está empezando a moverse durante la impresión.

Recomiendo que se cambie el Kapton del extrusor por un trozo nuevo.

Gracias por la documentación @pamarinp , nos ponemos con ello!

Siotma, 2017-05-25:
Hoy hemos repuesto el Kapton en Hija Resurrection. He traído un poco del mío mientras se adquiere el de la asociación ( https://github.com/asrob-uc3m/actas/issues/62 ).

Ha quedado muy bien, afianza el termistor firmemente y sobre todo ¡¡No se chamusca!! XD

Adjunto imagen que muestra el estado en que se encontraba el extrusor:
[image]

Y ahora:
[image]

La impresora ha estado imprimiendo dos horas seguidas sin errores, por lo que damos por validado el arreglo:

[image]

closed 2017-05-25 by Siotma