asrob-uc3m/impresoras-asrob #24: Soldar y colocar los ventiladores de la electrónica de Blacky

Siotma, 2017-05-25:
La posición más óptima parece ser crimpado directamente a los cables de la fuente de alimentación correspondientes al conector de 5A ( [ver imagen](http://www.reprap.org/wiki/File:Arduinomega1-4connectors.png) )

Siotma, 2017-10-13:
Dado que se cambió la fuente de alimentación ( #28 ), ahora debería ser mucho más fácil añadir los ventiladores utilizando una nueva línea de 12V y conectores apropiados.

GioSunrider, 2017-10-30:
Se han soldado y colocado los ventiladores a la electrónica.  Se han soldado a la fuente en paralelo en un extremo de un canal de 12V. Adjunto fotos de la colocación.
[image]
[image]

Siotma, 2017-10-30:
¿Qué ha pasado en esta Issue? Aparece un tal @ghost 😕  y se ha perdido el comentario de @GioSunrider... 

Jorge, has eliminado tu usuario?

No entiendo nada...

Siotma, 2017-10-30:
Ya me aparece todo bien. Habrá sido un fallo de GitHub.

Lo has hecho fantástico: Asignar, cerrar y mover la ficha a Done.

GioSunrider, 2017-10-30:
Gracias 😊😊