asrob-uc3m/impresoras-asrob #26: Thermal Runaway 2 en Blacky y problema de fijación de espejo en Hija Resurrection

pamarinp, 2017-05-31:
Incluso con el nuevo kapton y lo bien fijo que está el termistor, al imprimir dos ruedas del protobot, ha saltado el mensaje de thermal runaway de nuevo. Siento no haber hecho más fotos. Cerré el programa y lo volví a abrir rápido para aprovechar y poner sólo una rueda y funcionó bien.
[image]

Fijáos bien siempre antes de imprimir que el espejo de la plataforma esté fijo si no os va a pasar como a mi
[image]

Siotma, 2017-06-04:
Gracias por dejar constancia de este problema.

Dado que la captura del error de Thermal Runaway está hecha en Windows, deduzco que has intercambiado los nombres de las impresoras en el título de la issue:
- Hija Resurrection imprime en negro y usa windows xp.
- Blacky imprime en amarillo y usa lubuntu.

Tras un Thermal Runaway lo más importante es **registrar la curva de temperatura** para poder saber si se trata de un error de interpretación del software o si por el contrario el software ha actuado bien protegiendo a la impresora de un mal funcionamiento de algún componente.

Tengo entendido que en las últimas impresiones que habéis realizado no ha vuelto a aparecer este mensaje por lo que no hay razones para pensar que la impresora no funcione.

Propongo dejar esta issue abierta durante una semana más, y en caso de que se vuelva a producir un error, os ruego que capturéis la forma de la curva de temperatura.

Respecto al problema con la fijación del espejo, sin duda es para tenerlo en cuenta y que no nos pase 🤣

Siotma, 2017-06-24:
Ayer durante el taller de Zowimanoides me fue imposible imprimir piezas con Hija Resurrection por el problema de Thermal Runaway. Olvidé hacer fotos, pero describo lo que observé:

- El extrusor y la cama caliente son capaces de alcanza correctamente las temperaturas consignadas y las mantienen.

- Al comenzar a imprimir, y por tanto, a moverse la impresora, la curva de temperatura del extrusor fluctúa significativamente, lo que hace saltar el error.

- Hice una prueba extensiva, manteniendo la impresora caliente más de media hora, pero sin imprimir, y mantuvo las temperaturas correctamente.

Urge, por tanto, revisar la impresora para identificar el problema.

jgvictores, 2017-09-21:
@Siotma ¿Crees que esto se merece un "label: NO FUNCIONA"? ¿Quizás deberíamos añadir una etiqueta de valor intermedio estilo WARNING?

Siotma, 2017-09-22:
Gracias por retomar el hilo, @jgvictores .

En absoluto. Una etiqueta WARNING no tiene sentido, ya que en última instancia significaría lo mismo que la etiqueta NO FUNCIONA, es decir, que la impresora no puede ser utilizada por el público. Olvidé por completo colocar la etiqueta antes de irnos de vacaciones pero ya mismo lo hago.

Recalcar que en la [última reunión de Printer One](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora#15_de_septiembre_de_2017) tratamos de imprimir dos piezas con resultado erróneo, por tanto, procede colocar la etiqueta "No funciona"

Siotma, 2017-09-28:
Hoy se han hecho nuevas pruebas en Hija Resurrection.

Se ha descubierto una posible causa del problema del Thermal Runaway, y es que la fuente de alimentación de Hija Resurrection únicamente tiene 17A en el canal de 12V. Este amperaje es muy reducido para alimentar cama caliente y extrusor, que llegan a demandar corrientes cercanas a los 30A.

Ello se puede comprobar en la siguiente imagen, en la que se ve cómo el sistema es incapaz de mantener el extrusor caliente mientras se calienta la cama; tardando todo ello mucho tiempo. 

[image]
Además, se ha comprobado que la impresora no es capaz de alcanzar 240 grados en el extrusor, siendo únicamente posible alcanzar 210, de lo que se deduce también falta de potencia.
Teniendo esto en cuenta, se ha analizado el comportamiento de las temperaturas en cama caliente y extrusor y se ha detectado que en los intervalos de tiempo en los que el relé está activo, la temperatura del extrusor cae, y únicamente se recupera en los intervalos del relé apagado. Ello se observa bien en la siguiente imagen.
[image]

Así pues proponemos como solución al Thermal Runaway cambiar la fuente de alimentación por otra idéntica a la utilizada en #28 , que ha demostrado tener potencia suficiente para calentar Blacky sin problemas.

Siotma, 2017-11-13:
Hoy se ha colocado una nueva fuente de alimentación en Hija Resurrection ( ver https://github.com/asrob-uc3m/impresoras-asrob/issues/35#issuecomment-344007565 ).

Se ha modificado la configuración de Marlin vía 83d843e02ac85237739cad3ec4fa42780e4ceed3 para aumentar la frecuencia de refresco de la temperatura de la cama.

Se han realizado dos pruebas de impresión, la primera de un cubo, satisfactoria:

[image]

La segunda prueba de impresión de una pieza de calibración, ha sido insatisfactoria por desplazamiento del eje Y:

[image]

Sin embargo, en ninguna de las dos ocasiones ha saltado el error de Thermal Runaway, y a la luz de las curvas de temperatura parece que también se ha corregido la falta de potencia:

[image]

Siotma, 2017-11-13:
Dada la antigüedad de esta Issue, queda sustituida por #35

closed 2017-11-13 by Siotma