asrob-uc3m/impresoras-asrob #27: Error en hija resurrection.

pamarinp, 2017-06-07:
Esta vez ha sido un error de conexión. A mitad de la impresión ha saltado el siguiente error.
[image]

Siotma, 2017-06-24:
Los errores de conexión son mucho más comunes al imprimir desde el PC. Imprimiendo desde tarjeta SD se suelen minimizar.

También creo que sería interesante actualizar la versión de Repetier a la última versión ya que creo que tienen mejoras para evitar estos errores.

David-Estevez, 2017-06-26:
En el caso de mi impresora de casa, los errores de conexión se resolvieron al usar cables USB con ferrita. No me he fijado en si los cables de las impresoras de la nave tienen o no ferrita, pero si no la tienen debería mejorar la conexión al cambiarlos.

Siotma, 2017-09-22:
Este error se volvió a manifestar en la [última reunión de Printer One](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora#15_de_septiembre_de_2017). Por tanto, urge repararlo.

Siotma, 2017-10-24:
La actualización en el software del ordenador de Hija Resurrection ( #3 ) debería haber solucionado este problema.

closed 2017-10-24 by Siotma