asrob-uc3m/impresoras-asrob #28: La fuente de Blacky ha muerto

Siotma, 2017-09-15:
La fuente de alimentación de PC que estaba instalada en Blacky, ha dejado de funcionar.

Hemos intentado imprimir con la impresora, pero la fuente no se encendía.

Hemos identificado que la fuente estaba rota y la hemos tirado a la basura.

Recomendamos, cambiar la fuente por una nueva del armario.

Siotma, 2017-09-16:
Aprovechando el cambio de la fuente de alimentación, sería bueno mejorar el uso de los cables, en la línea de lo que comenzó a hacer @Morgloz ayer.

En la fuente antigua únicamente se habían dejado dos canales de 12V que llegaban directamente a la RAMPS. Sin embargo, las necesidades de 12V de la impresora también pasan por la conexión de ventiladores y el relé de la cama caliente. Por ello, se había recurrido a puentear las conexiones, soldando varios cables en paralelo bajo el conector. Este tipo de instalaciones dificultan el mantenimiento de la impresora.

De esta forma, sería bueno aprovechar los múltiples cables de 12V de la nueva fuente para que cada uno de servicio a un sistema concreto de la impresora. Además, sería muy positivo utilizar conectores para facilitar la desconexión o sustitución de algún componente en caso de mantenimiento.

Asimismo, recuerdo la intención de poner ventiladores a la electrónica de Blacky #24 , por lo que recomiendo que esto se tenga en cuenta a la hora de dejar un canal de 12V libre y accesible para este fin.

Siotma, 2017-09-28:
Se ha colocado una nueva fuente de alimentación en Blacky. 

No se ha utilizado la fuente que comenzó a modificar @Morgloz, ya que aquella tenía poca potencia en el canal de 12V, solamente 21A. La fuente finalmente utilizada estaba en el armario y tiene dos canales de 12V de 18A cada uno, lo que hacen un total de 36A, por lo que tiene potencia suficiente para calentar cama y extrusor rápidamente. Por comparar con la fuente anterior, tenía únicamente 20A en el canal de 12V.

La única pega es que la fuente nueva tarda en encenderse desde que se activa el interruptor, y por eso puede parecer que algo está mal conectado. Únicamente se debe tener paciencia y esperar unos segundos.

La siguiente imagen muestra la curva de temperatura de un calentamiento simultáneo de cama y extrusor, y se puede comprobar la gran velocidad a la que se produce.
[image]

Finalmente, se ha impreso satisfactoriamente un cubo de calibración, que se muestra a continuación.
[image]

Siotma, 2017-09-28:
Créditos a @JorFru por coordinar el proceso de soldado y realizar el cable management. Creo que te has ganado cerrar esta issue.

JorFru, 2017-09-28:
Blacky ya funciona!

closed 2017-09-28 by JorFru

Siotma, 2017-11-29:
Finalmente, Blacky ha vuelto a un funcionamiento fiable con una fuente de alimentación distinta a la documentada en esta issue (ver problemas y resolución en #33 )