asrob-uc3m/impresoras-asrob #29: Actualizar lista de operadores

jgvictores, 2017-09-21:
La [lista de operadores](http://asrob.uc3m.es/index.php/Operadores) no parece muy actualizada. Deberíamos de actualizarla. :-)

davidvelascogarcia, 2017-10-07:
@jgvictores
Perfecto, por mí parte podría encargarme de ello si no hay nadie disponible.
Comentándolo con @Siotma lo único que necesitaría sería una cuenta para poder editar, el listado de nuevas incorporaciones y creo que nada más dado que @Siotma , me ha facilitado un documento con el lenguaje para editarlo.

jgvictores, 2017-10-08:
@davidvelascogarcia Te he creado una cuenta wiki con el mismo nombre de usuario que el de github, te llegará un email con contraseña aleatoria a cambiar al email que tengo asociado a tí.

davidvelascogarcia, 2017-10-08:
@jgvictores Perfecto, muchas gracias. He seguido el proceso de configuración, me he estado adaptando al entorno de la wiki y al lenguaje de edición. He empezado a actualizar la lista de operadores, de momento solo me he incluido porque no tengo los datos del resto de las nuevas incorporaciones a la lista. No obstante, ya se manejarme un poco con la edición, así que podría incorporar todos los demás datos en cuanto consiga el listado.
Por lo demás, si se requiere alguna modificación adicional a parte de actualizar el listado, podría encargarme de ello.
Un saludo.

Siotma, 2017-10-13:
Hoy, me he tomado la libertad de auto-añadirme a la lista.

jgvictores, 2017-10-13:
Gracias @davidvelascogarcia @Siotma !!

Creo que la lista aun así no está actualizada del todo. Ideas:
1. Orden: ((Invertir la lista daría más visibilidad a los más nuevos y por tanto mayor posibilidad de seguir activos... peero debemos fiarnos de que se siga esta práctica, que editando la wiki es un lío de cuidado, es mucho más fácil anexar al final)). Mi propuesta es ordenar un operario por fila, y se pueden poner más datos, como fecha de hacerse operario, operarios formados, y por último definir la tabla como class="wikitable sortable", véase https://www.mediawiki.org/wiki/Help:Sorting#Sorting_rows_of_a_table
1. Limpieza: Enviar emails a los operarios preguntando si quieren seguir. Puede ser largo y tedioso, creo que con implementar el punto anterior bastaría.

¿Cómo lo veis?

David-Estevez, 2017-10-13:
> Limpieza: Enviar emails a los operarios preguntando si quieren seguir. Puede ser largo y tedioso, creo que con implementar el punto anterior bastaría.

Existe una lista de [antiguos miembros](http://asrob.uc3m.es/index.php/Antiguos_Miembros) en la que encontrar a gente que estuvo en la asociación y ya no está. Se puede usar como punto de partida para quitar a esa gente de la lista de operadores y evitar mandar tanto email.

davidvelascogarcia, 2017-10-13:
@jgvictores  @Siotma  @David-Estevez 

Por mi parte, estoy de acuerdo con las propuestas, me parece muy interesante añadir datos como el número de operarios formados o la fecha de fin de formación. Al situarlo por filas, puede parecer más organizado, lo único siendo tantos puede extenderse mucho la longitud en la web, a lo mejor de dos en dos.
He preparado un boceto, para ver que os parece algo del estilo.

[image]

Figura 1. Boceto lista de operadores

Respecto al orden, e invertirlo, puede ser buena idea, pero quizá, no solo situar por novedad, sino por disponibilidad y actividad, me refiero:
En el momento que un nuevo interesado, quiera formarse como operador, bien por tener acceso a las impresoras, para proyectos o derivados, o que quiera formar posteriormente a nuevas incorporaciones, se desplazará a la web y por comodidad salvo que conozca a alguno de los operadores previamente, es posible que se decante por las primeras líneas.
Dado que hay antiguos miembros en la lista, quizá el orden en sentido descendente pueda ser de mayor a menor nivel de actividad y disponibilidad, para reducir los tiempos.
En el punto de limpieza, podría usarse la lista de antiguos miembros y preguntar si siguen interesados y si retomaran el papel de operador y si están interesados en formar...

Siotma, 2017-10-13:
El formato de la tabla que propones me parece muy bueno. Tal vez se puede añadir un nuevo campo que contenga la última actividad del operador, y que ese sea el criterio de ordenación descendente. En caso de que el operador se encuentre inactivo durante mucho tiempo, este mismo parámetro puede servir para sacarlo de la lista y que ésta se auto-mantenga fácilmente. La disponibilidad siempre es un poco relativa, y creo que es mejor que cada persona interesada la consulte en el momento del contacto.

jgvictores, 2017-10-13:
Si veis el tipo de tabla que propongo (en el móvil sólo me funciona en modo PC), está guai porque lo puedes reordenar según el criterio que quieras. Pero claro, sería un operario por fila. https://www.mediawiki.org/wiki/Help:Sorting#Sorting_rows_of_a_table

Siotma, 2017-10-14:
Me gusta mucho ese tipo de tabla. Sería muy interesante probar qué se puede hacer con él.

davidvelascogarcia, 2017-10-18:
@jgvictores @Siotma @David-Estevez 

He estado mirando el formato de tabla propuesta por @jgvictores , y he preparado un pequeño boceto y un poco el código, basado en la propuesta para ver que os parece como punto de partida. Tiene usuario por linea.

[image]
Figura 1. Boceto operadores

Aproximadamente, el código podría ser como el siguiente:

`{| class="wikitable sortable"
|-
! style="background-color:navajowhite" | Foto!! style="background-color:navajowhite" | Nombre!! style="background-color:navajowhite" | Apellidos!! style="background-color:navajowhite" | Correo!! style="background-color:navajowhite" | Sitio web!! style="background-color:navajowhite" | Fecha de formación!! style="background-color:navajowhite" | Nº de operadores formados
|-
| [[Archivo:foto1.jpg|foto1.]] || Nombre1 || Apellido1 Apellido2 || Correo1 || [http://example.com] || 18/10/2017|| 1
|-
| [[Archivo:foto2.jpg|foto2.]] || Nombre2 || Apellido12 Apellido22 || Correo2 || [http://example.com]|| 18/10/2017 || 6
|-
| [[Archivo:foto3.jpg|foto3.]] || Nombre3 || Apellido13 Apellido23 || Correo3 || [http://example.com]|| 18/10/2017 || 7
|-
| [[Archivo:foto4.jpg|foto4.]] || Nombre4 || Apellido14 Apellido24 || Correo4 || [http://example.com]|| 18/10/2017 || 15
...
|}`

jgvictores, 2017-10-18:
@davidvelascogarcia Buena pinta!
PD: Te edité el markdown en tu comentario, de comilla simple a triple, que es el que sirve para múltiples líneas.

davidvelascogarcia, 2017-10-18:
@jgvictores  
Perfecto, muchas gracias.
Lo estoy mirando en previsualización, sin tocar el código original, pero de todos modos he hecho un backup de todo el código por si acaso.

[image]

Figura 1. Tabla en previsualización

Voy a finalizar el código con los datos de todos los operadores para dejarlo listo con este formato, que en 15 minutos aproximadamente esta terminado, y subo por este hilo una previsualización de como sería.
Me quedo a la espera de vuestra respuesta, para ver si lo termino de subir, o dejo el formato actual y debajo el nuevo...

jgvictores, 2017-10-18:
Muy buena pinta. Dale sin miedo, en wiki queda el histórico y es fácil de revertir (rollback) en caso de hacer algo mal.

davidvelascogarcia, 2017-10-18:
@jgvictores 
Perfecto, muchas gracias. Voy a extenderlo a todos los operadores.
He estado haciendo pruebas y funciona todo bien, correos y sitios web incluídos.

[image]

Figura 1. Prueba de tablas

davidvelascogarcia, 2017-10-18:
@jgvictores @Siotma  @David-Estevez 

He terminado con las actualizaciones de la web de operadores. Añado una foto y adjunto el link, para ver que os parece. He estado haciendo pruebas y en principio funciona todo correctamente, si encontráis algo extraño se cambia en un momento.

[image]

Figura 1. Web de operadores

Link:
http://ieee.uc3m.es/index.php/Operadores#Operadores_Autorizados

Lo único que haría falta ahora serían las fechas de formación y número de formados, aunque fuera orientativo.

jgvictores, 2017-10-18:
Está genial! 
Como única mejora, bajar los honoríficos al final, que confunde un poco.

davidvelascogarcia, 2017-10-18:
Muchas gracias.
Listo, ya esta actualizado.

davidvelascogarcia, 2017-10-18:
@jgvictores Por cierto, he hecho un par de backups del antes y el después por seguridad, no vaya a ser que haya algo mal, aunque se puede revertir. Son 14.7 KB, como ocupa nada, esta en un txt, lo he subido en una carpeta a Sourceforge y en otra en Impresoras en Github.
Espero que no os importe, sino lo borro.

Siotma, 2017-10-19:
Me encanta el resultado!! Sin embargo, creo que sigue siendo necesario una limpieza de esta lista, ya que aunque todos los operadores que figuran en ella supieron, en algún momento, utilizar las impresoras, creo que es importante resaltar el detalle de que las impresoras disponibles actualmente, así como los softwares que las controlan han cambiado desde que ellos se formaron.

Tal vez podamos implantar algún modelo de "reválida" aprovechando la casilla "Fecha de formación", de manera que ningún operador pueda tener una antigüedad superior a un año, por ejemplo. Para renovar el rango, podemos plantear una pequeña prueba bajo la supervisión de algún otro operador con rango en vigor, de que efectivamente sabe manejar las impresoras. De esta forma, en caso de haberse producido cambios en el manejo, el operador puede ponerse al día con la ayuda del otro. Tras pasar la prueba, la fecha de formación se actualizaría.

Este es solo un ejemplo, pero creo que podría servir. Siempre pienso que este tipo de cosas que se hacen una vez pero según pasa el tiempo se quedan obsoletas necesitan un método de automantenimiento definido, de forma que los que vengan después puedan mantenerlas actualizadas para que sigan cumpliendo su función.

En caso de aprobarse algo así, debería reflejarse en los [Estatutos de la Asociación](https://asrob-uc3m.gitbooks.io/estatutos/content/definiciones.html) y notificarse a todos los operadores de la lista.

Qué os parece.

David-Estevez, 2017-10-19:
Por lo que veo, de toda esa lista en activo sólo hay 4 personas, las demás dejaron la Asociación/Universidad/no han impreso nunca con nuestras impresoras. Esto se puede comprobar mirando la lista de [antiguos miembros](http://asrob.uc3m.es/index.php/Antiguos_Miembros) y comparando nombres y correos. 

Respecto a lo que propone @Siotma, me parece bien que "caduque" el título de operador, pero no por fecha de formación, sino por fecha de última impresión. No tiene mucho sentido que alguien que se formó hace un año pero ha impreso de forma asídua y sabe como funcionan las impresoras pierda el título de operador porque sí que sabe hacer uso de ellas. 

PD: kudos a @davidvelascogarcia por el curro de la nueva tabla

davidvelascogarcia, 2017-10-19:
@Siotma @David-Estevez

Muchas gracias, me alegro de que os guste.

Coincido con los dos,  podría aprovecharse las casillas de formación.
Respecto a la limpieza estoy de acuerdo. Lo siento, no tenía conocimiento de que la gran mayoría ya no se encuentran en la universidad y por lo tanto no están disponibles para realizar formaciones ni tener constancia de las actualizaciones del software, método de uso...
Podrían pasarse a una nueva pestaña como antiguos operadores, sería muy simple y el código el mismo. Y así dejar la web de operadores para activos.
Sugiero la web de antiguos operadores que sin problema podría encargarme yo, suponiendo que posiblemente algunos operadores en su momento fueron muy activos y formaron a muchos miembros y aportaron a la asociación, por otorgarles un reconocimiento.
Me podría poner en contacto con ellos para preguntarles por el número de miembros que han formado y fechas orientativas.
¿Qué os parece?.

Siotma, 2017-10-19:
> me parece bien que "caduque" el título de operador, pero no por fecha de formación, sino por fecha de última impresión.

La idea, es que la renovación del título de operador la realice un tercero, de manera que haya más control. Controlar el momento en que una persona realizó su primera impresión es difícil. Si hay un supervisor que "da fé" puede ser más fácil.

> Podrían pasarse a una nueva pestaña como antiguos operadores, sería muy simple y el código el mismo. Y así dejar la web de operadores para activos.

A favor de una página de antiguos operadores, y que se trate de dar crédito a cada uno de ellos por sus contribuciones.

Como aportación propia, y aprovechando que ya se ha hablado de migrar la página de operadores a la [Web](http://asrob-uc3m.github.io/), quiero compartir una idea que se me ha ocurrido: A lo mejor es deformación, pero creo que las Issues serían un sistema ideal para gestionar la formación y renovación de operadores. Todos los interesados podrían abrir una issue para solicitar formación, los operadores disponibles se podrían poner en contacto para concertar la sesiones y se podrían subir fotos de las impresiones realizadas durante el proceso. Por eso, si finalmente se hiciera una página de operadores en la [Web](http://asrob-uc3m.github.io/), propongo que se haga al estilo de la [página de impresoras](http://asrob.uc3m.es/printers/), de manera que haya un repo dedicado donde se encuentre el código de la página así como las issues que gestionen la formación de los operadores.

David-Estevez, 2017-10-19:
Entendido, en ese sentido estoy totalmente de acuerdo con tu idea.

La [página de impresoras](http://asrob.uc3m.es/printers/), como tú dices, podría ampliarse para mostrar información sobre los operadores  y tomar esa información de un repositorio. Ahora mismo estoy a tope con subvenciones y memorias, pero cuando pase la tempestad se puede mirar cómo implementar esa mejora. Es más, si tú tienes ya alguna idea de cómo sería, lo puedes proponer en ese repo mediante [una issue](https://github.com/asrob-uc3m/printers-status-webpage/issues).

davidvelascogarcia, 2017-10-19:
Durante la tarde, he estado preparando la página de Antiguos Operadores, basándome en la propuesta de @David-Estevez  de la lista de antiguos miembros.
Link: [http://ieee.uc3m.es/index.php/Antiguos_Miembros](url)

Todos los operadores incluidos en Antiguos Miembros, han sido trasladados a la página Antiguos Operadores.
Link: [http://asrob.uc3m.es/index.php/Antiguos_Operadores](url)

Ahora están separados en dos web.
Operadores, para los no incluidos en antiguos miembros.
Link: [http://ieee.uc3m.es/index.php/Operadores](url)

¿Qué os parece?¿se os ocurre alguna mejora hasta incorporar el código y las issues?

davidvelascogarcia, 2017-10-20:
Acabo de añadir un enlace en cada una las páginas para poder acceder de una a otra sin problema.
Gracias por la revisión @Siotma .

Siotma, 2017-11-29:
La [nueva página de operadores](http://asrob.uc3m.es/printers/operators.html) está activa y funcionando.

En este momento, nos enfrentamos a un proceso de transición en el que se abren distintas tareas:

- [ ] Hacemos el cambio ya o damos un tiempo de latencia para detectar bugs?
- [ ] Si se decide hacer la transición, habría que indicar con una nota en la antigua página que ya no se actualizará más, y poner un link a la nueva.
- [ ] Si se decide hacer la transición, habría que establecer una fecha límite para que los operadores que figuran en la actual lista migren automáticamente a la nueva al estilo de https://github.com/asrob-uc3m/operadores/issues/4
- [ ] Asimismo, habría que establecer la duración máxima del título de operador entre renovaciones.
- [ ] Por último, habría que detallar, el procedimiento de formación, tanto para el nuevo operador como para el formador, e indicarlo claramente en la web y en el repo de https://github.com/asrob-uc3m/operadores/

Pregunta: ¿migramos esta discusión a https://github.com/asrob-uc3m/operadores/ ?

jgvictores, 2017-11-29:
> Pregunta: ¿migramos esta discusión a https://github.com/asrob-uc3m/operadores/ ?

Sin duda! https://github-issue-mover.appspot.com/ funciona muy bien!

jgvictores, 2017-11-29:
PD: Soy muy dado a hacer la transición, para forzar a ponernos en modo beta-testers. :-)

Siotma, 2017-11-29:
> Sin duda! https://github-issue-mover.appspot.com/ funciona muy bien!

No me refería tanto a migrar totalmente la issue, sino a cerrar esta dándola por finalizada y abrir otra en https://github.com/asrob-uc3m/operadores/ para aclarar los puntos de https://github.com/asrob-uc3m/impresoras-asrob/issues/29#issuecomment-347841784

Siotma, 2017-11-29:
Discusión migrada a: https://github.com/asrob-uc3m/operadores/issues/14

 Creo que @davidvelascogarcia tiene el derecho más que ganado de cerrar esta issue.

davidvelascogarcia, 2017-11-29:
@Siotma muchas gracias. Dado que se ha migrado y estamos de acuerdo, cierro la issue.

closed 2017-11-29 by davidvelascogarcia