asrob-uc3m/impresoras-asrob #30: Propuesta: Préstamo de uso de Dédalo

David-Estevez, 2017-09-30:
Actualmente tengo una Prusa i2 ([clon 28, Dédalo](http://reprap.org/wiki/Clone_Wars:_El_imperio_de_los_clones/es#Jedis)) cogiendo polvo en casa porque desde que me compré una [m-prime](http://mprime.io/) ya no la uso.

[image]

Usa plástico de 3mm y necesita un buen calibrado, pero por lo demás funciona. El principal problema que tiene es que, al usar plástico de 3mm, la fuerza del hilo de plástico al entrar en el hotend hace que el mismo oscile, dejando un patrón característico en las piezas que imprime.

Abro este hilo para preguntaros si estaríais interesados en un préstamo de la misma y para negociar las condiciones del mismo. Mi propuesta es la siguiente:

* Yo cedería temporalmente la impresora y todo el plástico de 3mm que me queda a ASROB para uso y disfrute de los miembros de la Asociación. Pero me reservo el derecho a terminar la cesión y llevarme la impresora cuando me apetezca.
* Si en su estado actual no se puede imprimir con ella, o consideráis que necesita alguna mejora como cambiar el hotend para usar hilo de 1.75mm o añadir LCD, yo correría con los gastos de la mejora, previa consulta a mi persona de dichos gastos.
* Los gastos de mantenimiento derivados de su uso en la Asociación correrían a cargo de la Asociación. (Estoy abierto a otras propuestas para este punto)

Comentadme qué opináis sobre la propuesta.

Siotma, 2017-10-01:
Teniendo en cuenta la gran cantidad de gente interesada en impresoras este año, cualquier nueva contribución es bienvenida. Lo más importante es disponer de la máquina y de filamento. Estoy de acuerdo con los criterios que propones, al fin y al cabo, quien pague cada cosa será su propietario.

Por otra, parte, en caso de que lo creas necesario, te recomendaría que hiciersas una copia de seguridad de los archivos de configuración si aún los tienes así como los parámetros de impresión, para que los guardes y también para nosotros podamos utilizarlos como punto de partida.

Sobre el tema de modificaciones te quería hablar. En Printer One una de las máximas es que se pueden realizar abiertamente modificaciones en la impresora a nivel de software o de hardware siempre que:
1. Sean reversibles. 
2. Se documenten los motivos, pasos realizados y resultado final.

Me gustaría que lo valorases, ya que creo que, si finalmente prestases la impresora, su utilidad más clara sería utilizarla para investigar el uso de distintas piezas o configuraciones, así como aprendizaje de los nuevos.

Dado que dices que necesita un buen calibrado, imagino que estarás de acuerdo en que se pueda modificar su firmware, y ajustar la mecánica, pero igual quieres acotar las posibles modificaciones a nivel de hardware (piezas, electrónica, etc) que se practiquen sobre ella.

David-Estevez, 2017-10-01:
Totalmente de acuerdo con la experimentación! Como ya he dicho, es una impresora que ya prácticamente no uso, así que podéis experimentar con ella todo lo que queráis. Eso incluye cambiar piezas y mecánica, ajustes, etc.

Con el firmware, lo mismo, yo os busco el `Configuration.h` de la impresora y vosotros hacéis las modificaciones que veáis oportunas. Hace tanto que no toco el firmware que ya ni me acuerdo de cómo estaba configurado, así que cambiad todo a vuestro gusto.

Respecto a limitaciones de modificaciones, la única que tengo es el presupuesto que le puedo dedicar a esta impresora. Por ello, lo mejor es que me comuniquéis las mejoras que proponéis y negociamos quién se hace cargo del gasto. Como criterio preeliminar propongo lo que os he comentado antes, que los gastos necesarios para el funcionamiento inicial (cambio de hotend, pantalla LCD, etc) corran a cargo mío y el mantenimiento derivado de su uso a cargo de ASROB. Para modificaciones que no sean ni una cosa ni la otra siempre podemos negociar quién lo financia. 

¿Estáis de acuerdo con estas condiciones?

David-Estevez, 2017-10-04:
Aquí os dejo el Configuration.h de mi impresora. Al final resulta que iba a 115200 baudios (my bad).

[Configuration.h.tar.gz](https://github.com/asrob-uc3m/impresoras-asrob/files/1355328/Configuration.h.tar.gz)

Tratádmela con cariño!!

Siotma, 2017-10-05:
> ¿Estáis de acuerdo con estas condiciones?

Totalmente! 🎉  

> Aquí os dejo el Configuration.h de mi impresora. Al final resulta que iba a 115200 baudios (my bad).

¡Muchas gracias por buscarlo!, como puedes ver [aquí](https://github.com/asrob-uc3m/impresoras-asrob/tree/master/Dedalo) , ayer ya estuvimos trasteando y sacamos algunos valores por si no encontrabas ese fichero 👍 Lo suyo sería disponer de ese fichero en la carpeta correspondiente del Repo. Súbelo tú mejor para que quede constancia de que es el original.

> Tratádmela con cariño!!

Pues claro! De hecho, voy a abrir una nueva Issue para gestionar su puesta a punto.

Dado que el préstamo ya se ha realizado, cierro esta Issue.

closed 2017-10-05 by Siotma

David-Estevez, 2017-10-05:
Archivo subido en fe378824ab6218fd56dc411012a1c01a6bfdb569

David-Estevez, 2017-10-05:
Resulta que en la wiki de ASROB hay más información sobre la impresora!

http://asrob.uc3m.es/index.php/Clone_wars:Impresora_Dedalo

Siotma, 2017-10-07:
Enlazado desde el [cuaderno de Bitácora](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora#3_de_octubre_de_2017)