asrob-uc3m/impresoras-asrob #31: Puesta a punto de Dédalo

Siotma, 2017-10-05:
En esta Issue se debe centralizar el trabajo hecho sobre Dédalo hasta que se encuentre operativa.

En la última reunión se propusieron las siguientes líneas de trabajo iniciales:

- [ ] Investigar la placa: versión, firmware que lleva incorporado y configuración.
- [ ] Perfeccionar la mecánica: el carro del eje X oscila mucho y ello es inadmisible en una impresora 3D.

Además, cuando la impresora cobre vida, también sería importante:

- [ ] Calibrarla, para que los movimientos de los ejes sean fieles a las dimensiones reales.
- [ ] Limpiarla: quitarle el polvo y engrasar los ejes.
- [ ] Realizar pruebas de impresión.

Si me he dejado algo, habría que añadirlo en este mensaje para que queden todas las tareas a realizar juntas.

Siotma, 2017-10-05:
En la [última reunión de Printer One](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora#3_de_octubre_de_2017) comenzamos a trabajar con Dédalo:
- Se sabe que dispone de una placa Sanguinololu de las primeras versiones.
- Se ha investigado al configuración del firmware que tiene actualmente esta placa:
-- Funciona a 115200 Baudios 
[image]
-- Aquí se relacionan los parámetros del Configuration.h obtenidos a través de Repetier
[image]
- Asimismo, se han comenzado los trabajos para sustituir el carro del eje X por uno más estable. @gonzalojs1 y @mmartinezt49 han encontrado [estas piezas](https://www.thingiverse.com/thing:181524/#files) y ya hemos impreso la primera con Blacky.

[image]

gonzalojs1, 2017-10-12:
Después de imprimir el carro para el eje x, se fijó este a Dédalo junto con unas patillas nuevas para juntar la correa a dicho carro. La falta de estabilidad en ela dirección x ya esta solucionada para esta impresora.

Siotma, 2017-10-13:
Efectivamente, se montaron las nuevas piezas y también se imprimió una nueva polea para la correa del eje X. Lamentablemente, el montaje realizado provocó que la correa roce con el motor del extrusor y hubo que insertar una tuerca adicional (ver imágenes).

[image]

[image]

Sin embargo, yo solo me he auto resuelto este problema, dándome cuenta de que los extrusores de las impresoras Prusa-i2 van casi siempre torcidos (ver [aquí](http://asrob.uc3m.es/images/thumb/2/24/Dedalo_final.JPG/602px-Dedalo_final.JPG)). Rotar el extrusor debería resolver el problema.

Por otra parte, aunque el carro de X ahora no baila, el extrusor sí lo hace, ya que la fijación es muy pobre, yo propondría imprimir una nueva pieza de extrusor como [esta](https://www.thingiverse.com/download:279662) sacada de [aquí](https://www.thingiverse.com/thing:141915/#files) que permita la fijación mediante tornillos.

Siotma, 2017-10-13:
Como apunte, destacar que la pieza del eje X impresa, sacada de [aquí](https://www.thingiverse.com/thing:181524/#files), está diseñada para rodamientos LM10UU, mientras que los de Dédalo son LM8UU. He encontrado [estas otras piezas](https://www.thingiverse.com/thing:173339), que parecen ser las mismas, pero para los rodamientos correctos, de manera que de ahora en adelante sería mejor imprimir estas.

GioSunrider, 2017-10-17:
Hoy @Sulphie, @carlosmgomezj y @metroid2010 y yo, hemos solucionado el problema de rozamiento del motor con la correa. Para ello se ha cambiado de posicion el lado derecho (segun foto adjunta) colocandolo en el agujero delantero. 

[image]

Una vez colocado esto @Siotma se ha percatado de que la correa no estaba alineada y se han colocado nuevos tornillos mas largos con unas tuercas para ejercer presión y fijar de nuevo el motor. 

[image]

De esta manera la correa tampoco roza con las paredes laterales, evitando su desgaste.
Posteriormente se ha procedido a su encendido, se ha probado con el Repetier, el movimiento de los motores manualmente (con la útima version de cura no se podia), y a calentar la cama caliente y el extrusor, que aparentemente funciona correctamente. Nos hemos quedado a medias de la configuración de la misma en uno de los portátiles. 
También al final de la sesión, nos hemos percatado del estado de la pieza que encaja en el rodamiento del eje Z, contrario a donde se encuentra la placa.

daavm, 2017-10-23:
Hoy hemos calibrado la cama e impreso las primeras pruebas. Con esta impresión nos hemos dado cuenta de algunos cambios y revisiones que todavía necesita Dedalo para su puesta a punto:
- Tensar correas, ya que están algo sueltas y creemos que puede afectar a la impresión
- Cambio pieza de una de las varillas del eje Z, que básicamente, se cae y está medio suelta
- Revisión motores eje Z. Mañana revisaremos la configuración de Marlin, ya que sospechamos que la aceleración puede estar causando algunos problemas.
- Mejora de calibración cama caliente. Quizá usar muelles en lugar del sistema que tiene ahora ayudaría a la hora de calibrar más rápido y mejor. 
Posiblemente me deje algo, en tal caso lo añadimos mañana tras la reunión!

David-Estevez, 2017-10-23:
No sé si lo comenté, pero tengo un par de acoples de estos para el eje Z (una mejora que nunca llegué a instalar). Mañana los llevo para que los podáis usar.

[image]

De hecho debería revisar la caja de repuestos por si hubiese algo más que daros.

Siotma, 2017-10-24:
Sin duda estos acoples mejorarían el funcionamiento del eje Z 😊

Siotma, 2017-10-24:
Para solucionar el patinaje del eje Z, he estado mirando el fichero [Configuration.h](https://github.com/asrob-uc3m/impresoras-asrob/blob/master/Dedalo/Configuration.h) disponible en este repositorio, concretamente esta parte:

```
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80.0801, 79.9306, 2560 , 568.18*15/11}  // default steps per unit for ultimaker 
#define DEFAULT_MAX_FEEDRATE          {500, 500, 5, 45}    // (mm/sec)    
#define DEFAULT_MAX_ACCELERATION {5000,3000 /*5000*/,50,5000} // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for skeinforge 40+, for older versions raise them a lot.
```

Según veo, el valor de aceleración del eje Z es conservador, solo 50 mm/s2. Podría bajarse más para evitar patinajes, pero yo creo que lo más crítico es la velocidad máxima, que está puesta en 5mm/sec.

Yo recomendaría bajar la velocidad a 1mm/sec antes de tocar la aceleración.

Para flashearle un nuevo firmware:
1. Entrad en el [repositorio de Marlin](https://github.com/MarlinFirmware/Marlin/releases) y descargad la última Release (1.1.6). 
2. Una vez descargada, descomprimid todo en una carpeta y abrid el fichero Marlin.ino con la IDE de Arduino.
3. En el fichero Configuration.h que se abrirá, copiad todos los valores del fichero que tenemos en el Repositorio.
4. Subir el firmware en la placa de la impresora, por USB, conectada como si fuera un Arduino Mega.
5. Comprobad si los cambios han corregido el problema.

daavm, 2017-10-25:
En la sesión de hoy hemos avanzado bastante con dédalo:

- @metroid2010 ha instalado los nuevos acoples del eje Z que @David-Estevez ha traído. Este cambio ha parecido mejorar el movimiento del eje z. Anteriormente, este eje se bloqueaba cuando lo intentabamos desplazar en incrementos de 10 a traves de Repetier. Esto ahora no pasa. 

- Hemos tensado las correas del eje Y, ya que se veían ciertas ondulaciones en las impresiones en dicho eje. Se ha impreso otra pieza de prueba y, aunque esas ondulaciones ya no estaban, hemos notado cómo el eje Y parecía desplazarse poco a poco (ver foto). De momento desconocemos por qué pasa esto (el extrusor está un poco suelto quizá?)

[image]

- Uno de los cables de alimentación de la placa (el amarillo, creo recordar), se ha roto. Tenía fácil solución así que ya está arreglado.

- Hemos probado a disminuir la velocidad y aceleración del eje Z, pero a través de la configuración EEPROM del Firmware, en Repetier. En mi opinión la impresion ha mejorado, pero tampoco teníamos muchas pruebas anteriores con las que comparar. 

- Respecto a las pruebas de impresión de ayer lunes, el resultado final es notablemente mejor, veanse videos y fotos: 

[image]

[image]
(No juzguéis la calidad del GIF, si queréis el video está en telegram!)

Dejamos para la próxima reunión pendiente de hacer: 

- Revisar eje Y para intentar evitar ese movimiento en las impresiones
- Buscar PC para Dédalo. Hemos hablado @davidvelascogarcia y yo de la posibilidad de usar el que estaba antes con la impresora hija. No tiene mucha ram (1GB) y el lector de discos no funciona, pero podríamos intentar meterle Linux y usarlo de forma provisional para Dédalo. 
- Imprimir o buscar un soporte para el filamento, para colocárselo a Dédalo. Tener el filamento colgado de uno de los motores...no es una opción D:

De momento no creo dejarme nada, corregidme los que habéis estado hoy si no! :+1:

JorFru, 2017-10-25:
Sugiero destensar un pelín la correa de Y. Puede que ese problema de
desplazamiento sea a causa de pérdida de pasos porque el carro no corre
bien al tener la correa tan tensa.

davidvelascogarcia, 2017-10-25:
Lo comento, a ver que os parece.
Ayer, @daavm y yo , hablamos del hecho de que dado que no hay otro tercer monitor para el futuro PC que controlará Dédalo, se podría utilizar un KVM, y emplear el mismo monitor, teclado y ratón que controla o bien Hija Resturrection o bien Blacky, y alternar de uno a otro con una sola tecla.

Siotma, 2017-10-25:
Muy buen trabajo chicos.

He condensado todas las tareas que habéis ido escribiendo en el primer comentario. Al utilizar el formato de task list cualquiera puede ir marcando las tareas como completadas.

Creo poder aportar luz en la parte de las imprecisiones del eje Y. Hay que tener en cuenta lo siguiente:
- El extrusor oscila un poco, por lo que puede producir errores si se mueve durante la impresión, para eso habría que imprimir un soporte con agujueros para anclarlo con tornillos como [este](https://www.thingiverse.com/download:279662) sacado de [aquí](https://www.thingiverse.com/thing:141915/#files).
- La correa de un eje debe estar tensa pero sin pasarse, como bien dice @JorFru más arriba.
- La corriente que suministra el pololu al motor debe ser suficiente como para que este no pierda pasos. Creo que [este vídeo de Obijuan](https://www.youtube.com/watch?v=iEf7EnCCUao) es perfecto para ello.

Estas tres mismas circunstancias pueden afectar igualmente al eje X. Sería interesante evaluarlo también.

GioSunrider, 2017-10-25:
He añadido a la lista de cosas de por hacer, la mejora que comentamos @daavm y yo que era buscar la manera de mejorar la calibracion de la cama con muelles, porque lo de los tornillos que hay ahora nos ha estando dando guerra en las dos sesiones de esta semana.

David-Estevez, 2017-10-25:
Dos cosillas:

* Respecto al eje Y, le he echado un ojo y lo que parece es que la correa roza con uno de los idlers (el que está por la parte de atrás del extrusor) por lo que se engancha y le cuesta más moverse. Esta puede ser una causa para los problemas en ese eje.

* Respecto a lo de la pantalla + KVM, yo propongo desempolvar el display LCD que hay en el armario y colocárselo a alguna de las dos impresoras que tienen RAMPS. De esta forma se podría empezar a experimentar con la idea que se gestó [aquí](https://github.com/asrob-uc3m/actas/issues/93#issuecomment-335903419) (y que muy probablemente se debería convertir en una issue independiente) y se liberaría un PC que se podría usar en Dédalo. Esto es importante sobre todo porque la idea es reducir el número de PCs, no aumentarlo :sweat_smile:

GioSunrider, 2017-10-30:
Voy por partes que hoy ha sido provechoso, lo primero que he estado haciendo ha sido mirar la configuración de Marlin, he bajado la nueva version y he pasado lo valores uno a uno. Adjunto el nuevo Configuration.h para que simplemente se copie para futuras pruebas. Ya que no se ha conseguido subirlo a la placa, debido a que daba error de velocidad. Se estaba cargando con el IDE de Arduino con la placa Sanguino de Configuracion y a distitas velocidades (16 y 8 MHz). El procesador de la placa es un Sanguino 644p, suponemos que la mejor velocidad es la de 16, pero no hemos conseguido subirlo. Los cambios realizados en Marin vienen marcados con un //DEDALO
```
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
[... 1617 lines cut ...]
  #define MEASUREMENT_DELAY_CM        14    // (cm) The distance from the filament sensor to the melting chamber

  #define MEASURED_UPPER_LIMIT         3.30 // (mm) Upper limit used to validate sensor reading
  #define MEASURED_LOWER_LIMIT         1.90 // (mm) Lower limit used to validate sensor reading
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #define DEFAULT_MEASURED_FILAMENT_DIA DEFAULT_NOMINAL_FILAMENT_DIA // Set measured to nominal initially

  // Display filament width on the LCD status line. Status messages will expire after 5 seconds.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H
```
Posteriormente se ha instalado el ventilador del extrusor, soldado a un canal de 12 V de la fuente de alimentacion, adjunto foto. Queda remarcar que para que no choque con las pinzas de la cama, se coloquen las pequeñas en la parte donde queda el extrusor.

[image]

Luego se ha procedido ha hacer el cambio desde EEPROM FIRMWARE de Repetier, se va a adjuntar el archivo que hay que cargar **CADA VEZ QUE SE CONECTE LA IMPRESORA A REPETIER**.  Una vez acabado esto, se ha hecho una prueba de impresión del mismo cubo que llevamos imprimiendo, pero esta vez sin escalar y con PLA, y el problema en el eje Y sigue siendo el mismo. La impresión que queda hace como ondulaciones. En la foto queda distinguir que no nos dimos cuenta que nos quedamos sin plástico y el final es impreso en vacío, de ahí esa forma tan rara.

[image]

Finalmente se ha intentado cambiar la corriente que reciben los motores, tanto del eje Y, como del eje Z. Ambos estaban a tope, el de Y en torno a valores de 0,54 A y el de Z a 0,20 A. Los motores no deberían perder paso, ya que es corriente suficiente para su movimiento.

Y creo que no me dejo nada mas de lo hecho hoy.

Siotma, 2017-11-06:
Teniendo en cuenta que tanto Blacky como Hija Resurrection no han podido imprimir las últimas sesiones, y dado que Dédalo necesitaba [estas piezas](https://www.thingiverse.com/thing:141915/#files) para continuar su puesta a punto, las he impreso yo:

[image]

Que aproveche!

GioSunrider, 2017-11-09:
Hoy en dedalo hemos montado la nueva pieza que @Siotma imprimió. Tanto como el extrusor como el HotEnd quedan mucho mejor y fijos, asi como la facilidad de meter el filamento. Se adjuntan fotos del montaje. 

[image]

[image]

Lo único que queda es cambiar el sistema de enganche del filamento, las actuales palometas no agarran bien con la nueva pieza, ademas de lo engorroso que resulta apretar con ellas. Se adjuntan fotos de las mismas.

[image]

daavm, 2017-11-10:
En la sesión de hoy yo he intentado (sin éxito) actualizar el firmware de Dédalo. Al intentar subir el firmware, Arduino IDE daba un error: 
`stk500_getsync() attempt 1 of 10 not in sync`
El mismo 10 veces (attempt n of 10). Buscando en foros y demás, la respuesta que más aparecía es la de que o no hay bootloader, o está corrupto. No soy muy hábil con arduino, pero esto reinstalar el bootloader no parece la tarea más sencilla del mundo, y ni siquiera sabemos si realmente eso solucionaría algo. 
@David-Estevez Ayudaría mucho si pudieses dar alguna idea, ya que imagino en su momento tú mismo instalaste el firmware a la impresora.

David-Estevez, 2017-11-10:
Esa respuesta es típica cuando el programador no consigue hablar correctamente con el bootloader, normalmente porque no se ha seleccionado la placa correcta (Sanguinololu).

GioSunrider, 2017-11-12:
@David-Estevez descarta el error de elección de placa, el IDE de Arduino lo tengo configurado para la propia placa Sanguino, y miramos el modelo del procesador. Respecto a la retroalimentación igual, hacemos todo con la fuente encendida. Así que como @daavm me comento, seguramente sea del bootloader, así que el próximo día habrá que ver cómo lo arreglamos.

David-Estevez, 2017-11-12:
Me gusta ir de lo sencillo (IDE mal configurado) a lo chungo (bootloader muerto), de ahí mi sugerencia. :)

Para cargar un nuevo bootloader see necesita o bien un programador (yo tengo en casa) o bien un Arduino programado como programador ICSP. Seguramente la Sanguinololu tenga un puerto ICSP para conectar directamente el programador.

Si esto no funcionase, o si queréis hacer directamente un upgrade, tengo un ATMega1280P que se puede programar de 0 y usar en Dédalo.

GioSunrider, 2017-11-12:
jajaajajajaaj  @David-Estevez decia lo del IDE no, porque lo reconfiguere hace dos semanas para ver si ese era el problema, igual plantear el upgrade no esta mal, si tienes un ATMega1280P nos vendria de perlas porque una de las cosas que comentamos @daavm y yo era ponerle pantallas e igual un lector de sd para no necesitar un ordenador, debido al problema de que no hay mas ordenadores disponibles en ASROB.

David-Estevez, 2017-11-12:
Tengo el micro en casa, lo compré precisamente para ponerle un display.

Si queréis ponerle un display puedo comprar uno como este:   [3D Printer Parts  SH139 Sanguinololu Board Reprap LCD2004 Controller And Adapter For 3D Printer](
 http://s.aliexpress.com/U3uAf6R3 ), ¿Qué os parece?

Siotma, 2017-11-12:
Sería una pasada tener a Dédalo con pantalla. No he visto si ese controlador que has pegado soporta tarjeta SD, eso sería muy importante.

Respecto al cambio de micro. Sería fantástico siempre y cuando el nuevo se pueda programar fácilmente a través del USB de la Sanguinololu. De lo contrario, esta parte seguiría siendo un problema si en un futuro se quisiera modificar alguna configuración.

David-Estevez, 2017-11-12:
Juraría que sí que tiene SD, pero tendría que comprobarlo.

Sobre el nuevo micro, está blanco de fábrica, la primera vez habría que cargarle el bootloader con un programador, luego se programaría como un Arduino normal por USB.

GioSunrider, 2017-11-12:
No se como lo veis pero a mi me parece muy guay el cambio, supondria una potencia de disparo para dedalo increible.

GioSunrider, 2017-11-13:
La impresión de dedalo de hoy, con el nuevo extrusor montado, ignorar la parte de abajo que se ha solapado a un pegote que habia.

[image]

David-Estevez, 2017-11-13:
Wow!!! Qué pedazo de mejora de calidad!!! :clap: :clap: :clap:  Estáis hechos unos pros!!

GioSunrider, 2017-11-13:
Hemos medido al Calicat y sale perfecto, ejes reales con los de la impresora con un error admisible de 0.1.

GioSunrider, 2017-11-13:
Hemos ido a imprimir una pieza de gran tamaño, en concreto un soporte para el ventilador externo que tiene, y nos ha fallado. La causa de esto es que el motor del eje X, el que esta junto al extrusor, pierde paso a la minima que encuentra un roce. Tambien nos hemos percatado del sobrecalentamiento de la Sanguino durante el proceso, por lo que hemos decidido colocar la pieza a medio imprimir para evitar mas accidentes con el ventilador, apuntando hacia la placa como se muestra en la foto. 

[image]

Para la proxima sesion deberiamos revisar el problema del motor ya que si continua no podremos imprimir nada.

David-Estevez, 2017-11-13:
Acabo de caer en la cuenta de lo enrollada que estaba la bobina de PLA morado, y aún así habéis sido capaces de imprimir sin ondulaciones!!! :scream: :scream: :heart_eyes: :heart_eyes:

Siotma, 2017-11-13:
Todo es cuestión de comparar la rigidez del filamento con la de la pieza más flexible o con más holgura.

A todas luces, el problema de las ondulaciones venía dado por la unión tan laxa que tenía el hotend al extrusor, utilizando aquella pieza azul. Ahora que todas las uniones están hechas con tornillos, y bien apretadas, la torsión de la bobina debería flexar la estructura de la impresora (varillas y tal), lo cual es mucho más difícil...

Respecto al problema de que el motor de X pierda pasos... yo creo que es problema de sobrecalentamiento. Deberíamos comprobar si la rigidez del motor es mayor cuando el driver está frío, y si siempre está tan flojo deberíamos volver a revisar la corriente del Pololu. En última instancia, podríamos cambiarlo por otro menos crítico, como el de Z, o comprar uno nuevo.

GioSunrider, 2017-11-13:
Vale, el próximo día podemos empezar por ahí, revisando otra vez todas esas configuraciones. Eso si, lo del bootloader desisto, igual cambiando el micro podemos solucionarlo. Y respecto a la solucion del cambiar por el motor del eje Z, no se yo si seria buena, debido al problema que tambien pierde pasos con bastante facilidad.

David-Estevez, 2017-11-15:
Ya he comprado la LCD que os puse por aquí hace un par de comentarios, y he encontrado el Atmega 1284p que tenía para el upgrade. Os lo llevo esta tarde junto con el programador.

daavm, 2017-11-16:
Bueno pues allá va lo que hemos hecho hoy: 

Nada más llegar le hemos cambiado el atmega que tenía por el 1284p que nos ha traído @David-Estevez. Con eso y el programados USBTiny ISP que también nos ha traído ha sido cuestión de minutos conseguir meterle el bootloader y, por tanto, el Marlin. Ya no es necesario un programador externo para actualizar el firmware de Dédalo.
 
Sobre Marlin, tras pruebas de impresión (fallidas), hemos reducido la velocidad y aceleración del eje Z a 1mm/s. También hemos cambiado algunos ajustes para evitar el error "Thermal runaway" en Configuration_adv.h. Tanto este archivo como el Configuration.h actualizado los subiré en cuanto pueda a la carpeta de Dédalo.

Se ha impreso en Blacky el soporte para el filamento, pero queda algo holgado, por lo que se ha propuesto añadir tuercas (habría que desmontar parte de la impresora), y añadir una varilla roscada también con tuercas a lo que es el soporte como tal, ya que estamos usando una barra que había por allí que sujetar, no sujeta muy bien. 
También hemos impreso, en Hija, un engranaje para la correa del eje X. Hemos visto que el que tiene está un poco desgastado.

Probablemente me dejaré algo, pero como siempre editaré el comentario cuando me acuerde :)

Siotma, 2017-11-16:
Añadir al comentario de @daavm que las piezas elegidas para el soporte de filamento de Dédalo han sido [estas](https://www.thingiverse.com/thing:91724).

Si disponeis de alguna foto de cómo queda con ello montado, subidla! 😉

David-Estevez, 2017-11-17:
Tengo una varilla roscada en el soporte de filamento de mi M-Prime a la que le sobra varilla por todos lados, pero que no he cortado por falta de ganas / medios. Si creéis que lo que le sobra es suficiente para Dédalo me la puedo llevar un día y la cortamos. Así matamos dos pájaros de un tiro.

[image]

GioSunrider, 2017-11-19:
No he visto el soporte que han impreso ni como queda, pero puede ser una buena solución.

daavm, 2017-11-25:
Pues aquí va lo que hicimos el martes y el miércoles. Si alguno de los que estuvo ve que me dejo algo, que lo añada 

El martes nos dedicamos a desmontar parte de la impresora para añadirle (8?) tuercas en las varillas superiores, que sujetarían los soportes del filamento que habíamos impreso la semana anterior. En la foto (gracias @JorFru) podéis ver mas o menos las que pusimos. 

[image]

Después de esto revisamos que todas las varillas estuvieran a la misma altura, e hicimos alguna prueba de impresión. No dio tiempo a mucho más.
El miércoles yo llegué algo más tarde, pero se puso la varilla que trajo @David-Estevez y que cortó @danielo16 (creo). Se puso con tuercas y todo así que el soporte del filamento es bastante rígido ahora mismo. Después nos pusimos a imprimir, y después de algunos problemas (por no cambiar el perfil :D), conseguimos poner a Dédalo a imprimir e imprimió su propio ventilador de capa, el cual quedó bastante bien. 

[image]

[image]

Ese ventilador inicialmente se iba a usar para enfriar el extrusor, así que se propuso buscar otra pieza para imprimir que tuviese 2 conductos, uno para el extrusor y otro para la capa. 
Y creo no dejarme nada de Dédalo 👍

Siotma, 2017-11-27:
Visto que [esto](https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-345018044) no ha cuajado ... La próxima reunión deberíamos hacer una foto en la que se vea toda la impresora para que conste cómo queda el soporte para el filamento.

davidvelascogarcia, 2017-11-28:
Durante la sesión de hoy, se han impreso e instalado 4 sujeciones para facilitar el calibrado de la cama caliente en la impresora Dédalo.
Adjunto captura.

[image]

Figura 1.  Captura

GioSunrider, 2017-11-28:
Hoy a Dedalo le hemos instalado la mejora de calibración de la cama caliente a muelles, con las sujeciones impresas por @davidvelascogarcia, no hemos hecho foto, el próximo día, se la tenemos que hacer. Ademas hemos hecho dos impresiones, la primera, otro gatito de calibración;

[image]

Y una pieza de calibración de primera capa. La pieza es una simple espiral. Para nuestro asombro, una vez hemos sacado la pieza de la cama, nos hemos percatado de la flexibilidad de esta pieza. Como se puede ver en la foto tiene como forma de tela de araña, y hemos probado a ejercer diferentes fuerzas para probar su flexibilidad. 

[image]

[image]

En un rato, voy a subier a dedalo el perfil de impresion a su carpeta, no he marcado lo de limpiar y engrasar los ejes porque no se hasta que punto se ha hecho, invoco a @daavm y a @danielo16  que son los que lo han estado haciendo para que lo marquen ellos si lo creen necesario.

Para la próxima sesión díria de buscar el tensor del eje Y, aunque viendo que esta bien y la calidad de impresión que tiene Dedalo, yo por mi, diría de cerrar ya el issue y abrir otro de mejoras, entre las que añadir dicho tensor. ¡Dedalo esta más que a punto! 😃

GioSunrider, 2017-11-28:
Añadido el Bundle de información a la [carpeta](https://github.com/asrob-uc3m/impresoras-asrob/tree/master/Dedalo) de Dedalo, así como relleñado el readme de información acerca de los archivos de configuración.

Siotma, 2017-11-29:
En la carpeta de Dédalo hay archivos duplicados (.mepr) y obsoletos (.png). Revisadlos cuando podáis y dejad solamente los más actualizados e imprescindibles (creo que el readme refleja bastante bien esto).

Además, he visto que hay un .stl, en caso de querer tener piezas de dédalo en el repo y para evitar que crezcan demasiado y tapen el resto de archivos, yo haría una subcarpeta llamada `parts` para ponerlas dentro.

GioSunrider, 2017-11-29:
He hecho limpieza de la carpeta y he añadido la carpeta `parts` con la pieza que estaba dentro. 
Creo que seria bueno incluir las piezas que se han usado en toda la puesta a punto, para que así, si algún día se rompe alguna, sabemos donde encontrarlas rápidamente. Y creo que no solo para Dedalo, si no para el resto de impresoras, ya que así nos aseguramos de que siempre sea la misma. No se que tal veis esta idea.

Siotma, 2017-11-30:
Me parece una gran idea, creo que incluso puedes abrir una issue dedicada a ella.

GioSunrider, 2017-11-30:
Vale, perfecto, voy al lio! jajaajaja

Siotma, 2017-12-09:
Sería ideal que ahora que ya se pueden configurar nuevos perfiles ( #38 ), y aprovechando que en la última reunión se instalaron los perfiles de Dédalo en el PC de Hija Resurrection, se proceda a instalar a Dédalo en el lugar de Hija Resurrection. De esta manera, ya no será necesario que utilicéis vuestros propios PCs para utilizarla.

Mientras tanto, Hija Resurrection, seguirá en el taller ( #35 ), y cuando funcione se le podría poner una pantalla LCD...

GioSunrider, 2017-12-11:
Vale, me parece perfecto hacer el cambio. una pregunta ¿Hay 3en1en la nave para engrasar a Dedalo y darla ya por operativa?

Siotma, 2017-12-15:
En la nave no hay nada para engrasar, me temo.

GioSunrider, 2017-12-15:
Pues entonces como querais, si quereis podemos comprar 3en1, o si no la dejamos como esta, y cerrar la issue en el momento que la instalemos al PC. Que visto lo visto, diria de ahora mismo dejar a Blacky libre, y colocar en ese lugar a dedalo, ya que todavia hay que investigar en Hija lo del LCD.

Siotma, 2017-12-15:
Me parece bien. No es necesario esperar a hacerlo absolutamente todo para cerrar la issue. Lo principal es que funcione.

GioSunrider, 2017-12-15:
Perfecto, pues ya tenemos algo importante que haces para la semana que viene, asi nos vamos de navidad con las tres impresoras operativas.

GioSunrider, 2017-12-21:
Se ha comprobado que no es necesario ni engrasarlo n iel tensor de la correa Y. Se le ha quitado el polvo y se ha instalado en el PC de Blacky, y a Blacky se le ha dejado independiente, ya que, se le instalo el LCD como se menciona en [#40](https://github.com/asrob-uc3m/impresoras-asrob/issues/40). Tambien se han dado los permisos del PC, se han cargado los perfiles de impresion, se ha configurado repetier y probado a imprimir. De esta manera queda oficialmente a Dedalo como operativa y completamente funcional.

closed 2017-12-21 by GioSunrider

David-Estevez, 2017-12-22:
Ueeeee!! Genial!!! menudo curro que os habéis pegado!! :clap: :clap: :clap: