asrob-uc3m/impresoras-asrob #32: Mejorar la precisión de las impresoras utilizando la compensación en XY

Siotma, 2017-10-13:
@carlosmgomezj y Diego propusieron en la [última reunión](http://asrob.uc3m.es/index.php/Printer_One:_A_Clone_Wars_Story:_Cuaderno_de_Bitacora#9_de_octubre_de_2017) mejorar el ajuste de la compensación en XY para conseguir medidas más precisas para las piezas impresas, lo cual es una gran idea.
Se pueden hacer pruebas por ejemplo con [esta pieza](https://www.thingiverse.com/thing:916214) hasta conseguir que la moneda encaje correctamente.

diegotrap, 2017-11-16:
Veo que este issue va a estar programado para los próximas sesiones, así que me gustaría dar mi opinión.

Yo creo que este tipo de calibración es incorrecta, para los ejes X, Y lo mejor es usar los valores nominales de steps per mm. Es un tema que suele salir de vez en cuando en impresión 3D y hay dos escuelas de pensamiento, pero dejo el vídeo de Thomas Sanladerer donde justifica por qué no es una buena manera de calibrar los ejes: https://www.youtube.com/watch?v=Mbn1ckR86Z8

Siotma, 2017-11-16:
He visto el vídeo entero y ahora entiendo bien tu comentario. Como dices, los pasos por mm no deben ser modificados, y no es nuestra intención hacerlo.

Lo que queremos utilizar es la opción de Slic3r de "Compensación en XY", la cual es mencionada en el vídeo al final como "la forma correcta de realizar la calibración".

Probablemente, el poco texto que hay en el primer comentario te haya conducido a error.

Aun así, muchísimas gracias por el vídeo, he visto que tiene algunos consejos y enlaces muy interesantes! 👍

diegotrap, 2017-11-16:
Vaya, pues ahora yo quiero aprender a usar esa función de Slic3r =)

Siotma, 2018-01-27:
Creo que ahora que Blacky y Dédalo funcionan tan bien, sería muy interesante comenzar a investigar en esta issue.

GioSunrider, 2019-11-15:
Actualmente Blacky no imprime en las medidas correctas porlo que esto podria ser la solucion correcta.

JorFru, 2019-11-16:
Compensación en XY se llama "horizontal expansion" en Cura. Hay ajuste de expansión para la primera capa, que corrige el "pie de elefante", y para el resto de capas, que corrige el tamaño de los agujeros.

Adjunto un pequeño croquis, sacado de https://www.spainlabs.com/foros/tema-Ender-3D

- cubo grande + agujero grande = muchos pasos
- cubo pequeño + agujero pequeño = pocos pasos

- cubo grande + agujero pequeño = poca compensación XY
- cubo pequeño + agujeros grande = mucha compensación XY

Modificar pasos/mm actua en todas las dimensiones de la pieza, es un multiplicador de dimensiones.
Compensación en XY (expansión horizontal) lo que hace es mover las paredes hacia dentro o hacia fuera, el número son milímetros. El signo define si es hacia dentro o hacia fuera.

JorFru, 2019-11-16:
Habría que identificar qué es lo que necesitamos.

closed 2023-09-28 by CarlosME17