asrob-uc3m/impresoras-asrob #33: Investigar el encendido de la nueva fuente de alimentación de Blacky

Siotma, 2017-10-13:
La nueva fuente de alimentación de Blacky (instalada en #28 ) parece requerir un mínimo de carga para arrancar. Se debe comprobar cuál la resistencia necesaria (probablemente lo sean en el canal de 5V, según he encontrado en [el paso 10 de este tutorial](https://makezine.com/projects/computer-power-supply-to-bench-power-supply-adapter/)) e instalarla para que la fuente sea fácilmente utilizable. 

La instalación de la resistencia debería realizarse preferiblemente de manera interna para reducir las posibilidades de rotura y al mismo tiempo facilitar la ventilación.

Solo para cerciorarme, @David-Estevez @jgvictores , ¿habría posibilidad de obtener una resistencia de potencia de 10ohm / 10W de la asociación o del departamento? ¿O debería abrir una issue para comprarla?

Siotma, 2017-10-13:
Cuando se realicen estas pruebas, sería interesante replicarlas en la otra fuente de alimentación idéntica a la de Blacky que hay en el armario. Una vez se haya resuelto esta incidencia, el siguiente paso será colocar la nueva fuente en Hija Resurrection.

David-Estevez, 2017-10-13:
He estado mirando en RS y no son muy caras, yo creo que sí que sería posible conseguir una del departamento. Pero que nos confirme @jgvictores a ver qué le parece a él.

JorFru, 2017-10-15:
Yo dono las resistencias.
Serán dos de 4R7 10W

No puedo comprometerme a llevarlas la próxima reunión, pero será pronto.

JorFru, 2017-10-15:
Llevo cuatro. Dos para cada máquina.

Siotma, 2017-10-17:
Hoy se han realizado pruebas con la fuente de alimentación, aunque no se ha dispuesto de las resistencias de potencia que prometió @JorFru.

En su lugar, se ha probado con resistencias de 470ohm / 1W disponibles en el armario. Se han realizado pruebas colocando estas resistencias en los distintos canales de la fuente de alimentación, incluso se han conectado varias en paralelo, pero sin éxito.

La mayor carga que podemos conectar a la fuente ahora mismo es la propia impresora, y se ha determinado un comportamiento curioso:
- Tras largo tiempo sin usarla si se intenta encender la fuente mediante el interruptor incorporado, ésta no se inmuta, permaneciendo apagada.
- Si se activa el calentador del extrusor, tras varios intentos de activación - desactivación la fuente acaba arrancando.
- Si se activa el calentador de la cama caliente en este momento, se puede comprobar cómo el relé no engancha y empieza a sonar.
- Tras varios intentos de activación y desactivación de la cama caliente, al fin el relé engancha y la impresora es funcional.
- Una vez alcanzado este punto, no pasa nada por apagar y encender la fuente en un corto espacio de tiempo, ya que la fuente de alimentación seguirá funcionando correctamente.

Da por tanto la sensación de que la fuente tiene algún tipo de memoria que condiciona su comportamiento. Por tanto, **para realizar una correcta resolución de esta Issue, se debe valorar el arreglo con un largo tiempo de inactividad de por medio.**

Finalmente, leyendo algunos foros, he encontrado una idea provisional para poner una carga en la fuente: conectar un disco duro o una unidad de CD. Este tipo de unidades deberían demandar suficiente corriente como para que la fuente de alimentación arranque. Se puede añadir esta ide a la idea de la resistencia previamente comentada.

JorFru, 2017-10-17:
Hay disponibles ahora mismo dos resistencias en la caja de Printer One,
dentro del armario.

Una de 47Ohm 10W, pensada para instalarse en el raíl de 12V
Una de 10Ohm 10W, pensada para el raíl de 5V.

Acabo de acordarme de que esa fuente lleva dos raíles de 12V, igual hace
falta otra de 47. Probaremos.

PD: despiste remediado y promocionado por Jupi, su tienda de desconfianza.

Siotma, 2017-10-18:
Gracias, JorFru!

Siotma, 2017-10-24:
En la reunión de ayer estuvimos haciendo más pruebas con esta fuente de alimentación. A continuación voy a intentar exponer los descubrimientos más importantes, y en mi opinión, más fiables. 

En primer lugar quiero incidir en los dos problemas conocidos que tiene actualmente esta fuente:

1. No sabemos exactamente qué requiere la fuente de alimentación para encenderse.
2. Una vez encendida, la fuente de alimentación no siempre es capaz de entregar potencia suficiente para alimentar la cama caliente, provocando un sonido de timbre en el relé.

Durante la reunión de ayer realicé innumerables pruebas, casi todas infructuosas, sin embargo saqué algunas conclusiones y hoy he vuelto 10 minutos esta mañana para contrastar los resultados de ayer. Son los siguientes:

- La fuente de alimentación de Blacky no necesita ninguna carga para encenderse.
- Para encenderse, la fuente parece requerir un orden específico de acontecimientos:
1. Encendido del interruptor de la fuente. El puente entre el cable verde y negro del [conector ATX](https://es.wikipedia.org/wiki/ATX) **no** debe estar hecho.
2. Espera de un tiempo, unos 30 segundos.
3. Conexión del puente entre el cable verde y negro. La fuente debe arrancar y el ventilador empezará a girar. 
4. Si no arranca, es que el tiempo de espera ha sido corto. Sacar el puente, esperar otro poco y volver al paso 3.

*Este comportamiento concuerda con un uso habitual de la fuente de alimentación, ya que cuando está instalada en un PC, la fuente siempre tiene el interruptor activado y es el cable verde del botón de encendido el que la activa. Por tanto, es posible que la fuente necesite un tiempo para prepararse hasta que pueda entregar potencia, y que hasta que no esté preparada, el cable verde no funciona, o incluso puede que _evite_ que funcione.

*No lo he comprobado, pero es posible que el momento en que la fuente pueda ser encendida, coincida con un valor correcto en el cable violeta (5Vsb) del [conector ATX](https://es.wikipedia.org/wiki/ATX).

*Sí he comprobado que una vez la fuente ha arrancado, el valor de cable gris (Power OK) del [conector ATX](https://es.wikipedia.org/wiki/ATX) toma un valor de 5V.

- Una vez en marcha, siguiendo el proceso anterior, se debe probar a calentar la impresora, pueden ocurrir diferentes escenarios:
1. Si se intenta calentar solamente el extrusor, es posible que lo haga, aunque también es posible que la fuente se apague al detectar un pico de consumo muy alto.
2. Si se intenta calentar solamente la cama caliente, seguramente la fuente se apague al detectar el pico de consumo.
3. Es posible que al intentar calentar la cama caliente, el relé suene como un timbre. Esto se produce porque la fuente de alimentación en vez de apagarse, se queda reiniciándose continuamente debido al súbito pico de consumo que supone la cama caliente. Este escenario debe evitarse rápidamente si se produce ya que puede dañar la fuente.
4. Si se intentan calentar ambas partes simultáneamente, puede ocurrir cualquier combinación de lo anterior.
5. Si el relé de la cama caliente se queda fijo, es señal de que la protección de pico de corriente no ha saltado y que la fuente sigue encendida.
6. Para comprobar que la fuente es capaz de alimentar correctamente la impresora, se debe verificar que puede calentar simultáneamente el extrusor y la cama (sin que el relé suene como un timbre), y además mover los motores. Para un comprobación más fiable, se puede incluso imprimir una pieza pequeña. 

*Sin embargo, el único escenario que debería darse es el número 6, evitando todos los anteriores, por ello se realizaron pruebas

- Se ha comprobado que se puede propiciar el escenario 6 colocando cargas en los raíles de la fuente, concretamente:
1. 10ohm en el raíl de +5V
2. 47ohm en el raíl de +12V2 con cables negros y amarillos
3. 3 resistencias en paralelo de 470ohm en el raíl de +12V1 con cables totalmente amarillos
4. Un lector de CD conectado al conector de alimentación SATA que utiliza tensiones de +5V, +3,3V y +12V1

*Es posible que el lector de CD no sea necesario si se coloca una carga mayor en el raíl de +12V1. Sin embargo, el lector es útil para conocer el comportamiento dinámico de la fuente ya que se abre y se cierra cada vez que la fuente se reinicia.

Este lector de CD se encuentra en la balda de Printer One, y las resistencias expuestas anteriormente se encuentran en la caja de recambios.

Además, abrí la fuente de alimentación hermana de esta y estuve haciendo peluquería de cables, aunque soldé el cable verde al negro antes de saber que esto iba a ser un problema, ahora mismo esta fuente no puede arrancar, lo que concuerda con el proceso de encendido expuesto al principio del mensaje. Lo que sí saqué en claro cuando abrí la fuente y quité unos cuantos cables, es que los raíles de 12V1 y 12V2 son efectivamente independientes, por lo que para debe utilizarse una resistencia para cada uno.

Ruego que en la reunión de hoy comprobéis el contenido de este comentario, pero aún no realicéis modificaciones irrecuperables en el cableado de la fuente de alimentación de Blacky. Postead las conclusiones y el próximo día, tras un triple check, la metemos mano.

Recuerdo el hecho de que una vez que la fuente ha arrancado del todo, es más fácil volver a hacerla arrancar, de modo que una vez que esté en marcha las pruebas que se hagan no serán tan fiables como las primeras. Por ello recomiendo que tras haberse pegado un poco con esto, se de paso a #32

Siotma, 2017-10-24:
A continuación, un poco de imagen a tanto texto.

Lector de CDs utilizado:

[image]

Durante la peluquería de la fuente de alimentación que tiene sonido electrónico, únicamente aprecié a nivel visual un condensador hinchado:

[image]

Finalmente, también pude comprobar cómo la fuente de alimentación tiene espacio suficiente en el interior para colocar las resistencias de carga que decidamos instalar finalmente:

[image]

David-Estevez, 2017-10-24:
@Siotma  Me quito el sombrero ante tu capacidad de documentación... :tophat:

JorFru, 2017-10-25:
Leñe!
Estuve leyendo todas las issues mientras comía, pero no leí esta, es posterior.
Lamento comunicaros que: Hemos perdido el tiempo realizando las comprobaciones que tú habías hecho por la mañana.

Sin embargo, algo más en claro habremos sacado, esperemos...
Hemos hecho prácticamente todo lo que tú has probado.
Resistencias de carga de la siguiente manera:
- 10Ohm en 12V1
- 10Ohm en 12V2
- 4.7Ohm en 5V

Como diferencia no hemos instalado la disquetera, aunque la carga resistiva era mayor que la de Manu.
Pues bien, no hemos notado ninguna mejoría respecto a que el ventilador gire cuando se lo pides, ni tampoco a la fiesta que se monta el relé.

PERO! Ahora que te leo, a posteriori y con detenimiento, todo me encaja.
La fuente se está reseteando continuamente. Ideas para solucionarlo:
1. Instalar una resistencia en serie con la cama para restarle potencia. No me convence, poco eficiente.
2. Instalar una bobina en serie con la cama. Lo cual aporta alta impedancia durante la avalancha de corriente al encender, y no lastra el rendimiento de la cama en régimen estacionario. Contras: desconocemos el comportamiento de la fuente frente a cargas inductivas (no debería de comportarse mal en teoría). posiblemente la bobina a instalar sea bastante grande queda pendiente calcular el tamaño.

@davidvelascogarcia estuvo conmigo y quizás pueda aportar algo que se me escape.

JorFru, 2017-10-25:
Le he dado más vueltas al asunto.
Ayer comprobé con el osciloscopio lo que pasaba en el raíl de 12V de la fuente cuando el relé oscilaba. No tengo foto de ello. Pero era un pico de 10-11V seguido de silencio a 0V. Por supuesto al mismo ritmo del clickeo.

Me percaté de que el LED que lleva la RAMPS y que indica que la salida del MOSFET está activada también parpadeaba al mismo ritmo. ¡Entonces empecé a pensar que era el propio Arduino que le mandaba encenderse y apagarse todo el rato! Pero acabo de buscar el esquemático de la RAMPS y me he heshecho de esa idea.
[image]
El Arduino se alimenta con el ordenador, no se va a apagar durante el reseteo de la fuente, y el mosfet que activa el relé funciona correctamente.
La luz se enciende cuando el mosfet cierra el circuito, sin embargo, cuando no hay 12V en la fuente, el LED no se puede encender ni aunque el mosfet esté cerrado. Esto explicaría este funcionamiento.
El contactor del relé tiene un muelle que fuerza a la posición abierta el relé, desconecta la cama y la fuente se vuelve a encender pensndo que el peligro ha pasado.
La fuente se resetea porque a priori ve un cortocircuito.
Esto en cuanto al problema del clickeo.

Sobre el problema de que a veces le cuesta arrancar, leyendo lo que dice David y Manu, la fuente tiene que estar conectada durante unos segundos antes de estar peparada para puentear el cable verde.
Sugiero instalar un interruptor nuevo qu puentee el cable verde y sirva para encender la fuente fácilmente. El interruptor que corta la alimentación AC se puede quedar siempre activado por comdidad o se puede desactivar cada vez, pero anotando el "modus operandi" para los no informados.

Sobre el tema de si son necesarias o no la cagas resistivas (o lector de discos) todavía no está comprobado. Pero estando al tanto de que el osciloscopio de ASROB funciona bien, se podría comprobar el rizado de la alimentación en ambos casos. Lo propongo para la siguiente reunión.

JorFru, 2017-10-25:
> @JorFru <https://github.com/jorfru> , totalmente de acuerdo, poco más
> puedo aportar. Lo único, @Siotma <https://github.com/siotma> , comentar
> que descubrimos una manera de encender la fuente, conectar y hacer
> funcionar la impresora, con pasos comunes a los realizados en la sesión del
> 23/10/2017.
> Para encender la fuente, como comentas, en caso de que no arranque a la
> primera, desconectar el puente y volverlo a poner, y funciona sin problema.
> Respecto al problema del relé, nos dimos cuenta, que invirtiendo el orden
> de encendido, no da ningún problema, me refiero, a encender primero el
> extrusor, y posteriormente la cama, habiendo esperado al menos que el
> extrusor alcance los 45-50ºC.
> Es un tanto curioso, que únicamente por encender uno antes que el otro
> salte o no.

El gran Iceflow, Aisflow, me comenta por el canal de Telegram de
Staticboards que es muy recomendable instalar un diodo en serie con la
cama. Así lo monta la RAMPS1.4SB de diseño nacional.
No garantiza un arranque más suave, sin embargo garantiza que la corriente
no circule en el sentido contrario, lo cual puede dejar KO la fuente. Y
esto puedo ocurrir si la carga (la cama) tiene algo de componente
inductiva, lo cual seguro que es cierto viendo que el circuito de la cama
caliente parece un pantallazo del Snake en el último nivel.
Igual con  esta solución deja de reiniciarse.
Se probará en la siguiente reunión.

Saludos!

Siotma, 2017-10-25:
Veo que en líneas generales estamos todos de acuerdo. Comentarios:

@davidvelascogarcia dijo:
> Para encender la fuente, como comentas, en caso de que no arranque a la primera, desconectar el puente y volverlo a poner, y funciona sin problema.

Hurra, hemos conseguido encenderla!

@davidvelascogarcia dijo:
>  invirtiendo el orden de encendido, no da ningún problema, me refiero, a encender primero el extrusor, y posteriormente la cama, habiendo esperado al menos que el extrusor alcance los 45-50ºC.
Es un tanto curioso, que únicamente por encender uno antes que el otro salte o no.

Me alegro de que hayais encontrado esta situación. Desde mi punto de vista no es en absoluto curioso, sino que puede explicarse con el mismo motivo por el que yo identifiqué que la fuente se apaga súbitamente cuando pides calentar. Debe tener algún tipo de protección contra picos de corriente, o si no, un apagado automático cuando detecta que no es capaz de dar la tensión nominal (ie detección de cortocircuito). Teniendo en cuenta que el extrusor demanda unos 3A y la cama unos 10A, como mínimo; si la fuente arranca en vacío, un pico súbito de 0->10A le puede ser más difícil de realizar que si es de 3->13A.

Por eso yo tengo fe en que al usar resistencias de carga que demanden una cantidad notable de corriente, estos escalones desde vacío se eliminen y funcione siempre, aunque según dijo @JorFru:

> 10Ohm en 12V1; 10Ohm en 12V2; 4.7Ohm en 5V 
Como diferencia no hemos instalado la disquetera, aunque la carga resistiva era mayor que la de Manu.
Pues bien, no hemos notado ninguna mejoría respecto a que el ventilador gire cuando se lo pides, ni tampoco a la fiesta que se monta el relé.

No parece que las resistencias eliminen el problema.

@JorFru dijo:
> El contactor del relé tiene un muelle que fuerza a la posición abierta el relé, desconecta la cama y la fuente se vuelve a encender pensndo que el peligro ha pasado.

Eso mismo pienso yo. Al encenderse el peligro regresa y entra en bucle.

@JorFru dijo:
> Sugiero instalar un interruptor nuevo qu puentee el cable verde y sirva para encender la fuente fácilmente. El interruptor que corta la alimentación AC se puede quedar siempre activado por comdidad o se puede desactivar cada vez, pero anotando el "modus operandi" para los no informados.

Yo también había pensado en colocar un nuevo interruptor. En mi opinión el que ya hay se puede dejar encendido y tapado con una pieza de plástico. En caso de emergencia se tira del cable y listo.
También colocaría un par de leds de estado para simplificar su uso si confirmamos la función de los cables violeta y gris:
- Led "Ready" rojo de baja luminosidad conectado al cable violeta del [conector ATX](https://es.wikipedia.org/wiki/ATX), que indicaría que la fuente está preparada para encenderse.
- Lez "ON" verde de alta luminosidad conectado al cable gris del [conector ATX](https://es.wikipedia.org/wiki/ATX), que indicaría que la fuente está funcionando correctamente y las tensiones están en rango.

@JorFru dijo:
> es muy recomendable instalar un diodo en serie con la cama.

En serie me parece raro, ya que debería admitir más de 10A y un diodo de 10A es muy gordo y se calienta con unos `10A * 0,7V = 7W 😨` . Imagino que te refieres a un diodo en antiparalelo para que descargue la cama al apagarse. Sin embargo, es posible que el relé ya disponga de un diodo de este tipo, ya que los relés suelen tener diodos para evitar chispazos cuando conectan y desconectan motores u otras cargas inductivas.

Siotma, 2017-10-25:
En mi opinión, la idea del interruptor en el cable verde unida a la bobina en serie con la cama más diodo en antiparalelo son nuestra mejor baza. Aunque me gustaría prescindir de la bobina en serie, no sé si vamos a poder prescindir de ella...

David-Estevez, 2017-10-25:
Pregunta tonta... la carga inductiva que hace saltar la fuente es la cama o la bobina del relé? el relé tiene actualmente algún diodo de flyback?

Siotma, 2017-10-26:
A eso me refiero con diodo en antiparalelo. Es cierto que el relé también es una carga inductiva, aunque está activado por un mosfet así que la fuente no lo tan directamente como la cama. Aun así, revisaremos ambas cosas.

JorFru, 2017-10-26:
Esto dice Javier de Staticboards

"@JorFru el diodo hace que la carga electrica tenga un camino de retorno y
no salga de la cama caliente
se queda atrapada dentro de la cama caliente hasta que desaparece."

Con tranquilidad asimilaré estas palabras.

Siotma, 2017-10-29:
Eso me suena a diodo en antiparalelo.

jgvictores, 2017-10-29:
Sólo viendo estas referencias en diagonal...

- https://en.wikipedia.org/wiki/Flyback_diode
- https://en.wikipedia.org/wiki/Antiparallel_(electronics)

...diría que la terminología correcta es la que usa @David-Estevez (flyback). Yo lo conocía como freewheeling, que esa entrada de wikipedia pone como sinónimo. La intención y el efecto será parecido, sólo que antiparalelo tiene más sentido con algo que tiene polaridad (p.ej. un transistor), y un poco menos en una carga inductiva tipo bobina.

Siotma, 2017-10-29:
Comprendo, y coincido contigo @jgvictores 👍 

Al final, todos nos referíamos a lo mismo, pero es más correcto llamarlo por su nombre ☺️

Siotma, 2017-10-30:
Hoy no hemos conseguido que el relé enganche en toda la mañana. 

Eso ha podido ocurrir porque se ha partido una de las patas de una de las resistencias y no hemos podido utilizarla... Además, tampoco tenemos diodos de potencia en el armario, que yo pensaba que sí.

En la próxima sesión se podrán seguir haciendo pruebas, pero cada vez me desmotiva más esta fuente de alimentación ya que está lastrando el uso de Blacky.

En caso de mantenerse esta situación habrá que cambiar otra vez la fuente de alimentación.

Siotma, 2017-11-03:
Ayer por la mañana me pasé por la nave e hice lo siguiente:

1. Quitar el puente de la fuente de alimentación.
2. Encender el interruptor de la fuente de alimentación.
3. Esperar mucho tiempo, al menos 20 minutos.
4. Poner el puente. La fuente arrancó a la primera.
5. Calentar el extrusor.
6. Cuando el extrusor iba por 50 grados, puse a calentar la cama.

La impresora calentó maravillosamente, sin saltar el relé ni una vez:

[image]

Seguiremos investigando.

Siotma, 2017-11-10:
Ayer en la reunión de Printer One, hice lo siguiente:

1. Quitar el puente de la fuente de alimentación.
2. Encender el interruptor de la fuente de alimentación.
3. Esperar mucho tiempo, al menos 30 minutos.
4. Poner el puente. La fuente arrancó a la primera.
5. Encender el ordenador.
6. Conectar Repetier a la impresora y poner a calentar la cama.

El relé enganchó a la primera ante mi atónita mirada. @javii9 y yo imprimimos una pieza para celebrarlo:

[image]

Por tanto, la teoría de que la fuente necesita estar encendida largo tiempo para "prepararse", "cargar condensadores", "inicializarse", o lo que sea, cobra mucha fuerza.

Así pues, @javii9 y yo hemos hecho un poco de bricolaje a la fuente con intención de que este proceso sea más sencillo para el usuario:

1. Hemos soldado un interruptor entre el cable verde y negro.
2. Hemos soldado unos leds (con sus resistencias de 680ohm):
- Led azul al cable violeta (5Vsb). Este led se enciende mientras la fuente esté enchufada y con el interruptor principal encendido. Tarda unos segundos en encenderse.
- Led verde al cable gris (Power Ok). Este led se enciende cuando la fuente está en marcha.

La idea es que la fuente quede finalmente con el interruptor principal permanentemente encendido y con una pieza de plástico evitando que nadie lo toque. El interruptor del cable verde puesto ayer, estará montado de forma firme en la fuente para que cualquiera que lo vea entienda que es ese el que debe usarse. Además, los leds darán en todo momento feedback del estado de la fuente, lo cual teniendo en cuenta los problemas que nos ha dado, sería muy útil.

Merece la pena decir que no es buena idea soldar los leds porque se estropean con el calor, así que habrá que crimparlos en un futuro si los queremos dejar puestos.

Al irnos, por precaución, ayer dejamos ambos interruptores apagados, de modo que para encender la fuente el próximo día habría que encender primero el general y un rato después el nuevo. Si la próxima sesión somos capaces de replicar el proceso de encendido, haremos permanentes los cambios.

Y ahora, unas fotitos:

Interruptor apagado: modo ready con led azul encendido.

[image]

Interruptor encendido: modo on con ambos led encendidos.

[image]

JorFru, 2017-11-13:
Genial trabajo.
Muy buena documentación

Siotma, 2017-11-13:
En la reunión de hoy, ha sido totalmente imposible que enganche el relé, así que seguimos sin saber cómo usar la fuente. Menuda decepción, a este paso, tendremos que desechar esta y utilizar otra con menos potencia...

Siotma, 2017-11-16:
En la reunión de ayer, tras varios intentos, el relé enganchó y se imprimieron unos soportes para filamento para Dédalo.

[image]

Siotma, 2017-11-23:
Ayer, sustituimos la fuente de alimentación por una distinta que había en el armario, con las siguientes características:

[image]

El canal de 12V2 (13A máx) se conectó a la cama caliente mediante bananas de 3,5mm que llevé para la ocasión.

El canal de 12V1 (8A máx) se conectó a:

- Las clemas que alimentan la RAMPS
- Los ventiladores de la RAMPS (mediante bananas de 2mm cogidas del armario)
- El ventilador del extrusor (mediante bananas de 2 mm cogidas del armario)

Esta nueva fuente demostró funcionar correctamente, tal y como muestra la siguiente curva de calentamiento:

[image]

La curva es muy similar a la alcanzada con la otra fuente (ver https://github.com/asrob-uc3m/impresoras-asrob/issues/28#issuecomment-332919775), por tanto, la diferencia de potencia no parece ser significativa.

Para encender esta fuente únicamente hay que activar el interruptor que lleva incorporado. El puente entre el cable verde y negro está hecho, soldado y termorretractilado.

Imprimimos una pieza de prueba correctamente:

[image]

La fuente no ha dado problemas de encendido ni de saltos del relé, por tanto, está totalmente operativa. Se pide encarecidamente realizar impresiones con ella para confirmar que está reparada.

En estos momentos, Blacky es totalmente funcional, y presenta el siguiente aspecto.

[image]

closed 2017-11-23 by Siotma

Siotma, 2017-11-29:
En la reunión de ayer se imprimieron piezas durante toda la tarde sin ningún problema.