asrob-uc3m/impresoras-asrob #34: Smooth con baño de acetona.

carlosmgomezj, 2017-10-17:
En esta sección se testeará con diferentes métodos de realizar el baño de acetona con las piezas de abs.
Principalmente se usará el suavizado por medio del contacto del vapor de acetona en vez de contacto directo con ésta.

carlosmgomezj, 2017-10-17:
En la primera sesión se ha usado un bote de conservas normal de cristal como contenedor para el baño, hay que tener cuidado al ser la base convexa de fijar bien la base a la que se colocará la pieza.
Este método ha consistido en:
1: Rellenar el fondo del bote a una altura de unos 5mm con Acetona industrial líquida.
2: Colocación de una base metálica para aislar la pieza del contacto directo.
3: Poner la pieza 
4: Tapar el botte.
5: Esperar....y esperar.

Durante el proceso se ha observado que el vapor de acetona ataca de manera mucho más significante la mitad inferior de la pieza, por lo que se ha dado la vuelta para igualar resultados. Pasado un tiempo se cayó accidentalmente la pieza dentro de la acetona, convirtiéndolo en un verdadero "baño de acetona", se ha sacado rápidamente la pieza y dejado secar. 
La suerte ha hecho que el Link (pieza en ABS) haya quedado al límite en lo que a observabilidad de detalles se refiere.

En la próxima sesión se probará a meter la pieza directamente en acetona en vez de esperar tanto tiempo.
[image]
[image]

carlosmgomezj, 2017-10-17:
ERROR!!

Despues de que @diegodevigo  imprimiera un par de piezas diseñadas por él mismo asemejando a dos "LEGOS" que no encajaban para nada, se ha intentado hacer un  suavizado con acetona metiéndo la pieza directamente y luego sacarla para el secado.

No hay ningún tipo de control fino con éste método y ha ocurrido además el problema de que la acetona se ha quedado atascada en algunos huecos interiores, derritiendo todo el ABS a su paso.

El resultado final ha  sido pésimo y se ha confirmado que tiene que ser siempre que se pueda con vapor.

carlosmgomezj, 2017-10-17:
Hoy se ha aplicado un nuevo método, siguiendo las instrucciones de este video: [Smooth ABS](https://www.youtube.com/watch?v=6xFUNFG-UKE)
Consiste en que la acetona esté en la parte de arriba de la pieza, por lo que el vapor ( más pesado que el aire) bajará hasta abajo sin salir gracias a una capa de agua en la base.

Se ha realizado con el mismo bote de vidrio de las pruebas anteriores colocando papel de cocina en la parte inferior y dándole la vuelta. Después de aproximadamente dos horas de contacto con el vapor ( no estaba herméticamente cerrado) el resultado ha sido mejor del esperado.

Una cosa a tener en cuenta es que el proceso termina bastante después del paso de sacarlo del bote, con la precaución importante de no tocar la pieza ya que en ese momento es fácilmente moldeable o sensible a huellas, etc.

En la siguiente sesión se intentará hacer una base parecida a la del video con un ventilador que haga que sea más uniforme el suavizado y más rápido.

En las siguientes fotos se ve como el baño de acetona ha ayudado además a disimular la cicatriz de unión entre las dos mitades del Pokémon:

[image]
[image]
[image]

Siotma, 2018-04-21:
Cerrando por inactividad. Si se retoman las actividades con acetona se podrá reabrir.

closed 2018-04-21 by Siotma