asrob-uc3m/impresoras-asrob #35: Hija Resurrection necesita una nueva fuente de alimentación

Siotma, 2017-10-26:
Creemos que la solución a #26 pasa por aumentar la potencia de alimentación de Hija Resurrection. Vistas las dificultades encontradas en #33 con la nueva fuente de alimentación de Blacky, y teniendo en cuenta que la fuente hermana de esta hace ruido electrónico y no nos podemos fiar de ella para la impresora, propongo, aunque sea de forma temporal:

# Instalar una segunda fuente de alimentación en Hija Resurrection.

Mi idea es que la nueva fuente únicamente alimente el secundario del relé para calentar la cama cuando la Ramps lo requiera. De esta manera se libera a la fuente actual de más de 100W, por lo que el resultado debería ser un calentamiento más ágil y la desaparición de errores de Thermal Runaway.

Para esta actuación se puede utilizar una fuente de alimentación que @Morgloz dejó casi preparada para Blacky en su día. Está en el armario en la balda inferior a la izquierda. Creo que esta issue se puede resolver en una sola sesión y ganaríamos una impresora que funciona y está calibrada, pero que ahora mismo no estamos utilizando en absoluto por los problemas de alimentación que tiene.

Siotma, 2017-11-13:
En la reunión de hoy, se ha instalado una nueva fuente de alimentación en Hija Resurrection.

Se trata de una fuente de alimentación con 8A para el canal 12V1 y 14A para el canal 12V2.

He conectado el canal 12V2 a la cama caliente de Hija Resurrecion utilizando bananas de 3,5mm disponibles en el armario.

La fuente antigua de Hija Resurrection aún alimenta el resto de elementos (incluido el relé), por tanto ahora hay dos fuentes en Hija Resurrection.

La nueva fuente de alimentación carece de interruptor y por tanto para encenderla es necesario poner el puente.

Además, se ha modificado la configuración de Marlin vía 83d843e02ac85237739cad3ec4fa42780e4ceed3 para aumentar la frecuencia de refresco de la temperatura de la cama.

Se han realizado dos pruebas de impresión, la primera de un cubo, satisfactoria:

[image]

La segunda prueba de impresión de una pieza de calibración, ha sido insatisfactoria por desplazamiento del eje Y:

[image]

Sin embargo, en ninguna de las dos ocasiones ha saltado el error de Thermal Runaway, y a la luz de las curvas de temperatura parece que también se ha corregido la falta de potencia:

[image]

Únicamente es notable un escalón en las medidas de temperatura en el momento en que comienza la impresión. Se me ha ocurrido que el momento en que empiezan a moverse los motores, haga caer la potencia de la fuente antigua, y que eso tumbe las referencias de los ADCs. Esto además concuerda con el desplazamiento del eje Y...

Seguiremos haciendo pruebas. Aunque, al menos parece, que hemos cerrado #26

Siotma, 2017-11-13:
Aunque esta impresora continúa teniendo la etiqueta "NO FUNCIONA", con los resultados de hoy estaríamos cerca de tenerla operativa. 

Lo que necesita ahora es, sin duda, **pruebas de impresión**. Así que... AL TURRÓN!

Siotma, 2017-11-16:
En la reunión de ayer se realizaron nuevas pruebas de impresión: un cubo y una polea dentada, ambas satisfactorias.

[image]

Además, adjunto una imagen de la configuración actual de la impresora con las dos fuentes de alimentación. Ayer utilizamos un carrete vacío que llevé para enrollar un poco de filamento amarillo que tiene bastante mejor calidad de impresión que el negro que había antes.

[image]

Siotma, 2017-11-29:
En la reunión de ayer, se comenzó el proceso para soldar conectores a la nueva fuente de alimentación para dejar a Hija Resurrection con una sola de ellas. Sin embargo, no dio tiempo a completar el proceso por lo que actualmente se encuentra sin fuente (están en el armario, en la balda de Printer One).

Siotma, 2017-11-29:
Asimismo, se realizó una revisión minuciosa de los sensores de temperatura que incluyó un re-soldado, y re-termorretractilado del termistor del extrusor, y la colocación de una pieza de cartón bajo la cama caliente para disminuir la disipación. El sensor de temperatura de la cama caliente pareció estar en buen estado y por tanto no fue necesario realizar ninguna operación sobre él.

Relacionado con esto, los cables de ambos sensores de temperatura estaban realmente tirantes y puede que ello fuera la causa de los problemas de lectura de temperatura. Se descubrió que estos cables se encontraban enrollados adrede para reducir su longitud, aunque tal vez demasiado. Se ha dado un poco más de juego a los cables de ambos sensores para reducir su tensión y las posibilidades de desconexión fortuita y/o rotura.

Siotma, 2017-12-02:
Ante la posibilidad de necesitar las impresoras para formar nuevos operadores la próxima semana, ayer me acerqué un rato a la nave para finalizar la instalación de la nueva fuente de alimentación de Hija Resurrection y así dejarla operativa. Sin embargo, aunque parecía sencillo en un principio no resultó serlo tanto:

- Tras conectorizar la fuente de alimentación y conectarla a la impresora, realicé una prueba de calentamiento insatisfactoria: el calentador del extrusor no funcionaba, como ya pasó en otra ocasión. Dado que pretendemos que la impresora sea fiable y aprovechando que teníamos otro calentador nuevecito en el armario, lo cambié.

- Para cambiarlo, me tocó desmontar el extrusor puesto que el prisionero que afianza el calentador era inaccesible ya que estaba orientado hacia arriba. Finalmente, monté el extrusor de nuevo, dejando el prisionero orientado hacia abajo para facilitar su accesibilidad en un futuro.

- Realicé una nueva prueba de calentamiento de cama y extrusor, satisfactoria, aunque el extrusor calentaba muy lentamente.

- Lo más preocupante vino cuando intenté mover los motores: al moverse, provocan una caída de potencia considerable en el resto de la impresora, y mientras están alimentados, la fuente de alimentación emite un ruido poco recomendable. Incluso, si se insiste mucho con el movimiento, la fuente acaba apagándose, imagino que por seguridad.

Con la información de los dos últimos puntos, me he dado cuenta de que nunca hemos evaluado la calidad de la electrónica de esta impresora: calidad de la RAMPS, calidad de los MOSFET  y calidad de los pololu, principalmente. Sospecho que unos MOSFET de baja calidad pueden provocar un calentamiento lento e ineficiente del extrusor y que si alguno, de los pololu estuviera defectuoso podría interferir en el funcionamiento del resto de los componentes.

Dado que tenemos en el armario otra RAMPS nuevecita de BQ (tiene 5 pololu y hasta pantalla LCD), he pensado que podemos evaluar la posibilidad de sustituir la actual por esa y ver si cambia el comportamiento de la impresora. Tengo confianza en que el cambio mejore globalmente el funcionamiento de nuestra querida Hija Resurrection. En dicho caso, además, podríamos dejar preparada la impresora para utilizarla con pantalla LCD y liberar un PC para Dédalo, por ejemplo.

Siotma, 2017-12-06:
En la reunión de ayer comprobamos que el problema de potencia no estaba relacionado con la RAMPS sino con la fuente de alimentación: la tensión cae enormemente cuando se enciende la cama caliente y ello perjudica el funcionamiento de los drivers paso a paso provocando fallos en el movimiento.

Se cambió la fuente por otra igual pero con mejores valores de tensión de salida. Asimismo se observó que el funcionamiento de la fuente mejoraba añadiendo una resistencia de 10Ohm al canal de 5V, por lo que se ha dejado conectada internamente.

Ahora mismo, Hija Resurrection dispone de una fuente de alimentación operativa y está lista para realizar pruebas de impresión.

Siotma, 2017-12-14:
Hoy, he realizado un agujero en el bloque calentador de Hija Resurrection para fijar el termistor con un tornillo y prescindir del Kapton. Ha quedado de maravilla:

[image]

Además, @danielo16 ha realizado varias pruebas de impresión satisfactorias. Los problemas de potencia han desaparecido tal y como muestra la siguiente curva de calentamiento. Se puede comprobar cómo la curva de calentamiento del extrusor es mucho más estable al estar el termistor bien sujeto.

[image]

Hija Resurrection ha vuelto a la vida, valga la redundancia 🤣 

Issue cerrada

closed 2017-12-14 by Siotma

JorFru, 2017-12-15:
Mera curiosidad,
Me imagino que el agujero lo has hecho con las nuevas herramientas
rotativas de ASROB
Pero cómo has hecho la rosca? Tenemos machos de rosca y yo no lo sabía?

Siotma, 2017-12-15:
El agujero con la rosca ya estaba hecho. Curiosamente este tipo de agujero en todas las impresoras que yo he visto viene roscado. Yo hice uno un poco más abajo para introducir el termistor. Y ese no necesita rosca.

JorFru, 2017-12-16:
Todo entendido!
Gracias Manu