asrob-uc3m/impresoras-asrob #36: Mejorar la calidad de nuestras impresiones

JorFru, 2017-11-13:
Creo que mejoraríamos notablemente nuestra calidad de impresión si cambiamos a plásticos PLA. Opino que el ABS está anticuado y además es muy difícil de imprimir.

Printer One lleva muchos meses trabajando en preparar las impresoras y ajustarlas para imprimir con calidad. Creo que hemos llegado al momento en el que el material nos limita para seguir mejorando.

Quizás @David-Estevez o @jgvictores nos pueden ayudar a conseguirlos a través del departamento.
¿Hacemos votación para decidir si continuamos con esta idea? ¿Votación del color?

David-Estevez, 2017-11-13:
Si consideráis que no se puede imprimir con suficiente calidad usando ABS, veo razonable pedir PLA. Se puede buscar incluso algún distribuidor de calidad y barato, como el plástico de BQ que vendían en PCComponentes, a ~15€/kg.

Links de ejemplo:
* https://www.pccomponentes.com/bq-bobina-de-filamento-pla-1-75-mm-azul-topacio
* https://www.pccomponentes.com/bq-bobina-de-filamento-pla-1-75-mm-violeta
* https://www.pccomponentes.com/bq-pla-premium-1-75-mm-amarillo-1kg-fabricado-en-espa-a

¿Qué opinas, @jgvictores ?

Siotma, 2017-11-16:
A mi me gusta mucho esta idea. Ayer @diegotrap nos ofreció traer recortes de PLA la próxima semana (Gracias Diego!). Podemos empezar por ahí, ya que no sabemos si los extrusores de las impresoras estarían preparados para PLA...

diegotrap, 2017-11-16:
Llevaré unos restos de bobinas que tengo por aquí, espero que no se me olvide.

Siotma, 2018-01-09:
Hoy, hemos colocado una bobina [Smartfil PLA Cobalt](http://smartmaterials3d.com/es/tienda-smart/51-smartfil-pla-cobalt.html#/tamano-m_750g/diametro-1_75_mm_0_03) adquirida en la última convocatoria de asociaciones en Blacky.

De momento hemos hecho una primera prueba con una pieza para la pantalla de Hija Resurrection, con muy buen resultado:

[image]

Como digo, se trata solo de una prueba:
- Hemos cambiado las temperaturas de extrusión a 210ºC extrusor y 50ºC cama caliente.
- Queda pendiente generar un perfil más ajustado, sobre todo para quitarle el Brim.

El resultado de hoy promete grandes impresiones con PLA en el futuro!

daavm, 2018-02-09:
La semana pasada probamos a usar el filamento flexible de 3mm que llevó @David-Estevez en Dédalo. No tuvimos que cambiar mucho, y las pruebas que hicimos (un cohete) salieron bastante bien. Subiré fotos cuando las tenga ^^

GioSunrider, 2018-09-13:
Esta issue me parece la mas indicada para plantear un nuevo metodo este año, idea de @Siotma. Se van a proponer retos semanales de impresion para asi mejorar la calidad de nuestras impresiones. De esta amnera en nuestra casa imprimiremos piezas necesarias para la implementacion de piezas en las impresoras. Todo se debe a que la calidad de las impresiones en nuestras casa es muy superior a la de la nave, y asi no estamos de brazos cruzados esperando a las impresiones terminesn, asi que doy por inaugurado el label de RETOS DE IMPRESION.

Siotma, 2018-09-14:
Acabo de crear un Project para gestionar los retos: https://github.com/asrob-uc3m/impresoras-asrob/projects/3

 También está disponible el primer reto en: https://github.com/asrob-uc3m/impresoras-asrob/issues/84

Siotma, 2018-11-09:
Hoy he subido perfiles para [Blacky](https://github.com/asrob-uc3m/impresoras-asrob/commit/5345085ce4c570454549ff38569d3621a9ac8971) e [Hija Resurrection](https://github.com/asrob-uc3m/impresoras-asrob/commit/b976956694d1994c347750beb9f09c914ec9e575) diseñados para Slic3r 1.3.0

Adjunto resultados de cada uno:

[image]

[image]