asrob-uc3m/impresoras-asrob #38: Problema de generación de nuevos perfiles de impresión

davidvelascogarcia, 2017-11-29:
Se ha detectado en los ordenadores de las impresoras Blacky e Hija Resurrection, que no permite la generación y almacenamiento de nuevos perfiles de impresión, tanto en los perfiles de invitado como en el perfil de administrador Asrob. Permite rellenar los campos y exportar el perfil, sin embargo no su almacenamiento.

davidvelascogarcia, 2017-11-29:
En la sesión de ayer @daavm , detecto que la ruta de almacenamiento, no era correcta y ni siquiera existía.

daavm, 2017-12-07:
En la última reunión este problema quedó resuelto. @Siotma y yo nos pusimos con ello y, aunque no descubrimos el fallo en concreto, al eliminar la carpeta .Slic3r de '/home/asrob' y forzar al propio Slic3r a crear esa carpeta de configuración de nuevo, se arregló el problema. En resumen, si vuelve a pasar, con reconfigurar Slic3r debería solucionarse.

closed 2017-12-07 by daavm