asrob-uc3m/impresoras-asrob #39: Incluir Piezas en los Archivos de Configuracion de las Impresoras

GioSunrider, 2017-11-30:
Como he propuesto en [#31](https://github.com/asrob-uc3m/impresoras-asrob/issues/31) creo que seria necesario incluir en el `master` de cada impreso las piezas de cada una. En Dedalo ya hay una carpeta `parts` dedicada a ello.
La finalidad de esto no es otra que en caso de rotura, mal-funcionamiento o aunque improbable perdida de alguna pieza, sepamos con exactitud la pieza que se encontraba. De esta manera, ademas de ser mas fácil encontrar las piezas, se evitan errores de cambio de piezas, facilitando así la reparación de las mismas impresoras.

Siotma, 2017-12-01:
A mi me parece muy buena idea. La mayor dificultad sería conocer qué piezas exactamente llevaban las impresoras antes de iniciarse Printer One. Sin embargo, todas las piezas que hemos instalado desde entonces sí que sería muy recomendable guardarlas 👍 

En caso de que sean piezas sacadas de Thingiverse, yo también dejaría un enlace de la fuente de cada una (por ejemplo, en el [Readme](https://github.com/asrob-uc3m/impresoras-asrob/blob/master/Dedalo/Parts/Readme.md)) por si en un futuro se quisieran comprobar mejoras o nuevas versiones.

GioSunrider, 2017-12-03:
Pues solo queda manos a la obra. Creo que lo más difícil es como dices, conocer las piezas que se encuentran instaladas, pero bueno, poco a poco, la única que podríamos "completar" sería Dédalo, el resto abra que apañarselas.

David-Estevez, 2017-12-04:
De Dédalo tengo yo por ahí guardados un par de STLs de las piezas que yo personalicé. Son sólo modificaciones estéticas, pero puedo subirlas también si queréis tener todas las piezas originales.

GioSunrider, 2017-12-04:
Estaría bien, sobre todo por ver cuáles hemos cambiado y cuáles no, y dejar las que ahora mismo están instaladas. Sobretodo por si por ejemplo se parte una rueda dentada o algo, aunque improbable puede pasar.

davidvelascogarcia, 2017-12-06:
@GioSunrider Subo al repo el STL de las bed leveling, adunto [link](https://github.com/asrob-uc3m/impresoras-asrob/blob/master/Dedalo/Parts/Bed%20leveling.stl).

GioSunrider, 2017-12-06:
Perfecto, @Siotma ¿puedes también subir la pieza de sujeción del extrusor de Dedalo? Esa que imprimiste en casa.

David-Estevez, 2017-12-06:
Subidas las partes de Dédalo modificadas por mi (https://github.com/asrob-uc3m/impresoras-asrob/commit/d79c3a6ed4321783ab0c38bb067298bb184c71cd).

También he encontrado este zip, que imagino que serán las piezas de la i2 tal cual las descargué para imprimirlas:
[prusajr-PrusaMendel-d2fde50.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/1536801/prusajr-PrusaMendel-d2fde50.zip)

Y un Configuration.h, que no puedo asegurar que fuese la última versión que usé:
[Configuration.h.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/1536811/Configuration.h.zip)

Y como bonus, el gcode de la marcha imperial! 
[imperial.gcode.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/1536812/imperial.gcode.zip)

GioSunrider, 2017-12-06:
El Configuración.h dejaría el actual, ya que seguro que le tenemos actualizado y no da problemas por el momento y el resto genial, sobre todo el bonus track ajajajaja

David-Estevez, 2017-12-06:
Totalmente de acuerdo, si la versión que hay ahora funciona es mejor que se quede esa.

Siotma, 2017-12-09:
Del hilo de #31 he sacado las siguientes piezas (igual me he dejado alguna más)

- Carro X: https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-334395480 . Tener en cuenta [esto](https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-336418170).

- Extrusor: https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-342093484 .

- Soporte de filamento: https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-345018044 .

- Ruedas de nivelación de la cama: https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-347659805 (no sé de dónde los sacó @davidvelascogarcia, le invoco para que la añada ).

- Ventilador de capa: https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-346937941 (no sé de dónde lo sacó @daavm, le invoco para que la añada ).

He subido las tres primeras via 318703f53975232296e441800cad2fe794a34a0a .

Readme actualizado via 7072c2cc6ab2d733ed316ab6299c90debbba9c4e

dmarcosm, 2018-09-19:
Todas las impresoras tienen ya su carpeta de piezas con las piezas que hemos sabido reconocer, por lo que procedo a cerrar esta issue.

closed 2018-09-19 by dmarcosm