asrob-uc3m/impresoras-asrob #40: Mejora de la impresora Blacky

daavm, 2017-12-07:
En la última reunión, decidimos ponerle una pantalla de BQ que teníamos en el armario. Se editó el Configuration.h (pendiente subir el actualizado) y se subió el nuevo firmware. 

Todo funcionó bien, pero a la hora de apagar el ordenador que controla a Blacky, la pantalla sigue encendida, ya que se alimenta a través del USB de la torre del ordenador. Trasteé un poco en la BIOS, pero no conseguí hacer que los puertos USB no reciban alimentación una vez el ordenador está apagado. Si alguien tiene idea de como evitar esto, que lo diga. La pantalla se queda encendida si no solucionamos eso, por lo que ahora mismo está desconectada y guardada en la caja blanca de Printer One.

GioSunrider, 2017-12-14:
Para la proxima sesion imprimir el recubrimiento del LCD que ira en Hija Resurrection, en [este enlace](https://www.thingiverse.com/thing:275352/#files) se pueden encontrar.

GioSunrider, 2017-12-15:
Se imprimió y se coloco el LCD de Blacky. Adjunto fotos del soporte, que a su vez queda añadido a la carpeta de `parts` de Blacky, y del LCD instalado.
Admas se probo su funcionamiento y no da ningun problema.

[image]

[image]

[image]

daavm, 2017-12-15:
A la hora de intentar solucionar el tema de la alimentación de la placa cuando el ordenador está apagado, decidimos optar por modificar los cables de las impresoras para que fuesen data-only.
Para hacer esto, no hay más que abrir el cable, y sabiendo que de los 4 cables que tiene dentro (blanco, negro, verde y rojo?), el rojo es VCC, procedimos a cortar el rojo, cubrir los 2 trozos que quedan tras cortar con termoretractil, y cerrar el cable con cinta. 
Tras hacer una prueba y ver que funcionaba, lo hicimos para las 3 impresoras. 
Sin embargo, en la impresora hija no funcionó, imaginamos que por la placa, así que la propuesta fue que si no conseguimos que funcione, cambiar la placa por la nueva de BQ que hay en el armario, ya que tenemos una pantalla que queríamos poner en hija. No sin antes probar que la pantalla efectivamente se queda encendida al apagar el ordenador.

Siotma, 2017-12-15:
Respecto a la colocación de la pantalla sobre el marco de la impresora, yo puedo aportar que en mi experiencia **es un mal sitio**: si se interactúa con la pantalla durante la impresión es muy fácil hacer oscilar el marco de metacrilato, y ello puede hacer que el extrusor se mueva (suba o baje), dejando imperfecciones en la pieza. Este tipo de montajes son más apropiados para estructuras más robustas como la de Hija Resurrection o Dédalo. El problema de oscilación del marco es inherente al modelo de impresora, yo hace poco puse [estas piezas](https://www.thingiverse.com/thing:603953) a la mía para afianzar el marco.

Respecto al problema de alimentación de la RAMPS de Hija Resurrection por USB, habría que hacer pruebas pero a mi me pasó algo muy parecido y el problema era de la MEGA. No funcionaba correctamente el pin de Vin y por tanto los 12V no la encendían. Creo que antes de cambiar la placa deberíamos revisar con el multímetro tensiones en algunos puntos: http://www.reprap.org/wiki/RAMPS_1.4/es#D1.2C_D2_-_Diodos

Siotma, 2017-12-24:
En la reunión del día 21, pusimos a Blacky a imprimir desde la SD, obteniendo muy buenos resultados:

[image]

Ahora mismo, Blacky es totalmente funcional mediante la LCD, por tanto, urge, actualizar el [manual de usuario](https://github.com/asrob-uc3m/impresoras-user-manual-i3), ver la issue correpondiente: https://github.com/asrob-uc3m/impresoras-user-manual-i3/issues/4

daavm, 2017-12-27:
En la sesión del pasado jueves instalamos también la pantalla de Hija, en este caso una full graphics. Actualizamos el firmware y los nuevos archivos de configuración quedaron subidos al repo. El próximo día añado fotos :D

daavm, 2017-12-28:
Añado foto de Hija con pantalla :) 

[image]

Siotma, 2018-01-04:
Alguna noticia del [LCD de Dédalo](https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-344561347) @David-Estevez ?

David-Estevez, 2018-01-04:
Pues algo me dice que la traerán los Reyes Magos....

David-Estevez, 2018-01-08:
Ya tenéis vuestro regalo de Reyes en el armario :wink: 

[image]

Siotma, 2018-01-09:
> Ya tenéis vuestro regalo de Reyes en el armario 😉

Muchas gracias!

Hemos venido entusiasmados a la reunión para poder probarla, pero nos hemos dado cuenta enseguida de que no vamos a poder usarla con la Sanguinololu. En realidad, no sabemos si la pantalla es compatible o no, pero el adaptador, es decir, la pieza pequeña blanca señalada en la siguiente imagen, es para RAMPS. Además, en la pegatina del paquete de la pantalla también se indica que es una kit para RAMPS.

[image]

La dejamos donde estaba, no la hemos llegado a abrir. Seguramente puedas poner una reclamación al vendedor para que te envíe el adaptador correcto. Algo como esto:

[image]

Con un adaptador adecuado, tal vez esta pantalla funcione. Nos vas contando @David-Estevez .

David-Estevez, 2018-01-10:
Disputa abierta con el vendedor, le he pedido que me envíe el adaptador correcto o que me devuelva 3€ para que me pueda comprar otro a otro vendedor. Os mantengo informados de lo que me vaya diciendo.

Siotma, 2018-01-11:
En la última reunión, se imprimieron [estas piezas](https://www.thingiverse.com/thing:638833/#files) para hacerle una carcasa a la pantalla de Hija Resurrection.

[image]

Además, se practicó un agujero en la parte inferior de la carcasa para dejar la pantalla atornillada sobre la impresora:

[image]

David-Estevez, 2018-01-11:
El vendedor aceptó el reembolsarme 3€, con los que he comprado otro adaptador distinto en Aliexpress. En lo que llega os dejo la placa otra vez en el armario, por si se la queréis poner temporalmente a otra impresora o algo.

GioSunrider, 2018-02-20:
Issue que queda en stand-by, debido a que la pobre placa de dedalo cuando se le conecto la pantalla, el adaptador tenia los pines desalineados e hizo una mala conexion y salio ardiendo como se refleja en [#46](https://github.com/asrob-uc3m/impresoras-asrob/issues/46). Cuando evaluemos el problema se vera que hacer en esta issue.

GioSunrider, 2018-04-26:
Hoy hemos montado la placa de Dédalo (#48) y hemos conectado la pantalla que llevaba 2 meses en el armario. ~10 asrobcoins para @dmarcosm

closed 2018-04-26 by GioSunrider

Siotma, 2018-10-14:
Piezas LCD Hija actualizadas via 0363e391158485261eb205cfc239e86b738a5702