asrob-uc3m/impresoras-asrob #41: Aprovechamiento de los nuevos puntos de luz y puertos Ethernet de la nave

Siotma, 2017-12-09:
Recientemente se ha añadido una instalación nuevecita en la nave, la cual incluye varias torres ancladas al suelo, cada una de ellas con:
- Enchufes de 220V.
- Puertos Ethernet.

Actualmente, las impresoras y los PCs que las controlan están enchufados mediante un árbol de regletas bastante poco robusto. Además, los PCs carecen de internet a no ser que se tire un cable de 20 metros. Se quiere mejorar la actual configuración de los PCs de las impresoras, aprovechando las nuevas posibilidades de la nave, para ello se debe:

- [ ] Enchufar cada impresora a un punto de luz independiente cercano.
- [ ] Conectar cada PC a internet a través de un puerto Ethernet cercano.

Siotma, 2017-12-09:
En la última reunión, @jgvictores y yo modificamos la infraestructura de enchufes dejando la siguiente configuración:
- Monitor de Blacky, PC de Blacky, y Blacky, enchufados a un punto de luz **único** mediante una regleta de 4 tomas sin interruptor.
- Monitor de Hija Resurrection, PC de Hija Resurrection, e Hija Resurrection, enchufados a un punto de luz **distinto** mediante una regleta de 3 tomas con interruptor.

El enchufe del que colgaban antes ambas impresoras está libre y disponible para ser utilizado con mucha mayor libertad durante las reuniones.

Siotma, 2017-12-09:
Faltaría solicitar IPs al servicio de informática para poder conectar los PCs. Lo puedes mover tú @jgvictores ?

jgvictores, 2017-12-10:
Sí, me encargaré!

jgvictores, 2017-12-14:
Petición lanzada.

Siotma, 2017-12-14:
Ya disponemos de las IPs. No las pongo aquí por seguridad ¿Verdad @jgvictores?

Hoy, hemos conectado un PC con uno de los cables ethernet del router y hemos configurado la primera IP. Es muy sencillo. Hay que ir a:

`Inicio -> preferences -> network connections -> add -> Ethernet -> create -> ipv4 settings -> manual -> add address.`

Allí escribir: `address = IP asignada`. `Netmask = 255.255.255.0`. `Gateway = 163.117.150.2`

Clic en save y funciona.

Siotma, 2017-12-14:
Faltaría pedir un par de cables Ethernet al departamento. Ya que los que tenemos actualmente se usan para el router, y además, tienen las pestañas rotas por lo que no son muy robustos.

¿ Hace falta poner una issue en actas @David-Estevez @jgvictores ? O se puede gestionar internamente con el departamento...

Siotma, 2017-12-14:
Acaba de llegar @jgvictores con dos cables Ethernet como si fuera Moisés con las tablas de la ley.

Ya se encuentran ambos PCs configurados mediante el perfil Ethernet Connection 2 y conectados a internet correctamente.

Issue cerrada.

closed 2017-12-14 by Siotma