asrob-uc3m/impresoras-asrob #42: Puesta a Punto de Hija

GioSunrider, 2018-01-10:
Visto el buen resultado que nos dio esta misma issue, abrimos una para Hija.
Antes de Navidad, se imprimieron dos árboles de navidad ([ver en issue #40](https://github.com/asrob-uc3m/impresoras-asrob/issues/40)) y comparando el impreso con Blacky y el de Hija se notó un gran cambio de calidad con el de Hija. Por ello @Siotma ha propuesto una solución, ya que nos comentó el cambio de pasos por milímetro que anteriormente se cambió en Hija.
Los cambios propuestos son los siguientes:
Cambiar la placa, con los mínimos cambios posibles en el Marlin e imprimir el mismo g-code exacto con ambas placas. 

Una vez realizado este cambio y comprobación de validará la puesta a punto de Hija.

Siotma, 2018-01-11:
En realidad se pueden plantear aún más cambios:
- Cambio de la polea de la correa del eje X para que la correa vaya siempre en horizontal.
- Añadir un tensor para el eje X [este](https://www.thingiverse.com/thing:247992) va muy bien.
- Añadir un tensor para el eje Y.
- Ajustar el extrusor, que ahora mismo tiene fugas de plástico.
- Reducir holguras, si las hay, en la sujección del extrusor al eje X, que pueden ser también la fuente de la baja calidad de las impresiones.
- Engrasar.
- En caso de no poder mejorar la calidad de las impresiones, cambiar el extrusor directo actual por uno tipo geared como [este](https://www.thingiverse.com/thing:512338).
- Modificar el cableado que llega al extrusor. Cuando el eje Z sube demasiado, los cables se tensan demasiado y llegan a impedir el movimiento del eje X.
- Yo aún diría más: replantear todo el cableado.

Por cierto, se puede leer más sobre el problema de pasos por mm que comentas en #4

GioSunrider, 2018-01-11:
Me parece más que perfecto. A añadir lista de tareas entonces y manos a la obra.

GioSunrider, 2018-01-11:
Pregunta: respecto al problema de los pasos por mm, si cambiamos la placa por la de BQ, ¿se solucionaria?
P.D. Cambios añadidos al comentario inicial.

Siotma, 2018-01-14:
Posiblemente los Drivers de la placa de BQ acepten un mayor microstepping (mayor resolución) y por tanto sí que se puedan utilizar los números habituales. Esto es una suposición pero lo digo teniendo en cuenta que Blacky utiliza la misma placa y los mismos drivers, por lo que parece factible.

GioSunrider, 2018-01-14:
Yo es que tengo la duda si el problema es de la placa o de los motores, al fin y al cabo, los motores son los uqe se mueven, la placa solo manda "pulsos" no?

Siotma, 2018-01-14:
Eso es totalmente cierto. Para discernir esto, deberíamos probar a mover los motores de Hija desde la placa de Blacky (por ejemplo, con conectar el motor de X o Y y comparar el comportamiento sería suficiente).

GioSunrider, 2018-01-14:
Pues el próximo día investigamos a cerca de ellos, porque es algo que me lleva dando que pensar.

GioSunrider, 2018-01-24:
He venido a imprimir con hija la pieza que partió ayer de dedalo y me la he encontrado con un atasco de PLA curioso, sigue saliendo PLA plateado y no he conseguido desatascarlo, por lo que esta ahora mismo fuera de servicio hasta que lo consigamos arreglar.

Siotma, 2018-01-27:
Me extraña muchísimo que esté atascada con PLA cuando el último día imprimió una pieza de ABS amarillo durante dos horas 😕 .

GioSunrider, 2018-01-27:
Cuando fui a imprimir con ella salió plateado, además de la fuga por la parte de arriba del extrusor.

Siotma, 2018-01-28:
En la reunión del martes 23 comenzamos a imprimir piezas para el tensor del eje X.

[image]
[image]

También se propuso la posibilidad de recablear la impresora utilizando cadenas portacables para darle más robustez al cableado:

[image]

Para las piezas de la cadeneta, me he basado en [estas piezas](https://www.thingiverse.com/thing:1874802/#files) y he creado una versión parametrizable [aquí](https://github.com/Siotma/developrints/tree/master/Cable_chain_parametric).

Por último, y vistos los problemas del extrusor reportados por @GioSunrider , me ofrecí a donar las piezas necesarias para montar un extrusor geared tipo Jonas. Las llevaré en la próxima reunión.

Siotma, 2018-02-02:
En la reunión de ayer se sustituyó el extrusor MK8 de Hija por este otro tipo Gregs Wade. Aún no funciona porque no dio mucho tiempo a calibrarlo y además se partió un cable del motor, que aún está pendiente de reparar:

[image]

Se actualizó el Marlin para soportar el nuevo extrusor. Los cambios debidos al extrusor se han actualizado mediante 16e581b25ed1901dbd130e5e137c9286cd2ff612

Además, fue necesario sustituir el carro del eje X para dar compatibilidad a este extrusor, ver c02ba1170be563f307f5c064870bd085740b524f

Las piezas del extrusor y del carro que se han quitado, están guardadas en una bolsa hermética en el armario, dentro de una de las cajas de Printer One.

Por último, he creado un Readme para las piezas de Hija vía 800594e925fdd73a3a910fa833f13106423d8b61

Siotma, 2018-02-12:
En la reunión del pasado jueves, se reparó el cable roto del motor del extrusor y se ajustaron los pasos por mm en fa5c5e1739b1cd279aa2b05e3c58197001cbdaf8

Con la configuración actual, Hija Resurrection es funcional para imprimir como demuestra la siguiente imagen.

[image]

Siotma, 2018-02-17:
En la última reunión, se sustituyó la pieza de la parte derecha del eje X por [esta otra](https://www.thingiverse.com/thing:247992), que incorpora un tensor. La siguiente imagen muestra el resultado

[image]

Tras la sustitución se comprobó el correcto funcionamiento del tensor y se procedió a nivelar el ejeX, con resultado correcto. Sin embargo, las últimas pruebas de funcionamiento mostraron un extraño sonido de traqueteo en el movimiento del eje. Analizamos su origen durante la última parte de la reunión sin éxito.

Dado que este sonido puede significar un rozamiento que comprometería la salud mecánica de la impresora, Hija Resurrection queda fuera de servicio hasta que se solucione.

GioSunrider, 2018-02-22:
Se ha instalado la nueva placa de Ramps de BQ en hija, sustituyendola por la anterior que ha sido almacenada. @Siotma finalmente ha actualizado la configuracion del Marlin, por lo que Hija queda de nuevo operativa.

Siotma, 2018-02-26:
En la última reunión, tras cambiar la placa, se actualizó el Marlin tal y como refleja c8df68e5eb48b71d093aa41e5ddc5d0902caf728

Con la nueva configuraciíon, se imprimió una pieza para el extrusor de Blacky satisfactoriamente.

Hija queda de nuevo en servicio.

GioSunrider, 2018-02-27:
Se ha impreso el mismo gcode en hija y en blacky como se describe en [#45](https://github.com/asrob-uc3m/impresoras-asrob/issues/45).

GioSunrider, 2018-07-02:
En la ultima reunion se sello con termofusibles el agujero que existia en la carcasa de Hija.

[image]

JorFru, 2018-07-02:
Que no cunda el pánico pero...
HIJA NO FUNCIONA.

Cuando llegué, hija tenía un atasco importante a la altura del barrel. Tirando por arriba imposible. Aguja por abajo, imposible. Probé todo tipo de temperaturas diferentes, hasta 240ºC, imposible. Desmonté el hot end. El nozzle bien. El barrel atascado. Lo desatasqué calentando esta pieza introduciéndola completamente en el heatblock.

Problemas observados durante la manipulación:

- El barrel tiene la rosca marcada y no rosca del todo bien. No sé si fuí yo al desmontarlo/montarlo,  usé un trapo y mucho cuidado para evitar marcarlo con los alicates (necesarios porque estaba muy apretado). He vuelto a montar el conjunto y ha quedado como se puede ver en la foto. En teoría está bien ensamblado, al tacto lo está, pero se ve demasiada rosca fuera y da la impresión a la vista de que no está del todo metido.
[image]

- El sensor de temperatura (termistor) del hot end está en sus últimas. Los cables que salen de él no tienen ningún tipo de aislamiento eléctrico y posiblemente hagan cortocircuito entre sí. El resultado es que Marlin registra lecturas de 700, 800ºC de vez en cuando y dispara las protecciones por máxima temperatura a veces, durante la impresión.

- Durante los calentamientos para la extracción del atasco observé que el hotend no subía de 170-180ºC, y comprobé que esto se debía a que el ventilador de capa enfría en exceso el conjunto caliente. Lo retiré y obtuve un calentamiento normal. Al volver a poner el ventilador, la temperatura volvió a bajar. Comprobado que este ventilador produce un funcionamiento anormal de la máquina en su posición actual, debe recolocarse y posiblemente necesite un soporte nuevo. También esto ha podido ser la causa del atasco, por enfriar la parte inferior del bloque mientras que permite que el calor fluya por la parte de arriba. No estoy seguro, sólo son especulaciones.

Actualmente el conjunto caliente está ensamblado pero los ventiladores y tal están desmontados, colgando de su cable. Eran ya las nueve y pico, tuve que irme sin terminar. Lo siento.

GioSunrider, 2018-09-13:
Segun la idea de @Siotma se va a proceder a un cambio completo del extrusor. Le dejo por aqui escrito para que nos diga el que usa.

Siotma, 2018-09-14:
La idea de este extrusor es que lleva el filamento a través de un mismo tubo de teflón desde la zona de empuje directamente hasta la zona caliente, por lo que no hay interfaces donde pueda quedar atascado. Desatascar este extrusor es tan fácil como cortar el trozo de tubo donde esté el atasco, en el peor de los casos.
En mi experiencia esto no es necesario ya que el plástico desliza por el interior del tubo sin problema aunque esté frío: he llegado a quitarle el filamento a mi impresora sin calentar y solamente hay que hacer una pequeña tracción para conseguirlo.

El concepto es el siguiente:

[image]

Siotma, 2018-09-14:
Para devolver a Hija a la vida, ya está disponible el primer reto en: https://github.com/asrob-uc3m/impresoras-asrob/issues/84

GioSunrider, 2018-10-13:
Cierro esta issue por la nueva politica de issues, actualmente hija vuelve a estar operativa, por lo que ademas se le quita el label no funciona.

closed 2018-10-13 by GioSunrider

Siotma, 2018-10-14:
Piezas actualizadas via 0363e391158485261eb205cfc239e86b738a5702