asrob-uc3m/impresoras-asrob #45: Cambio de Extrusor de Blacky

GioSunrider, 2018-02-06:
En la pasada reunion se decidio cambiar el extrusor de Blacky debido a que el que se encuentra ahora mismo esta un poco roto. Se decidio, ya que habia que cambiarlo, @Siotma propuso el extrusor [Greg's Wade's Geared Extruder](https://www.thingiverse.com/thing:65939) y decidimos cambiarlo. Se imprimio la pieza idle y faltaria el cuerpo del mismo.

GioSunrider, 2018-02-08:
Se ha impreso parte de las piezas del extrusor. Decir que la pieza de la foto es el resltado final, despues que por ansias y por bruto me cargara un pequeña parte de ella pero ya esta como nueva.

[image]

Siotma, 2018-02-11:
Para instalar este extrusor, hay que disponer de un carro del eje X que disponga de una configuración de agujeros compatible. 

En mi impresora, monté este modelo: https://www.thingiverse.com/thing:701087

Me gusta mucho por el sistema de fijación de la correa que luego facilita el tensado. Presenta la dificultad de pausar la impresión a la mitad e introducir la tuerca en el interior de la pieza, pero no es muy difícil si se siguen las instrucciones que vienen en la propia página de Thingiverse.

En caso de que se decida no imprimir esta pieza por cualquier motivo, cualquier otra con la misma disposición de 4 agujeros M4 sería válida.

Siotma, 2018-02-18:
Dado que Blacky se atascó el último día al imprimir con PLA, propongo que el próximo día se desmonte el extrusor y se compruebe el estado del tubo de teflón interno al barrel. En caso de no tenerlo, es muy importante ponérselo. Creo recordar que en el armario había un barrel nuevecito con tubo de teflón

[image]

GioSunrider, 2018-02-22:
Se ha instalado el carro del extrusor, así como el nuevo extrusor. Aunque por lo visto no ha dado tiempo de su finalización, se quedara pendiente para la próxima sesión. Se le coloca la etiqueta 'No Funciona' hasta que se termine dicho proceso.

Siotma, 2018-02-26:
Tal y como dice @GioSunrider , tan solo falta instalar el motor del extrusor. Debe sujetarse con 3 tornillos. Como apunte, es muy importante que ambos engranajes queden bien alineados. Para conseguirlo, propongo utilizar tornillos M3 de 16mm y el número de arandelas que sean necesarias.

Una vez sujeto el motor, habrá que introducir el filamento y colocar las varillas roscadas con sus palometas y muelles, para fijar el idler. Se pueden utilizar las mismas que tenía el antiguo extrusor. Dejamos todas las piezas necesarias en un Tupper transparente.

Por último, es muy importante realizar pruebas de impresión para verificar el resultado final alcanzado.

Me gustaría destacar el detalle de que en la última reunión, **se cambió el barrel, que no tenía teflón, por otro que había en el armario, que sí lo tiene.** Así pues, confío en que pronto Blacky pueda imprimir correctamente con PLA sin atascarse.

GioSunrider, 2018-02-27:
@Tamayoo3 , @AlejandroAlv96 , @luisen30 y yo, hemos terminado de colocar el extrusor de Blacky. Adjunto fotos del resultado final. 
[image]

[image]

[image]

GioSunrider, 2018-02-27:
Como se muestra en la foto el cambio de extrusor se ha completado. Por lo que Blacky vuelve a estar operativa. Ademas se han impreso el mismo gcode en blacky y en hija, con la conclusion de que necesitamos ajustar los perfiles de impresion. Adjunto foto de ambos calicats, el azul impreso por blacky y el gris por hija. 

[image]

closed 2018-03-08 by GioSunrider

Siotma, 2018-03-10:
Añadidos @Tamayoo3 , @AlejandroAlv96 y @luisen30 al Team Impresoras

Siotma, 2018-03-10:
Aunque hace bastante tiempo de esto, lo comento para dejarlo documentado.

El día 27 de febrero imprimí con Hija Resurrection el Idler_holder sacado de [esta página de Thingiverse](https://www.thingiverse.com/thing:1855673/#files).

Para colocarlo en Blacky, limé un poco el Idler para hacerle una ranura y que quede bien encajado con el saliente del holder.  Una vez colocado, reemplacé las tuercas M3 de los tornillos de sujección por tuercas M3 autoblocantes.

El objetivo de esta pieza es simplificar el proceso de fijación del Idler. Al tirar del holder hacia arriba, el idler se libera y se puede acceder al filamento. Cuando se ha finalizado, es muy sencillo volver a colocar la pieza en su sitio tirando de ella hacia atrás. La principal ventaja es la **repetitividad del proceso**: como la pieza tiene una ranura siempre queda encajada en su sitio, y como las tuercas son autoblocantes, el idler siempre ejerce la misma presión una vez está todo colocado.

Una foto para ilustrar el resultado:

[image]