asrob-uc3m/impresoras-asrob #46: Puesta a punto de Dédalo [RETURNS]

daavm, 2018-02-09:
Ayer @David-Estevez nos trajo el adaptador para colocarle la pantalla a Dédalo, y nosotros super contentos corrimos a ponérsela. La buena noticia es que la pantalla funciona!! La mala es que uno de los pololus ardió, así que dédalo está sin funcionar ahora mismo. 
Cuando se le cambie la placa a Hija, podremos usar los pololus de esa placa en la sanguino de Dédalo.
Por otra parte, intentamos actualizarle el software para que admitiese la pantalla, pero volvemos a tener el problema de la última vez, por lo que necesitaremos de nuevo un programador como el que nos prestó @David-Estevez la última vez (ISP).

- [ Cambiar pololu/s y probar si se vuelve a quemar algo]
- [ Actualizar firmware]

David-Estevez, 2018-02-09:
Os lo presto otra vez? 😂

daavm, 2018-02-09:
> Os lo presto otra vez? 😂

Por favor, jajajaja

GioSunrider, 2018-02-20:
@David-Estevez siento decir que es imposible cargar la placa después de que saliera ardiendo, así que creo que la conclusion fue dar la placa por muerta T.T

David-Estevez, 2018-02-20:
Si el microcontrolador no responde es probable que muriese al colocar el adaptador de forma incorrecta.

En ese caso yo propondría los siguientes pasos:
1. Asegurarse de que nada más se quemó (ver si hay algo negro en la placa además del microcontrolador).
2. (Antes o a la vez que el punto 1) Usar el microcontrolador antiguo, el que no era compatible con la LCD para descartar más fallos en la placa y volver a tener la impresora funcional lo antes posible.
3. El microcontrolador que usamos era una muestra gratuita de ATMEL. Si sólo se ha quemado el microcontrolador, propongo pedir más muestras o comprar uno de repuesto.

¿Qué os parece?

Siotma, 2018-02-21:
Me parece perfecto. Podrías facilitarnos otra vez el microcontrolador antiguo?

David-Estevez, 2018-02-21:
Os lo llevo mañana

GioSunrider, 2018-02-21:
A mi tambien me parece una muy buena solucion, se quemo tambien un pololu, pero creo que cuando cambiemos la ramps a hija quedara alguno libre. En la revision no parece nada mas quemado, pero sin micro poco hemos podido probar.

David-Estevez, 2018-02-22:
De momento seguimos con el plan porque aunque se cambie el pololu si no funciona el micro no hacemos nada. Si el micro va buscaremos repuesto.

David-Estevez, 2018-02-22:
Ya tenéis el micro antiguo en el armario.

GioSunrider, 2018-02-22:
Estoy con el, gracias @David-Estevez ! Pero sigo sin poder cargarle el Marlin, y no reconoce el programador, solo el puerto convencional y ese es imposible de cargarle.

GioSunrider, 2018-02-22:
Se ha determinado que la placa se encuentra en buen estado. Se ha dejado instalado el micro antiguo, con los parámetros de Dedalo de cuando se trajo, ya que es imposible actualizar el Marlin. Se va a sustituir uno de los pololus que se extralleron de la placa de ramps de Hija [ver issue #42](https://github.com/asrob-uc3m/impresoras-asrob/issues/42). Quedaria ver si con un nuevo micro volveria a quedar como nueva.

David-Estevez, 2018-02-22:
Respecto a esto... tengo malas noticias. Al parecer el modelo Atmega1280 se ha descontinuado, y ahora sólo se vende en sus versiones SMD. 

Habrá que buscar a ver si alguna tienda tiene stock antiguo, pero la cosa no pinta demasiado bien...

Siotma, 2018-02-26:
Qué faena!!! :cry:

David-Estevez, 2018-02-26:
He encontrado un par de micros de repuesto del 1280p... un par de reliquias que vais a tener que tratar con mucho cuidado, porque no hay más...

GioSunrider, 2018-02-26:
Seran como nuestros hijos! jajajajaaj pero prometemos que no saldran ardiendo.

GioSunrider, 2018-03-08:
Despues de fallidos intentos, ha sido imposible hacer revivir a Dedalo. Conseguimos manejar la temperatura tanto de extrusor, pero los motores no quieren funcionar. Se ha verificado las conexiones, el estado de la placa e incluso sustituido el micro. Probando motores nuevos sin usar, con resultado fallido. Abro issue nueva con la propuesta de cambio de placa.

David-Estevez, 2018-03-08:
Me tiene tremendamente intrigado, porque en la  placa no hay nada roto a simple vista. Tras inspeccionar un poco más detenidamente el [esquemático](http://reprap.org/mediawiki/images/f/f5/Sanguinololu_1.3a.png), lo único que se me ocurre que puede afectar a todos los motores a la vez es: 

* VDD (debería ir a 5V)
* VMOT( debería entregar 12V)
* X-STEP, Y-STEP, Z-STEP, E-STEP (conectados al micro)
* X-DIR, Y-DIR, Z-DIR, E-DIR (conectados al micro)
* STEP-ENABLE y STEP-ENABLE-Z (conectados al micro, lógica negada [1 siempre a través de pull-up, 0 forzado por el micro para desactivarlo])
* GNDVDD, GNDMOT

También habría que revisar cortos entre 12V y GND (condensadores estropeados) o VDD y GND (pistas cortocircuitadas).

Siotma, 2018-03-10:
Comento para añadir que en la última reunión se intentó quemar el bootloader en un micro Atmega1280 completamente nuevo utilizando el mismo programador USBtinyISP que utilizamos [la última vez](https://github.com/asrob-uc3m/impresoras-asrob/issues/31#issuecomment-344783262), así como el mismo procedimiento, sin éxito. 

Esto complica el uso de Dédalo con la Sanguinololu, y también puede ser un síntoma de algún timpo de mal funcionamiento de la placa...

David-Estevez, 2018-03-14:
Y a unas malas....

https://www.youtube.com/watch?v=tzG5zHn_ysM
[image]

Siotma, 2018-04-21:
Viendo que #48 ha prosperado, cierro esta issue como wontfix, ya que no se va a seguir por esta vía de momento.

closed 2018-04-21 by Siotma

David-Estevez, 2018-04-21:
Le he cambiado el nombre a la issue para que se pueda localizar más fácilmente en el futuro, espero que no os importe.