asrob-uc3m/impresoras-asrob #47: Extrusor de Blacky atascado

David-Estevez, 2018-02-28:
Esta tarde durante la reunión de RoboFactory hemos intentado imprimir unas piezas en ABS negro con Blacky.

La impresora no extruía nada, así que procedimos a intentar extruir a mano sin éxito. Al intentar extraer el filamento no salía, y lo hemos acabado partiendo.

GioSunrider, 2018-03-01:
Tiene, tanto extrusor como el barrel recién puestos, igual es el filamento ese negro que es muy mierda y le usamos como señuelo. Si vuelve a pasar, en el armario hay agujas para desatascar (son las de acupuntura) en el armario que nos trajo Diego. Aún así el próximo día le echaremos un vistazo a ver cuál es la causa.

David-Estevez, 2018-03-02:
Como se partió el filamento y había poco margen de maniobras preferimos no seguir tocando nada para no empeorar la situación. Pero sí que es posible que fuese por usar el plástico negro ese malucho.

GioSunrider, 2018-03-08:
Se ha limpiado el atasco. Este se debió a la mala, malisima calidad del teflón en el barrel. Solo quedan hacer las pruebas de impresión tras la limpieza.

daavm, 2018-03-08:
Añadir que se ha hecho una pequeña modificación en el barrel para poder usarlo en el extrusor que actualmente usa Blacky, la cual arregla el problema por el cual se producía el atasco.
El barrel que se utiliza en este tipo de extrusor es tal que así: 

[image]

Este barrel tiene en la parte de arriba un trozo liso, sin rosca. Al tener los barrels que habíamos probado rosca hasta el final, no llegaba a entrar del todo, y quedaba un hueco en el cual el plastico se expandía, y, por tanto, se atascaba. Para solucionarlo hemos lijado la rosca sobrante para que entrase del todo. De esta forma Blacky quedaría (a falta de pruebas) operativa con PLA. La otra solucion era dejarle el barrel sin teflon que ya tenia ese trozo liso, pero solo serviría para usarla cn ABS. 
No hay fotos, el próximo día :)

David-Estevez, 2018-03-08:
Ahhh, así que para eso era la Dremel... :joy: :joy:

daavm, 2018-03-16:
En la sesión de ayer pusimos a Blacky a imprimir una carcasa para la Raspberry Pi y lo hizo de maravilla, así que damos por solucionado este problema :+1: 

[image]

closed 2018-03-16 by daavm

GioSunrider, 2018-03-19:
Cabe destacar, que hoy me ha costado bastante colocar el filamento, esto se debe a la modificacion del barrel, ya que se nota donde choca por la cantidad de filamento introducida. Si el filamento no entra recto cuesta bastante de colocar, a pesar de todo, Blacky ha impreso correctamente como se puede demostrar en la siguiente foto.

[image]

Siotma, 2018-03-20:
Es importante que al introducir el filamento, la punta de este no sea demasiado ancha. Esto ocurre habitualmente cuando el filamento se saca del extrusor para guardar. La punta se ha ensanchado como producto del calor y al intentar volver a ponerlo le cuesta entrar. La solución de baja tecnología para esto es romper la punta.

Además, como tú dices, el filamento debe estar recto para entrar más fácilmente. Para ello se pueden enderezar fácilmente con los dedos.

GioSunrider, 2018-03-20:
Creeme que hice todo, pero bueno, quiero que quede reflejado que da problemas aun haciendo todo.