asrob-uc3m/impresoras-asrob #48: Remplazar placa a Dedalo

GioSunrider, 2018-03-08:
Aprovechando el dinero de la subvención, propongo, ya que como se especifica mas detalladamente en [#46](https://github.com/asrob-uc3m/impresoras-asrob/issues/46), se compre una placa Arduino Mega y un nuevo juego de Pololus, para que Delalo vuelva a la vida. La Ramps hay una en el armario que funciona a la perfección que esta disponible después del remplazo de la misma a Hija.
¿Cual seria el método que habría que seguir?

David-Estevez, 2018-03-08:
Como es dinero de subvención, es dinero que gestionáis vosotros. No harían falta issues a nivel asociación, simplemente que os pongáis de acuerdo entre vosotros.

GioSunrider, 2018-03-08:
Pues... COMPAÑEROS DE PRINTER ONE! En nombre de Azazel YO OS INVOCO! 
¿Os parece bien esta idea? y ¿alguno podría hacer cargo de los costes que supondría esta compra?

David-Estevez, 2018-03-08:
Respecto a los costes, yo puedo ofreceros un pacto... (:smiling_imp: )

Siotma, 2018-03-10:
A mi me parece perfecto. @GioSunrider , podrías hacer una pequeña búsqueda para comparar precios y hacer una propuesta? Así podemos saber cuánto nos costaría y poder opinar sobre ello. Si además dejas algunos links, sería fantástico.

GioSunrider, 2018-03-10:
Sin problema, me pongo a ello, preferimos que sea original o con una imitación nos vale?

Siotma, 2018-03-10:
El presupuesto es limitado, con una de imitación puede valer perfectamente. Si encuentras alguna original bien de precio, pásala también.

daavm, 2018-03-10:
Yo creo que mientras no arda nos vale. Si es imitación pero sabemos que no es mala, pues adelante

GioSunrider, 2018-03-12:
Voy con las placas que he mirado. 
[OPCION 1](https://www.amazon.es/Tarjeta-Microcontrolador-ATmega2560-ATmega16U2-Compatible/dp/B06Y3ZHPWC/ref=sr_1_3?ie=UTF8&qid=1520853461&sr=8-3&keywords=Arduino+mega&dpID=51w0PuG6wHL&preST=_SY300_QL70_&dpSrc=srch)
[OPCION 2](https://www.amazon.es/XCSOURCE®-Atmega2560-16AU-Tarjeta-Versión-TE629/dp/B01N51Y3EJ/ref=sr_1_11?ie=UTF8&qid=1520853461&sr=8-11&keywords=Arduino+mega)
[OPCION 3](https://www.amazon.es/SODIAL-desarrollo-ATMega16U2-Mega2560-Arduino/dp/B00PXOGG74/ref=sr_1_14?ie=UTF8&qid=1520853461&sr=8-14&keywords=Arduino+mega)
Esto para las placas Mega Imitación, deberíamos valorar tiempos, por eso lo pongo por Amazon.
Y estos los pololus.
[OPCIÓN 1](https://www.amazon.es/Popprint-DRV8825-Controladores-disipadores-impresora/dp/B06Y6536CB/ref=sr_1_1?s=electronics&ie=UTF8&qid=1520853767&sr=1-1&keywords=Pololus)
[OPCION 2](https://www.amazon.es/BIQU-StepStick-Controlador-disipador-elevadores/dp/B06WVBMTKF/ref=sr_1_7?s=electronics&ie=UTF8&qid=1520853767&sr=1-7&keywords=Pololus)
[OPCION 3](https://www.amazon.es/Stepper-Driver-Pololu-Impresora-stepstick/dp/B071DQ7Y45/ref=sr_1_2?s=electronics&ie=UTF8&qid=1520853767&sr=1-2&keywords=Pololus)
Las diferencias entre las opciones son básicamente precio y entrega.

daavm, 2018-03-12:
Las opcion 1 de cada cosa pintan bien, en mi opiinión

GioSunrider, 2018-03-12:
A mí también me parecen las mejores.

David-Estevez, 2018-03-12:
Os habéis planteado algo más pro? (No sé si entrará en presupuesto) 

Algo como esto: https://www.staticboards.com/products/ovm20-lite/

daavm, 2018-03-12:
Es una buena opción, y tampoco se va demasiado de precio.

GioSunrider, 2018-03-12:
Lo había planteado, y más que mirado. Me parece la mejor opción en cuanto a calidad, pero el tema del presupuesto no se si será viable, pero puede quedar como otra opción.

David-Estevez, 2018-03-14:
Y a unas malas...

[image]

Siotma, 2018-03-15:
Yo también veo bien la opción 1 de ambas cosas. El coste total son 24€ aprox :+1: 

En cuanto a la placa OVM20 lite, a parte de que es más cara, a mi no me convence la idea de que tenga los drivers soldados, en cuanto haya algún problema en alguno, adiós placa...

Cuáles serían las ventajas de una OVM20 exactamente?

David-Estevez, 2018-03-15:
¿Alguna vez has tenido algún problema con algún driver? Yo la única vez que los he tenido han sido por conectarlos mal en su socket... Aparte de eso, la principal ventaja que veo yo es que la placa y los componentes son de calidad.

JorFru, 2018-03-15:
Propongo usar una Ramps 1.4 SB.
SB viene de StaticBoards.
Y ponerle un Arduino chinaco.

Ahorraríamos pasta y mantendríamos una electrónica de calidad, que lo
importante es la parte de potencia. Traen un MOSFET de cama correctamente
dimensionado a si que se podría usar sin relé, aunque no es obligatorio
quitarlo.
Luego a parte habría que comprar los drivers (otra razón por la que la OVM
renta, no hay que rallarse por comprar los drivers) y estos pueden ser
muuuuuy chinos o ser un poco mejores y comprar los de StaticBoards. (En ese
caso a estaríamos en el mismo precio de la OVM en sí)

GioSunrider, 2018-03-15:
yo sinceramente no compraria la Ramps, precisamente es lo que tenemos. O la placa completa o elegir de las opciones.

Siotma, 2018-03-15:
Una RAMPS, ya tenemos, aunque pensándolo bien es muy china.

Teniendo en cuenta que Dédalo no tiene Relé, y que antes tenía una Sanguinololu, creo que una OVM20-Lite puede ser buena idea para mantener la filosofía de placa monolítica.

Yo tengo la experiencia de que un Driver se me estropeó de un momento a otro mientras imprimía en una Feria de Asociaciones de esto hace años. Otra experiencia más reciente es cuando ardió el Driver de Dédalo al conectar la pantalla.

Considero que si tenemos cuidado al instalar la placa, una vez instalada ya no tendremos que tocarla y probablemente nos dure mucho tiempo y nos de muchas alegrías. Teniendo en cuenta que las otras impresoras tienen placas de BQ, y que esta es la única placa que _presumiblemente_ vamos a comprar con la subvención, creo que puede merecer la pena tenerlas a todas con electrónica buena.

OVM20 :hand:

GioSunrider, 2018-03-15:
Pues ya esta, la OVM20 creo que practicamente por unanimidad. Solo queda pensar quien adelanta la pasta?

Siotma, 2018-03-18:
Según he visto, el precio de la placa OVM-20 son 42€ + 6 de envío = 48€ .

La puedo comprar yo esta noche. Hasta entonces, se admiten nuevas sugerencias :)

Siotma, 2018-03-18:
Parece ser que el precio de 34,71€ es sin IVA. En Orballo Printing la tienen al mismo precio, pero el envío es más barato: 42€ + 1,20€ = 43,20€.

Yo creo que esta opción es más económica y nos puede interesar.

GioSunrider, 2018-03-18:
voto por la opcion economica.

Siotma, 2018-03-19:
Ayer compré la placa por 43,20€. Hoy me han dicho que ya la han enviado, así que en la próxima reunión la tendremos :+1:

Siotma, 2018-04-07:
De momento la placa no llega y estoy empezando a preocuparme. Ya he escrito a la tienda para pedir explicaciones.

Siotma, 2018-04-21:
Os cuento: 

El viernes día 13 llamé a Orballo Printing, la tienda donde había comprado la placa. Les expliqué que no me había llegado aún y que había pasado a preguntar por mi oficina de correos sin que pudieran darme ninguna solución. Al haberlo enviado por carta ordinaria no tenía número de seguimiento.
La persona que me atendió fue muy amable en todo momento y consultó con otros compañeros para tratar de darme una solución.
Sin que yo se lo pidiera, me dijo que no había ningún problema y que me iban a mandar otra placa, esta vez por correo registrado.
La placa llegó ayer a mi casa :tada: :tada: 

[image]

En la próxima reunión tocará probarla. Mientras, aún espero la factura por lo que dejo la Issue abierta.

David-Estevez, 2018-04-27:
>  La conclusión que hemos sacado es que a los Drivers paso a paso no les gusta que se conecten o desconecten los cables de los motores "en caliente". Si detectan algún problema en un motor se desactivan.

Por si fuese útil:

## 8.3.7 Protection Circuits
The DRV8825 is fully protected against undervoltage, overcurrent, and overtemperature events.

### 8.3.7.1 Overcurrent Protection (OCP)
An analog current limit circuit on each FET limits the current through the FET by removing the gate drive. If this analog current limit persists for longer than the OCP time, all FETs in the H-bridge will be disabled and the nFAULT pin will be driven low. The device remains disabled until either nRESET pin     applied, or VM is removed and reapplied. overcurrent conditions on both high-side and low-side devices; that is, a short to ground, supply, or across the motor winding all result in an overcurrent shutdown. Note that overcurrent protection does not use the current sense circuitry used for PWM current control, and is independent of the ISENSE resistor value or xVREF voltage.

### 8.3.7.2 Thermal Shutdown (TSD)
If the die temperature exceeds safe limits, all FETs in the H-bridge will be disabled and the nFAULT pin will be driven low. After the die temperature has fallen to a safe level, operation automatically resumes.

### 8.3.7.3 Undervoltage Lockout (UVLO)
If at any time the voltage on the VM pins falls below the UVLO threshold voltage, all circuitry in the  device will be disabled and internal logic will be reset. Operation will resume when V(VMx) rises above the UVLO threshold.

Siotma, 2018-04-28:
Concuerda completamente con el comportamiento que observamos en la última reunión.

JorFru, 2018-09-18:
¿Esta placa OVM20 está colocada en Dédalo?
No se puede saber porque no hay foto!!!