asrob-uc3m/impresoras-asrob #49: OctoPrint para nuestras impresoras!

daavm, 2018-03-11:
Hace unas semanas @GioSunrider sugirió usar una sola raspberry para controlar todas las impresoras (actualmente, 2), usando lo que dicen en este link: http://thomas-messmer.com/index.php/14-free-knowledge/howtos/79-setting-up-octoprint-for-multiple-printers
Básicamente lo que propone este hombre es crear varias instancias de Octoprint, de forma que cada una de esas instancias controle cada una de tus impresoras. 
Para esto se podría usar una de las Raspberry que hay en el armario si es posible usarlas. De todas formas, y aprovechando la subvención, compraría al menos una cámara para una de las impresoras (no se recomienda poner 2 camaras usb a la vez porque tiran de mucha CPU según he leído), así como un cable ethernet decente para conectarla a internet. 
Comentad!

Siotma, 2018-03-11:
Disponiendo ahora mismo de placas RaspberryPi y tarjetas SD en la Asociación, yo propongo muy fuerte usar esas, al menos para poner a punto todo el sistema, y probar distintas alternativas.

Esto tiene la ventaja de que no tiene retrasos de compras o envíos, y además, el uso de las placas para este fin no va a poner en peligro su integridad. Propongo, de hecho, que esta vía se comience a investigar en la próxima reunión con los medios disponibles. En el momento en que se tenga clara la configuración definitiva y el número de placas necesarias, se puede revisar si estas eran necesarias para otros proyectos y hacer una compra para reponerlas.

Respecto a comprar una cámara web, te digo lo mismo que a @GioSunrider en #48, haz una búsqueda y propón opciones para valorar precios, y lo mismo te digo con los cables Ethernet :wink:

David-Estevez, 2018-03-11:
En RoboFactory disponemos de Raspberry Pis (y tarjetas) que no estamos usando actualmente y que os podemos ceder para que no tengáis que esperar a que os llegue el material para empezar a trabajar :smile:

GioSunrider, 2018-03-12:
El próximo dia, si tenemos MicroSD de sobra puedo ponerme a ello sin problema que tengo todo en el portátil de cuando lo hice en casa.

David-Estevez, 2018-03-12:
No tenéis más que recordarme qué día os reunís esta semana para que os deje preparado el material.

GioSunrider, 2018-03-12:
Esta por decidir @Siotma tiene hoy que cerrar el doodle jajajajaja cuando lo cierre que confirme por aquí.

daavm, 2018-03-12:
Jueves de 18.00 a 21.00 :)

GioSunrider, 2018-03-15:
Gracias a la RPi y a la microSD prestadas por @David-Estevez se ha instalado en ella Octoprint. Siguiendo la pagina oficial de [Octoprint](https://octoprint.org/) presenta un manual de instrucciones para su uso e instalacion en su [pagina de descagas](https://octoprint.org/download/).

Ademas se han impreso un protector para la RPi, como se ve en la foto.

[image]

[image]

Ademas se ha Configurado Octoprint para Hija. El proximo dia quedaria probar su funcionamiento, para posteriormente comenzar con la gestion de varias impresoras con la misma RPi.

[image]

GioSunrider, 2018-03-19:
Hoy se ha probado Octoprint con hija. Para ello se ha conectado el router al ordenador y a la RPi. La dirección IP de Octoprint conectada al router es: `222.1.1.54`
Ademas se ha instalado el Plugin de Telegram y @daavm a traves de `@BotFather`, ha creado un bot para gestionar Octoprint desde Telegram. Adjunto la imagen resultado del uso del comando `/status`.

[image]

Ademas se ha impreso, como se muestra en la foto, usando Ocoprint un cilindro para una de las bobinas ya que actualmente solo disponiamos de uno.

[image]

Queda pendiente para la próxima sesión ajustar la RPi para el uso de una segunda impresora (Blacky).

Y por ultimo abro una issue para la compra de dos cámaras de RPi (Una por balda de la granja), ya que para las pruebas realizadas hoy, se ha dispuesto de una que había en el armario perteneciente a Robofactory (creo).

Siotma, 2018-03-20:
Excelente trabajo! Esto marcha!

David-Estevez, 2018-03-21:
Añadiendo a lo que dice @jgvictores , no sé si este router lo soportará, pero hay algunos que tienen dns local y permiten definir dominios locales tipo "dedalo.local" o "blacky.local".

O si no gestionarlo a través de avahi o algo así para no tener que recordar las ips.

jgvictores, 2018-03-21:
En TEO simplemente recomendamos un cierto contenido en `/etc/hosts`, pero claro, eso lo tiene que configurar cada usuario.

David-Estevez, 2018-03-21:
Yo lo del dns lo tengo configurado en casa y va guay, y lo del avahi no recuerdo exactamente cómo se configuraba, pero también iba guay (aunque los dispositivos Android, por ejemplo, no lo soportaban).

jgvictores, 2018-03-21:
Hombre, lo del DNS mola muchísimo.

Siotma, 2018-04-28:
Recuperando esta issue, hago la misma pregunta que en #54 : deberíamos comprar nuestras propias RPis y tarjetas SD?

dmarcosm, 2018-09-28:
Bueno, como prometí estos días he estado trabajando en Octoprint para poder usarlo en varias impresoras con una sola raspberry. He conseguido que 2 instancias funcionen de forma simultánea, a falta de probar en la nave si es capaz de imprimir en las 2 a la vez sin generar muchísimo lag. Si funciona, y ya por curiosidad, intentaré ver si es capaz de tirar 3 a la vez. De esta forma, solo necesitaríamos 2 Pi's y 5 cámaras.
En caso contrario, necesitaremos 3 Pi's y 5 cámaras, ya que ahora tenemos 5 impresoras y estaría bien dejarlas ya todas controlables vía Octoprint. 
Tengo pendiente hacer un script para generar la nueva instancia en cualquier raspberry (así nos ahorramos trabajo en el futuro). Lo subiré a una carpeta llamada "Octoprint" en el repositorio de impresoras, si os parece correcto.

GioSunrider, 2018-09-29:
No te compliques, 3Pi's mejor, así tenemos capacidad de extensión a otra impresora en caso de donación! Pero gran trabajo!

Siotma, 2018-09-30:
> Lo subiré a una carpeta llamada "Octoprint" en el repositorio de impresoras, si os parece correcto.

Me parece bien!

GioSunrider, 2018-10-02:
Necesitamos la compra de 3 RPi 3 y tres tarjetas microSD, @dmarcosm si puedes decidir tu la capacidad seria lo mejor, ya que eres el que ha hecho las pruebas.

GioSunrider, 2018-10-02:
Link compra Raspberries: https://es.rs-online.com/web/p/products/1373331/

dmarcosm, 2018-10-02:
> Necesitamos la compra de 3 RPi 3 y tres tarjetas microSD, @dmarcosm si puedes decidir tu la capacidad seria lo mejor, ya que eres el que ha hecho las pruebas.

Mañana compruebo, pero 8-16Gb deberían ser más que suficientes.

dmarcosm, 2018-10-05:
Me comenta @GioSunrider  (no tiene portátil a mano para comentar en la issue) que con las raspberries de GAME nos da de sobra, con SDs incluidas, así que no hace falta que las compres @Siotma

David-Estevez, 2018-10-05:
Como antiguo responsable de GAME confirmo la "cesión" del material.

GioSunrider, 2018-10-08:
Cierro issue y reabro para la intalacion de octoprint, dado que no se necesita la compra de ninguna de los componentes.

closed 2018-10-08 by GioSunrider