asrob-uc3m/impresoras-asrob #50: Construir una estantería para colocar las impresoras 3D

Siotma, 2018-03-11:
Esta es una parte fundamental del proceso de creación de la Granja de impresoras.

En la propuesta del proyecto consideramos comprar perfiles y baldas metálicas. La idea es construir una estantería en la que quepan al menos 4 impresoras. Se realizaron medidas descubriendo que cada impresora mide unos 50x50x50cm. Tal vez esto habría que volver a medirlo.

Se encontraron perfiles de 2m y baldas de 100x60cm. La idea es construir una sola estantería utilizando 4 perfiles y tres o cuatro baldas. La idea es colocar una balda a ras de suelo y otra arriba del todo para que hagan de estructura, y luego dos intermedias para disponer de 3 huecos con una altura aproximada de 2m/3=66cm. Con esta configuración se pueden poner 4 impresoras en los 2 huecos superiores y dejar la parte inferior para almacenar materiales.

Enlace a baldas: http://www.leroymerlin.es/fp/14118923/balda-balda-gris-fondo-60-cm?pathFamilaFicha=220116

Enlace a perfiles: http://www.leroymerlin.es/fp/10913070/perfil-perfil-gris-40-x-40-mm

Además de esto, haría falta comprar tornillería para montar la estantería correctamente.

Probablemente se puedan valorar otras alternativas por lo que se abre el debate!

daavm, 2018-03-11:
Por sugerir, sugeriría buscar baldas algo más anchas para poder poner 2 impresoras por balda. No veo util poner una impresora en una supuesta 3ra o 4ta balda, sería muy incómodo usarla, no?

GioSunrider, 2018-03-11:
A mi modo de ver, lo primero a realizar seria medir el espacion en la nave donde vamos a colocar el parque, porque a raiz de ahi podremos decidir estanterias y demas. Dibujar un boceto y empezar a buscar todo lo que @Siotma menciona, tornilleria, baldas y perfiles.

daavm, 2018-03-12:
Anoche y hoy hemos estado hablando en el grupo sobre la posibilidad de usar mesas lack del IKEA apiladas. Puede sonar un poco a chapuza, pero dejo una foto por aquí de cómo quedaría: 

[image]

Las piezas para unirlas las podríamos imprimir nosotros, y cubrirlas por los lados con metacrilato y ponerles una puerta que se cierre con un imán. Con 4 mesas iríamos sobrados y cuestan 10€ cada una, ahí dejo la idea.

Siotma, 2018-03-15:
Las mesas lack están bien, pero son endebles, están hechas de cartón y yo creo que se deteriorarían con el tiempo, especialmente con la humedad de la nave.

Yo propongo la configuración mostrada en la siguiente imagen, las unidades están en cm:
[image]

Utilizando las piezas de leroy merlin que puse en el primer comentario ([baldas](http://www.leroymerlin.es/fp/14118923/balda-balda-gris-fondo-60-cm?pathFamilaFicha=220116), [perfiles](http://www.leroymerlin.es/fp/10913070/perfil-perfil-gris-40-x-40-mm)), la estantería tendría 60cm de fondo, dejando 50x60x66cm para cada impresora. La estantería tendría capacidad para hasta 6 impresoras y costaría 75€ más tornillería, conque calculo que unos 80€.

Respecto a la posición en la nave, se proponen tres ubicaciones posibles:

1: junto a la columna de la lona de Robot:
[image]

2: en la zona del segundo armario (reubicando el segundo armario en alguno de los otros dos lugares):
[image]

3: junto a la pared de la nave:
[image]

Siotma, 2018-03-18:
A mi, la zona que más me gusta es la número 3. Pegada a esa pared se puede poner la estantería  de las impresoras y también una mesa para el PC donde generar el g-code.

Desde mi punto de vista tiene la ventaja de ser una zona apartada del paso, pero fácilmente visible.
La principal desventaja es que habría que reubicar (y/o desmontar) el terreno de juego para hacer hueco.

Visto que esta issue tiene poco movimiento, no sé si deberíamos tomar la justicia por nuestra mano, o replicarla en actas. 

Opiniones?

JorFru, 2018-03-18:
¿La jurisdicción del Dohyo de peleas es completamente de ASROB?

Siotma, 2018-03-18:
A qué te refieres exactamente?

JorFru, 2018-03-18:
Me refería a que si es 100% nuestro y no compartido con ningún departamento ni cosas de esas. Pero creo que sí, comenta @David-Estevez 

Mi lugar favorito para la granja de impresión es la posición 3. (Que lamentablemente no queda a la vista desde la pasarela, hubiese sido completamente épico).

También hay que pensar, y creo que es muy importante, qué se va a hacer con las mesas que actualmente soportan las impresoras. Se van a quedar muy solitarias y si se quedan solitarias son razones que estamos regalando para que otros laboratorios se recalifiquen el espacio. En algunas ocasiones este espacio se puede usar para exhibir y hacer usables los inventos de GAME: recreativa y hóckey.

Actualmente hay una issue abierta respecto a "decoración de la nave" https://github.com/asrob-uc3m/actas/issues/37 **en la que se puede comprobar lo siguiente: una de las propuestas es tirar la cosa verde**. Además el rack de impresoras puede cumplir con la función de decorar la nave, por supuesto. Y si hay alguna balda en desuso se puede habilitar como "hall of fame de PrinterOne"

David-Estevez, 2018-03-18:
Yo voy a contribuir recordando que hay una serie de propuestas en actas sobre qué hacer con la nave, de las que salió la idea de la granja: https://github.com/asrob-uc3m/actas/issues/93#issuecomment-335903419

Siendo coherente con la issue, estoy a favor de tirar la cosa verde y poner en su lugar la granja.

GioSunrider, 2018-03-18:
yo tambien estoy a favor de desmontar todo eso. El otro dia comentamos que una de las mesas quede como lugar de soldadura, con la fuente y el osciloscopio. y creo que una vez desmontemmos todo eso, tambien se podria recolocar el armario 2, para asi "hacer bulto".

jgvictores, 2018-03-18:
Tema cosa verde, mínimo votación a nivel Asrob. Tengo la idea intuitiva de que conviene conservarlo, prefiero comentar por qué en persona. 

Tema granja en general, no quería intervenir porque no quería cortar el rollo... Pero si es algo muy grande es posible que fuese bueno consultarlo a nivel departamento.

GioSunrider, 2018-03-18:
Mañana tenemos reunion de 11 a 15. Igual puedes pasarte y nos cuentas. Tema de la granja, creo que en el diseño de @Siotma vienen medidas "minimas" que tendremos que respetar para que entre todo, no se si eso se considera grande o no.

JorFru, 2018-03-19:
Para tranquilidad de @jgvictores puedo justificar que la tendencia actual para competiciones de robótica destructiva (uno de los pocos nichos de la robótica que usarían el citado deshecho verde) consiste en miniaturizar los competidores y limitarlos a materiales impresos para hacerlos más baratos y menos peligrosos.
Adjunto imagen de un competidor de tendencia actual.
[image]
Fuente: @esdrufao en Telegram, grupo @bricogeek

jgvictores, 2018-03-19:
Hemos estado midiendo esta mañana, hay espacio en la esquina para meter el estante de planta 100 cm * 60 cm sin tener que desmontar ni tirar lo verde.

La propuesta está en que una Raspi controle las impresoras vía Octoprint. Puede conectarse a directamente por cable a Internet de la toma de al lado, o vía wifi a router donde PCs actuales.

Si alguien intuye que lo que trato es que se perciba el menor número de cambios para que no peligre nuestro actual espacio, está en lo cierto.

Siotma, 2018-03-19:
Teniendo en cuenta los casos de uso que va a tener la granja de impresoras (uso por terceros, mantenimiento, paso de gente a su alrededor, ...), creo que es importante que se ubique en una zona no solamente con espacio suficiente para poner la estantería, sino también con buena accesibilidad y margen de maniobra para realizar la instalación de forma cómoda y poder hacer un mantenimiento mínimo sin quitar las impresoras de la estantería. Además, si la ubicación está apartada del paso, mejor para que nadie pueda derribar ni desplazar la estantería a su paso. Este asunto mejoraría si la estantería se pudiera anclar a la pared.

Dicho esto, también estoy de acuerdo con reducir los cambios en nuestro espacio de trabajo, a no ser que se tenga un apoyo general de la Asociación, pero **no considero que el espacio actualmente ocupado por el escenario verde sea espacio útil para ASROB**, por lo que si eventualmente perdiéramos un espacio de tamaño equivalente, seguiríamos exactamente igual que ahora.

Creo que **es importante que los trabajos en la granja comiencen pronto** para poder tenerla lista antes de que lleguen los exámenes y el verano, ya que si se queda a medias sería un gran problema. Como solución transitoria propongo que la granja se coloque inicialmente en la ubicación número 1, retirando la mesa de la fuente y el osciloscopio. En esa zona se podrían realizar los trabajos de construcción de la estantería y también se podrían hacer las pruebas necesarias con las impresoras al estar físicamente cerca de los ordenadores. **Sin embargo, esta ubicación provisional no es la mejor, y por tanto, no me gustaría que fuera definitiva.**

> Tema granja en general, no quería intervenir porque no quería cortar el rollo... Pero si es algo muy grande es posible que fuese bueno consultarlo a nivel departamento.

Todo lo contrario Juan, no cortas ningún rollo. Desde Printer One estamos dispuestos a hacer este trabajo para mejorar las condiciones de espacio de la nave, y las condiciones de uso de las impresoras. Creo que el Departamento podría ver con buenos ojos esta iniciativa, y tal vez hasta pueda darnos facilidades para encontrar una ubicación entre todos, o medios para anclar la estantería y darle más estabilidad.

Propongo que todos aquellos trámites que se crean necesarios hacer previamente se discutan por aquí, y se pongan en marcha para tenerlo claro a la vuelta de Semana Santa:

- [ ] Consultar el proyecto a nivel Departamento para analizar posibilidades de ubicación y anclajes.
- [ ] Discutir los cambios a realizar en nuestro espacio para ubicar la granja y consultarlos a todos los miembros de ASROB mediante una issue en actas.
- [ ] Poner en práctica las iniciativas que se desprendan de los puntos anteriores y **montar la granja**.

GioSunrider, 2018-03-19:
Yo veo incluso que podemos sobredimensionar la estantería de la granja, de esta manera "ocuparíamos espacio" y ademas seria mas fácil su manejo.

Siotma, 2018-03-20:
No hay baldas mucho más grandes que las que hemos propuesto. Por otra parte, la idea de la granja es desde su inicio, "mejorar el aprovechamiento del espacio". Me parece contraproducente cualquier intento de agrandar innecesariamente su tamaño.

GioSunrider, 2018-03-20:
Entonces a favor del comentario previo, en semana santa puede ser una buena oportunidad para aclarar esto.

David-Estevez, 2018-03-21:
> Consultar el proyecto a nivel Departamento para analizar posibilidades de ubicación y anclajes.

Yo me ofrezco voluntario para hablar con la persona correspondiente del Departamento (imagino que Ramón Barber),explicarles qué pretendemos hacer y consultar qué implementación sería la más apropiada.

David-Estevez, 2018-04-09:
Acabo de hablar hace un rato con Ramón Barber. Dice que le parece interesante la idea de la granja de impresoras y que, siempre que no nos salgamos del espacio que tenemos actualmente, no hay problema por parte del departamento para instalarla en nuestro espacio.

daavm, 2018-04-09:
Dado que la nueva impresora es tan grande, quizá deberíamos replantear el concepto de la estantería, ¿no?

Siotma, 2018-04-09:
Muchas gracias por gestionarlo @David-Estevez 

Respecto a lo que propone @daavm , habría que medir las dimensiones definitivas de la nueva impresora (#57) y en base a ello plantear una estantería distinta o plantearse colocar esta nueva impresora en otro sitio. Sin embargo, las otras tres impresoras van a seguir necesitando un sitio y la estantería puede seguir siendo una solución para ellas...

daavm, 2018-04-09:
64x64cm @Siotma

JorFru, 2018-04-09:
Las impresoras viejas almacenadas de cara a la pared. Y la TronXY en medio
de La Nave en un altillo.

Siotma, 2018-04-21:
Finalmente, esta issue se ha resuelto a nivel de asociación mediante https://github.com/asrob-uc3m/actas/issues/139.

La disposición de las impresoras será pegadas al muro de separación con la nave anexa, conforme a lo que indica el siguiente dibujo:

[image]

La estantería será de [este modelo](https://www.ractem.es/estanterias-media-carga/81-estanteria-c-max-200.html#/Color-Azul%20y%20gris/Niveles-4/Balda-Melamina/ancho-1000_mm/ancho-1000_mm/capacidad_de_carga-200_kg/fondo-700_mm/fondo-700_mm/alto-2000_mm/alto-2000_mm) y tendrá las siguientes medidas:
- 2000mm de alto.
- 1500mm de ancho.
- 700mm de fondo.
- 4 baldas.

Haré el pedido de forma que lo envíen directamente a la Universidad a nombre de @David-Estevez .

JorFru, 2018-04-27:
> El fabricante dice esto:
>
> Aguanta mejor la melamina, ya que la capa superior melamina esta preparada
> para resistir mejor el agua.
>
> Opiniones? Mañana hago el pedido .

Por mi dale!

GioSunrider, 2018-04-27:
PALANTE!

Siotma, 2018-04-27:
Comprada: 149,74€

Siotma, 2018-04-27:
[image]

David-Estevez, 2018-05-04:
El paquete ha sido recibido hoy con éxito, y se encuentra en nuestro espacio de trabajo habitual. 

* El transportista se puso en contacto hoy conmigo para gestionar la entrega, pues no tenía muy claro dónde había que dejar el paquete y era voluminoso (y pesado). 
* Me puse en contacto con el resto de asociados para tener ayuda de cara a llevarlo a la nave.
* Indiqué al transportista que descargase en conserjería, punto en el cual @davidvelascogarcia se unió para echarme una mano. 
* Usando el carrito que se encuentra en la nave, llevamos los paquetes hasta el espacio de trabajo de ASROB. (Ver foto)

[image]

Gracias a @davidvelascogarcia por echarme una mano, porque sin duda no hubiese podido llevar ese pedazo de paquete yo solo hasta allí.

Siotma, 2018-05-11:
@David-Estevez @GioSunrider y compañía, deberían contarnos qué tal se dio esto... :wink:

David-Estevez, 2018-05-11:
@GioSunrider llegó hoy un poco antes de la reunión de Printer One, cuando estábamos reunidos los de RoboFactory, y comenzó a montar la estantería. Para evitar accidentes, y ya que estábamos un poco ociosos, Cecilia y yo decidimos ayudarle a montar la estructura, que finalmente colocamos contra el muro, en el lado más cercano al armario de los autómatas.

En ese punto los de RoboFactory nos fuimos al bar, así que @GioSunrider puede contar el final de la historia.

GioSunrider, 2018-05-15:
Lo primero adjunto fotos del unboxing de la estanteria.

[image]

[image]

[image]

[image]

Como cuenta @David-Estevez , él y Cecilia me ayudaron a contruir ya que todo era muy endeble y podria causar accidentes. La estructura se decidio colocar en esas pestañas para que Geonosis pudiera entrar y ser manejada sin problemas. Seguimos el manuel de instrucciones para su montaje y lo colocamos a la derecha en el muro como se muestra en la foto.

[image]

Posteriormente se instalaron las maderas y se colocaron las impresoras al final de la reunion y encadenaron debidamente como resultado final que se puede ver en la foto.

[image]

[image]

Siotma, 2018-05-19:
@GioSunrider cierra esta Issue que te lo has ganado

closed 2018-05-19 by GioSunrider