asrob-uc3m/impresoras-asrob #51: Comprar un kit para construir una nueva impresora 3D

Siotma, 2018-03-11:
Esta es una parte muy atractiva de la subvención de Mejora y Modernización del parque.

Teniendo en cuenta que el importe total de la subvención han sido 455€, yo creo que deberíamos limitar el presupuesto de la impresora a 250€.

Además, como recomendación, creo que merece más la pena comprar una sola impresora de precio medio que intentar comprar dos de bajo presupuesto. Si compramos dos, tal vez no quepan todas al final en la Granja :rofl: 

Propongo seguir la siguiente estructura:
- [ ] Elegir modelo de impresora.
- [ ] Comprar el kit de la impresora.
- [ ] Recibir el kit de la impresora.
- [ ] Bautizar la nueva impresora.
- [ ] Abrir una nueva issue para gestionar su construcción.

Se abre el debate para elegir modelo, precio y proveedor!

daavm, 2018-03-11:
Creo que sabéis de sobra qué haría yo (2 Anets jej), pero creo que también sería buena idea comprar algo que no tengamos, algo de lo que podamos aprender. Quizá buscaría una tipo Delta, o una Core como decía @GioSunrider (aunque ésta se nos iría de precio probablemente).

JorFru, 2018-03-11:
Una CoreXY/HBot o bien una Delta.
Porfiiiii

GioSunrider, 2018-03-11:
Yo mi posicion tambien esta clara, creo que es mas atractiva una CoreXY, por tener una diferente, igual si valoramos precios podemos encontrar buenos precios, incluso en vez de kit, podriamos directamente comprar las piezas y montarla de 0.

JorFru, 2018-03-12:
La impresora se va a montar de cero y componente a componente fijíiiiisimo! Eso no lo dudes Giorgio.

Para convencer a @Siotma con a añadir un par de enlaces que se pueden leer en cosa de dos minutos. Y que presentan diseños cartesianos del tipo CoreXY. La Ultimaker y la Witbox, máquinas comerciales muy famosas de 1500€ usan este sistema. Por supuesto utilizando herramienta y mano de obra maker salen mucho más baratas, y estos diseños son de coste bajo, que ronda los 300€.

Esta primera usa varillas y rodamientos lineales para permitir el movimiento.
https://www.thingiverse.com/thing:2422839
Esta otra también usa varillas. Es mucho más grande y muchísimo más fea, pero está muy bien documentada y tiene un BOM muy claro con los precios puestos.
http://www.instructables.com/id/Vulcanus-V1-3D-Printer/

Esta otra difiere en que no usa varillas, si no que apoya ruedas directamente en el perfil de aluminio extruido.
https://openbuilds.com/builds/c-bot.1146/

Dando por hecho que se hará con perfiles de aluminio extruido, he estado mirando las opciones disponibles. Los famosos y disponibles V-Slot tienen la particularidad de que hay que comprar tuercas específicas o bien tunearlas con la Dremel
[image]
He visto los perfiles G-Slot, de MakerGal. Son de fabricación nacional y creo que en general molaría apostar por ellos. https://www.makergal.es/product-page/perfil-g-slot-2020-anodizado-negro
Más o menos se van entre 40 y 50€ en la perfilería comprándolo en MakerGal unos 8m. (he calculado que se usan 6m).

Termino aquí este comment, para reservar fuerzas si todavía no os he conseguido convencer.

GioSunrider, 2018-03-12:
No si eso ya sé que seguro @JorFru pero me refería a comprar rollo los perfiles, tornillería, etc en una ferretería, la placa elegir nosotros cual, la cama hacerla de la medida que queramos; vamos a montarnos un diseño y no un kit que ya viene todo preparado.

daavm, 2018-03-12:
A mi no me termina de convencer, pero si tuviésemos un precio detallado quizá sí. Veo que hay cosas que quizá ya tengamos en la nave que se puedan reutilizar. Alguien se anima a buscar el presupuesto? :) 
Lo de tunear tuercas con la Dremel suena bien jeje

Siotma, 2018-03-16:
A mi me gustan todas las ideas que se han propuesto, aunque no podemos dejar de pensar en el global de la subvención para tener cuidado con los costes. Si queréis, el próximo día hacemos un inventario de los materiales disponibles en la nave para construir una nueva impresora. Mientras tanto, estaría genial tener, como dice @daavm, presupuestos detallados de las diferentes opciones.

GioSunrider, 2018-03-21:
Hemos encontrado [está impresora](https://m.gearbest.com/3d-printers-3d-printer-kits/pp_701647.html?wid=89&lkid=13503442) modelo CoreXY por un precio menor de 300 euros. La impresora puede ser un buen bicho metiéndole Bowden y doble extrusor. No sé si os convence.

JorFru, 2018-03-22:
La máquina sugerida por Jorge viene desde el almacén europeo, ya sabéis a
qué me refiero. Significa que no habrá sobresaltos.

Tiene cama caliente, por lo que permitirá a Printer One hacer el bruto con
ABS.
Tiene un tamaño de impresión muy chulo, 330 330 400. Y la máquina tiene
forma de cubo y mide 65cm. Es más grande que las Prusas que tenemos, que
ocupaban 50cm. Habrá que tenerlo en cuenta para el mueble del rack. Aunque
esta máquina quizás se puede plantear que su estructura sirva para soportar
otra máquina encima. No tengo ni idea, lo dejo como idea loca.

Yo me animaría a comprar esta máquina. Pros que veo:

   - Viene desmontada.
   - Trae todas las piezas(o debería)
   - Parece muy modificable
   - Empezaremos (por fin) a hacer pruebas con tubo bowden
   - Es tribarata!
   - Le gusta a Jorge

Contras no le veo ninguna destacable.

Siotma, 2018-03-22:
A mi me encantaría poder comprar esta impresora, la verdad. Si fuera posible buscar por internet alguna referencia sobre cómo gestionan las facturas en Gearbest, tal vez en algún foro o algo. Es mucha pasta y hay que estar seguros.

Por otra parte, habría que adelantar la pasta...

Si resolvemos estas dos cuestiones antes de mañana, puede quedar pedida antes de irnos de vacaciones!

Siotma, 2018-03-23:
Acabo de hacer el pedido de la [impresora](https://www.gearbest.com/3d-printers-3d-printer-kits/pp_701647.html?wid=89&lkid=13503442).

Ha costado 232,51€. :tada:

JorFru, 2018-04-05:
[image]
Votación:
Cometer la obscenidad de colocar el embellecedor a la TronXY?

Siotma, 2018-04-07:
El martes día 3 recibí el aviso de llegada de la impresora en mi casa. Como no había nadie, la dejaron en un punto de recogida cercano.

El miércoles día 4 fui a por el paquete:

[image]

Y el jueves día 5 lo abrimos y empezamos el montaje!

[image]

La verdad es que queda de maravilla al lado de las demás :heart: 

[image]

De momento la llamamos TronXY, pero se admiten nuevas sugerencias de nombres.

Siotma, 2018-04-07:
Issue de construcción abierta en #57 

Cerrando.

closed 2018-04-07 by Siotma

Siotma, 2018-05-16:
Recibido reembolso en mi cuenta.