asrob-uc3m/impresoras-asrob #52: Sustitución de los rodamientos líneas.

GioSunrider, 2018-03-18:
Dado a que se han percibido "traqueteos" en Hija, propongo sustituir los rodamientos líneas por igus. 
Una de las ventajas que ofrecen es menor ruido, además de que no requieren ser engrasados.
Solo los he visto disponibles en Amazon. Adjunto [enlace](https://www.amazon.es/Igus®-Rodamiento-deslizamiento-drylin®-rj4jp/dp/B01M4JZ81S/ref=sr_1_1?s=tools&ie=UTF8&qid=1521332722&sr=1-1&keywords=Igus&dpID=41-MnCzDNXL&preST=_SX300_QL70_&dpSrc=srch).

Siotma, 2018-03-18:
Me parece muy buena idea comprar estos rodamientos para probarlos.

Yo he tenido buenas experiencias en [esta tienda](https://www.hta3d.com/es/igus-drylin-rj4jp-01-08-equivalente-a-lm8uu). Tienen envío 24h y si queremos comprar varias cosas igual nos interesa mirar aquí.

GioSunrider, 2018-03-18:
He estado echando un vistazo a la tienda, y tiene muy buenos precios, una vez hagamos lista de materiales podremos incorporar los rodamientos.

David-Estevez, 2018-03-18:
Por añadir experiencia de terceros, comentar que durante el tiempo que trabajé en BQ, mis compañeros probaron los rodamientos de IGUS y comentaron que no les terminaban de convencer porque se acababan desgastando muy rápidamente (no necesitan lubricación porque ellos mismos son el lubricante) y aumentando el juego de los mismos.

No obstante, y como esto son cosas que dijeron otros y no mi propia experiencia, os animo a que los probéis y documentéis si son adecuados para esta aplicación o no.

jgvictores, 2018-03-19:
En su día, lo que hablé con técnicos es que IGUS mola precisamente por la variedad de gamas que ofrece. Esto es, más que marca, hay que ver el producto en cuestión.

Tambien me comentaron que la tecnología esta donde el propio material es el lubricante sufre de lo que comenta @David-Estevez, con lo cual conviene pensar en la caña que se le va a dar. :-)

GioSunrider, 2018-03-19:
Muchas gracias por las aportaciones, lo tendremos mucho en cuenta, yo es un material que desconozco completamente, solo de lo poco que he visto por ahí. Valoraremos entonces el sustituirlos o no.

David-Estevez, 2018-03-19:
Si tenéis dudas puedo preguntar a ver si alguno se acuerda de lo que les pasaba en BQ a los rodamientos...

Siotma, 2018-03-19:
Sería muy útil conocer la experiencia de otros usuarios. De todas maneras, posiblemente la mejor opción sea realizar pruebas con nuestras impresoras ya que estos rodamientos no tienen un precio elevado.

Además, se al realizar el cambio, se podría aprovechar para identificar los rodamientos lineales LM8UU que se hayan estropeado; estoy casi seguro de que nos vamos a encontrar más de uno. Así, en caso de que los IGUS no nos terminen de convencer, podemos volver a colocar los LM8UU sustituyendo los estropeados por otros nuevos que compremos.

En ambas situaciones ganamos!

GioSunrider, 2018-03-19:
ME parece una gran opción, y respecto al la "autolubricación" del IGUS, seguro que echándole 3en1 se desgasta menos, aunque no lo necesite.

JorFru, 2018-03-20:
Echar cualquier otro tipo de lubricante es completamente contraindicado con IGUS.
Y no es que lo diga sólo el fabricante.

Lo siento pero no tengo tiempo para traerte las fuentes de mi argumento. Son casi las pi horas de la noche, quizás mañana...

daavm, 2018-03-20:
Yo uso 3en1 con los IGUS desde hará 5 meses y funciona de maravilla :)

GioSunrider, 2018-03-20:
Han tirado del lado consumista y de sacar pasta y a @JorFru han conseguido engañarle xD

David-Estevez, 2018-03-21:
Hombre, la gracia de los IGUS es precisamente que no hace falta lubricarlos... 

Preguntaré y os digo si averiguo algo

JorFru, 2018-03-22:
Lo que hay que escuchar xD
qué poco confiaís!

Investigaré yo tambien! Reportaré por aquí los resultados

2018-03-21 11:42 GMT+01:00 David Estévez Fernández <notifications@github.com

>:

> Hombre, la gracia de los IGUS es precisamente que no hace falta
> lubricarlos...
>
> Preguntaré y os digo si averiguo algo

David-Estevez, 2018-03-26:
Lo que he averiguado preguntando ha sido básicamente lo que os dije:

> Hay de todo, yo de siempre he pensado que eran una caca, las primeras witbox las llevaban e iban un poco mal
> En le grupo de la prusa original la gente los usa y hablan bien de ellos
> Creo que afecta mucho mucho las varillas
> No vale de nada comprar los igus y tener unas varillas del chino
> Tienen que estar perfectamente alineadas
> Creo que hay varios modelos y unos van bien y otros no

Y
> Los de IGUS a nosotros en las pruebas mecánicas que hicimos se atascaban con el polvo... se generaba una capa en la superficie que hacía que con el tiempo no deslizasen como el primer día

Más que poner unos IGUS  me recomiendan comprar unos rodamientos de bolas no chinos y tener las varillas bien limpitas y engrasadas.

David-Estevez, 2018-03-26:
Más feedback, esta vez positivo:

> Yo me pasé hace poco de los LM8UU a los igus... y estoy MUY contento. También hay que decir que los LM8UU eran chinos, pero las vibraciones que tenía se han ido, ya no hace apenas ruido y no tengo que engrasar las varillas para que vayan de lujo 👍

Siotma, 2018-03-31:
Vistas las opiniones creo que merece la pena probarlos en nuestras carnes. Sobre todo teniendo en cuenta que no son muy caros...

GioSunrider, 2018-04-01:
Me parece perfecto, el próximo día podríamos ver cuántos necesitamos para poder realizar el pedido.

diegotrap, 2018-04-19:
Acabo de llamar a IGUS, os lo recomiendo, atienden muy bien dudas técnicas :) Me han desmontado un mito que tenía, lo que he aprendido:

* No hace falta que los rodamientos vayan precargados sobre el eje. Yo pensaba que sí y que eso era lo que causaba el juego y atascos de los que habla mucha gente. Ahora pienso que debe ser bastante sensibles a la alineación y por eso pueden tener desgaste prematuro, atascos o juego.
* No hace falta lubricar pero tampoco pasa nada si se lubrican.

Siotma, 2018-10-11:
Esto sigue pendiente, se le podría dedicar algo de presupuesto...

dmarcosm, 2018-10-16:
IGUS: https://fervi3d.com/mecanica/380-igus-rj4jp-01-08-equivalente-a-lm8uu.html 

El precio unitario es de 1,30€, habría que ver cuanto queremos gastarnos y para cuántas de las impresoras daría :)

GioSunrider, 2018-10-16:
Yo de momento solo colocaria para hija, para probar. Si vemos que va bien podemos enfocarlo de cara al año que viene mas. con 8-10 igus tendriamos recambios.

Siotma, 2018-10-16:
El carro de X lleva 3 rodamientos. El carro de Y lleva 4. Si compramos alguno extra igual se pierde, a no ser que compremos suficientes como para ponerlos en otra...
Va a ser necesario tener recambios?

dmarcosm, 2018-10-16:
Yo llevo con los míos más o menos un año y no he necesitado recambios para nada. Yo compraría 14 y tendríamos para 2 impresoras :)

Siotma, 2018-10-16:
> 
> 
> Yo llevo con los míos más o menos un año y no he necesitado recambios para nada. Yo compraría 14 y tendríamos para 2 impresoras :)

14 son 18 euros. Igual es mucho. El objetivo de esta issue era probarlos para ver qué tal van.

A Blacky yo no le pondría porque no es nuestra. Geonosis no necesita porque solo tiene en Z. Spicy y Dédalo no recuerdo cuántos llevan...

De todas formas, yo soy partidario de comprar 7.

dmarcosm, 2018-10-16:
Spicy es Anet A8, lleva 7. Se los ponemos a Hija?

GioSunrider, 2018-10-16:
Yo soy partidario de solo 7 y para Hija.

GioSunrider, 2019-01-13:
Bloqueado por #83

GioSunrider, 2019-02-07:
Avances en #83

David-Estevez, 2019-02-07:
Si queréis comento cuando lleguen por aquí y así no tenemos que estar pendientes de #83 (y se puede cerrar)

GioSunrider, 2019-02-07:
Me parece perfecto, puedes cerrar la otra.

David-Estevez, 2019-02-12:
Habemus rodamientos:

GioSunrider, 2019-02-12:
Genial, asi el proximo dia podremos empezar a colocarlos y a probarlos.

David-Estevez, 2019-02-12:
Están ya en el armario, en la balda de Printer One

closed 2023-09-28 by CarlosME17