asrob-uc3m/impresoras-asrob #53: Mejoras y proposiciones varias para nuestras impresoras

daavm, 2018-03-18:
Dado que hemos empezado a usar PLA tanto en Blacky como en Hija, propongo, tirando de la subvención, comprar ventiladores de capa para ambas, ya que esto mejoraría las impresiones. 

Por otra parte, sugiero plantearnos la posibilidad de ponerle un segundo a extrusor a una de las impresoras (hija?). Lo dejo caer, para que lo discutamos.

JorFru, 2018-03-18:
El uno de los objetivos de ASROB y en particular de Printer One es llevar
la tecnología al máximo exponente.
Tenéis mi voto para el doble extrusor.

Siotma, 2018-03-18:
Los ventiladores son muy necesarios, y además son baratos. Podemos valorar los que hay [aquí](https://www.hta3d.com/es/componentes/electronica/ventiladores), y así agrupamos el máximo número de cosas en un mismo pedido.

Respecto a extrusores, también hay algunos [aquí](https://www.hta3d.com/es/componentes/extrusor/extrusores-completos).

GioSunrider, 2018-03-18:
Respecto a los ventiladores sin problemas, pero respecto a los extrusores, diría que mas que extrusores completos, tirar de motores, rodamiento,... Rollo a parte porque muchas de las piezas las podemos imprimir nosotros.
Propongo el siguiente [Hot End doble](https://www.hta3d.com/es/hotend-dual-v6-estilo-chimera-filamento-1.75-solo-piezas-metalicas), ya que también apoyo el doble extrusor.

Siotma, 2018-03-19:
Ese modelo que propones es fantástico!

Siotma, 2018-03-19:
Por cierto, yo recomiendo decantarnos por la opción con doble garganta de teflón. Este sistema es perfecto para imprimir PLA con Bowden, ya que el teflón llega desde el motor hasta el nozzle a través del mismo tubo, evitando cualquier atasco.

He leído que este sistema aguanta bien hasta los 245 grados, ya que por encima de esa temperatura el teflón se degrada; pero teniendo en cuenta el uso que se le va a dar en la nave, creo que es suficiente.

Si ocurriese la desgracia de que se estropeara el tubo de teflón, bastaría cortar la parte dañada, o si no, reemplazarlo. En [Aliexpress ](https://es.aliexpress.com/item/1M-PTFE-tube-Teflon-Pipe-for-1-75mm-or-3mm-Filament-of-3D-Printer-Parts/32818240272.html?spm=a2g0s.13010208.99999999.265.uCRLvS) es super barato.

Siotma, 2018-05-06:
He encontrado ventiladores a buen precio en eBay:

- [40x40x20](https://www.ebay.es/itm/201633351488)
- [50x50x15](https://www.ebay.es/itm/202065718094)

He pensado en pedir uno de cada para probarlos.

Siotma, 2018-05-11:
Realicé el pedido de los ventiladores y llegaron el miércoles. El jueves los llevé a la asociación. Tienen buena pinta, están disponibles en una caja en la balda de Printer One.

jorgercobo, 2018-06-11:
Posible ventilador de capa: https://www.thingiverse.com/thing:2120906