asrob-uc3m/impresoras-asrob #54: Compra de cámaras para RPi

GioSunrider, 2018-03-19:
Dado que Octoprint esta completamente operativo, mencionado en [#49](https://github.com/asrob-uc3m/impresoras-asrob/issues/49) y funcional, propongo la compra de dos cámaras para RPi que se colocaran una en cada balda para así controlar dos impresoras con la misma. 
Propongo [este modelo](https://www.amazon.es/MagiDeal-M%C3%B3dulo-Megap%C3%ADxeles-Seguridad-Raspberry/dp/B06XCSQN7Q/ref=sr_1_4?ie=UTF8&qid=1521500247&sr=8-4&keywords=camara+raspberry+pi) ya que lo considero que tiene buena calidad y no es caro.

daavm, 2018-03-20:
Dado que la Pi solo acepta una cámara "oficial", propongo comprar al menos una de las 2 USB. Así, si en algún momento decidimos usar una sola raspberry, no se nos queda una de las cámaras inutilizable.

Siotma, 2018-03-20:
Me preocupa la idea de poner una cámara "pelada".
Por una parte me gusta porque es barata y ocupa poco espacio, pero me disgusta que el cable sea corto y nos condicione para la ubicación de la Raspberry. Además, el sensor y la electrónica quedan muy expuestas y se pueden manchar o dañar.
Yo diría que fueran todas USB, para que la longitud del cable nos de más juego y esté más protegida con la carcasa. No hay ningún problema en utilizar un hub USB si se quieren conectar más de 4 dispositivos...

GioSunrider, 2018-03-20:
Era una de las propuestas pero si, me parece bien la compra de Camaras USB, lo unico que se nos sube de precio. El mas barato que he encontrado es [este modelo](https://www.amazon.es/Creative-73Vf077000001-Webcam-c%C3%A1mara-HD/dp/B00935Y7SE/ref=pd_lpo_vtph_147_bs_t_1?_encoding=UTF8&psc=1&refRID=WJ6J8PYZ99W17RAE4HT9).

David-Estevez, 2018-03-21:
Puede que también os pueda ayudar con esto, aunque sea para las pruebas. ¿Cuántas cámara necesitaríais?

GioSunrider, 2018-03-21:
Cualquiera USB o incluso si hay de Pi nos vale. no se necesita tampoco una calidad excesiva.

David-Estevez, 2018-03-21:
La pregunta es... ¿cuántas?

daavm, 2018-03-21:
En un mundo ideal, necesitamos 4. Pero con 2 de momento nos vale

David-Estevez, 2018-03-21:
Pues os podría proporcionar dos cámaras como la que habéis cogido prestada de RoboFactory.

GioSunrider, 2018-03-21:
Perfecto, muchas gracias. Ahora mismo nos valdría con solo una hasta que se haga el mueble de la granja 😊

David-Estevez, 2018-03-21:
Pues lo mismo de siempre, avisadme cuando sepáis la fecha de la próxima reunión y me paso a daros el material.

GioSunrider, 2018-03-21:
Perfecto jaajajajaj Muchas gracias 😄

Siotma, 2018-04-27:
Sé que ahora mismo nos las han prestado, pero creo que es algo temporal. Podríais referenciar por aquí una cámara válida para que las compremos con el dinero de la subvención? Si es en Amazon o similar, mejor que mejor.

David-Estevez, 2018-04-27:
La idea original de esas cámaras era formar parte de robots, pero de momento no hay ninguno planificado a corto plazo. De todas formas, no está de más conseguir alguna cámara más si tenéis presupuesto.

Por si sirviese de algo, aquí tenéis el link de las cámaras que os prestamos: https://www.pccomponentes.com/logitech-hd-webcam-c270

Siotma, 2018-04-27:
Lo suyo es que las de la granja, se compren con dinero de la granja que para eso se pidió. Esas que tenemos, pueden valer para cualquier otra cosa.

Alguien puede aportar algo sobre el modelo? Es adecuado? Compatible? Caro o barato?

Siotma, 2018-09-14:
En la reunión de ayer se planteó la idea de comprar estas cámaras en tiendas de segunda mano tipo CeX o CashConverters.

Recordad que se debe solicitar factura conforme con el artículo 12.B) de las condiciones de la convocatoria: https://www.uc3m.es/ss/Satellite/Cultura/es/TextoDosColumnas/1371240285442/

Creo que es una gran idea porque no se necesita de cámaras con grandes capacidades y tampoco tienen por qué ser todas iguales, así se dispondría de más dinero para otros gastos.

dmarcosm, 2018-10-02:
Link compra Raspberries: https://es.rs-online.com/web/p/products/1373331/

dmarcosm, 2018-10-02:
Link compra cámaras: https://www.amazon.es/Creative-73Vf077000001-Webcam-c%C3%A1mara-HD/dp/B00935Y7SE/ref=sr_1_2?ie=UTF8&qid=1538517394&sr=8-2&keywords=camara+usb+hd

GioSunrider, 2018-10-02:
Actualmente disponemos 5 impresoras, luego necesitamos 5 cámaras para poder gestionar todas por igual.

Siotma, 2018-10-12:
Acabo de realizar el pedido de 5 cámaras.

GioSunrider, 2018-10-12:
Hooray!!!

Siotma, 2018-10-16:
Hoy he llevado las cámaras a la nave:
[image]

closed 2018-10-16 by Siotma