asrob-uc3m/impresoras-asrob #55: Quedada de impresoras: trae la tuya!

Siotma, 2018-03-20:
Muchos de los miembros de ASROB tenemos impresoras en propiedad. Todos nosotros queremos nuestras impresoras como a la que más, y las tenemos bien cuidadas y puestas a punto.

Creo que sería muy interesante hacer una quedada en la Universidad en la que cada uno nos llevemos nuestra impresora y las pongamos todas a imprimir:
- Primero, porque **molaría muchísimo**. Calculo que podemos llegar a juntar más de 10 impresoras fácilmente, imagina todas imprimiendo a la vez.
- Segundo, porque es una oportunidad para **aprender unos de otros**. Podremos comprobar qué configuración electrónica, mecánica y de firmware utiliza cada uno, intercambiar impresiones sobre dificultades encontradas durante su uso o calibración y comprobar cómo lo ha resuelto cada uno.
- Tercero, porque con tantas impresoras juntas se pueden **hacer grandes cosas**. Podríamos plantear un reto de impresión, en el que entre todos imprimamos, por trozos, algo grandioso, como por ejemplo, [la mascota de ASROB](https://github.com/asrob-uc3m/recursos-digitales/tree/develop/logos/3d).

Para participar, solo haría falta que cada asistente aporte:
- Al menos, una impresora 3D, preferiblemente dotada de tarjeta SD para no tener que andar con PCs y aprovechar mejor el espacio.
- Filamento y soporte (se puede llegar a un acuerdo a priori sobre qué colores usar).
- Laca y herramientas de despegado de piezas.

Dadas las características de la actividad, creo que debería durar al menos "toda una mañana" o "toda una tarde". Molaría poder hacerlo un sábado pero yo no tengo potestad para abrir la universidad en sábado, así que no lo propongo.

Dejo aquí un doodle para que cada cual indique su preferencia de día y hora: https://doodle.com/poll/x4w4piwathbssqwd

GioSunrider, 2018-03-20:
I'M IN!

David-Estevez, 2018-03-21:
Igual es una idea de olla, pero si lo movemos con gente de fuera (Clone Wars, etc) igual podríamos montar un evento guapo, guapo. 

* Pros:
    * Conocer a un montón de gente guay
    * Igual hasta dan créditos por una Printer Party :joy: 

* Contras:
   * Limitación de espacio (aunque se podrían buscar alternativas)
   * Organización más costosa

Bueno, yo lo propongo como locura y a ver qué pasa...

daavm, 2018-03-21:
Me gusta la idea de @David-Estevez ,

Quizá lo más problemático como dices sería encontrar el espacio (sin salir de la uni), pero todo es ponerse

GioSunrider, 2018-03-21:
Me gusta también la idea, de hecho yo lo que propuse fue pedir permiso a la uni para organizar este evento en un aula grande y hacer eso, una Printer Party. Igual nos metemos en un jardín, pero molaria.

David-Estevez, 2018-03-21:
También se puede hablar con gente para intentar organizarlo en otro sitio con más espacio. Así sin pensarlo mucho se me ocurre el Medialab Prado, por ejemplo.

daavm, 2018-03-21:
Por listar otros sitios que estarían bien, se me ocurren el Open space de leganes (50 personas?), La Nave en villaverde (lanavemadrid.com)

GioSunrider, 2018-03-21:
Pero para eso habría ya que poner pasta verdad?

daavm, 2018-03-21:
Sería cuestión de mandar mails y ver qué necesitamos.

David-Estevez, 2018-03-21:
Si os mola la idea puedo hablar con la gente del MakeSpace Madrid a ver si se animan a ayudarnos a organizarlo.

daavm, 2018-03-22:
A mi me parece buena idea, así que diría que adelante con ello.

GioSunrider, 2018-03-22:
A mi también. Así que podemos mirarlo.

JorFru, 2018-03-22:
Si la organización de el resto de eventos que tengo me lo permiten, estoy
dentro chavales.
Me refiero al hackatón ASROB-UC3Music

Siotma, 2018-03-22:
Por una parte, me encantaría poder aspirar a un evento tan chulo como el que proponéis. El problema que le veo es que luego igual la asistencia es un fracaso... 

La finalidad de esta issue no es tan ambiciosa y creo que podría ser un buen punto de partida organizar algo pequeño a nivel de ASROB, buscando fomentar la participación interna, para en el futuro organizar algo más grande si nos vemos con ganas.

Dado que actualmente en el doodle solamente hay apuntadas 3 personas, no sé si es buena idea querer volar tan alto cuando aún no hemos empezado a caminar.

daavm, 2018-03-22:
Si organizaciones más grandes como MakeSpace Madrid nos ayudan yo creo que la asistencia no tendría por qué ser un fracaso, no?

David-Estevez, 2018-03-27:
Yo he mandado un mail por si acaso y os voy diciéndo qué les parece la idea.

GioSunrider, 2018-03-31:
Como va a quedar la cosa al final?

Siotma, 2018-03-31:
De momento somos 3 apuntados! :sweat_smile:

JorFru, 2018-03-31:
Sí, y uno es un tío random que no conocemos de nada.
https://thesongswelove.files.wordpress.com/2014/11/giorgio.jpg?w=625&h=390&crop=1

Siotma, 2018-04-07:
Me parece que de momento esta idea va a tener que replantearse, porque la participación no ha tenido mucho éxito.

jgvictores, 2018-04-08:
La idea mola... A ver si conseguimos a más gente para el día 12-13. :- )

Repito link: https://doodle.com/poll/x4w4piwathbssqwd

cc: @diegotrap ??
cc: ((@asrob-uc3m/impresoras inserten vuestros contactos aquí))

jgagom, 2018-04-08:
Me apunto a llevar la mía si es el 12! Aunque habrá 3h en las que tendré que estar en la nave con el autómata porque tengo clase. Tiene muy buena pinta el evento, a ver si se anima más gente!

Siotma, 2018-04-09:
Parece que podemos todos el jueves por la tarde, conque voy a establecer ese día y lo rulamos por los grupos para ver si más gente se anima! :wink:

Siotma, 2018-04-09:
Yo no podré llegar hasta las 18:00 o algo más tarde, pero en cuanto llegue me pongo al día!