asrob-uc3m/impresoras-asrob #56: Actualizar router ASROB

David-Estevez, 2018-03-21:
Ya que estáis en proceso de montar una granja de impresoras, y nuestro router está empezando a fallar (yo por ejemplo soy incapaz de conectarme a él desde el portátil), propongo la compra de uno en condiciones.

Si no teníais contemplado el gasto, puedo cederos un par de webcam de RoboFactory (https://github.com/asrob-uc3m/impresoras-asrob/issues/54#issuecomment-374954608) para que podáis comprar uno con el dinero que os ahorréis, y así ganamos todos.

GioSunrider, 2018-03-21:
Tengo que ver por casa, igual nos vale uno que tengo yo que está sin usar y que lo podría ceder sin problema. Pero igual nos merece la pena comprar uno en condiciones. Podemos debatirlo sin más. Y gracias @David-Estevez por el ofrecimiento de las cámaras.

JorFru, 2018-03-22:
Yo tengo en casa cinco o seis router de timofónica y otras empresas
similares. Sé que no valen para nada.

Si hubiese que comprar uno nuevo, hace no mucho discutí con Daeste cuál era
el adecuado para ASROB. Los miré en Amazon y salían a 40 paveles. Creo que
era un ASUS con Gigabit y con WiFi bibanda. Tendría que refrescarlo.
Si hay que decidir cuál, avisadme y lo busco.

Siotma, 2018-03-22:
Hay que tener en cuenta que el caso de uso del router de la granja es bastante básico:

- Un router con varios puertos capaz de establecer una LAN cableada con las raspis y con un PC de control.

No necesitamos ni Gigabit, ni doble banda ni nada :laughing: el único requisito es la fiabilidad para que el aparato funcione de forma estable durante un tiempo indefinido.

Por tener, yo también tengo un router en casa. Es un comtrend con 4 puertos y wifi g.

David-Estevez, 2018-03-22:
Todas esas features son pensando en que ese router pueda ser aprovechado por el resto de grupos activos de la asociación sin perjudicar el buen funcionamiento de la granja.

Pero oye, abiertos a donaciones estamos :blush:

Siotma, 2018-03-31:
El router que pongamos en la granja, se debería quedar en ella permanentemente. Es posible que a la hora de realizar ciertas acciones con él, el servicio pudiera quedar afectado o interrumpido. Pudiendo dedicar un router cutre a la granja, yo no me complicaría la vida compartiendo uno muy bueno.

Siotma, 2018-05-01:
Related https://github.com/asrob-uc3m/impresoras-asrob/issues/65#issuecomment-385659743

GioSunrider, 2018-11-27:
Cierro esta issue dado que como se muestra en #98 se ha instalado un router permanente para gestionar la granja de impresoras, ademas de que el router de la asociacion, actualmente en uso por Robodevastation, se le ha habilitado una conexion en la roseta como se muestra en #65 . Abrire una issue en actas cuando encuentre el router por casa para donarlo.

closed 2018-11-27 by GioSunrider