asrob-uc3m/impresoras-asrob #57: Montaje de la TronXY

Siotma, 2018-04-07:
Esta issue es para documentar y gestionar el proceso de montaje de la nueva impresora.

Para cerrar esta issue al menos:

- [ ] Se construirá la impresora por completo, documentando gráficamente el proceso y el resultado.
- [ ] Se reflejarán los problemas surgidos durante el montaje y se aportarán ideas para resolverlos.
- [ ] Se realizará la primera impresión.
- [ ] Se documentarán los perfiles de impresión y los archivos de configuración en este repositorio.

El último día estuvimos siguiendo [este manual](http://nathan.vertile.com/blog/2018/01/11/tronxy-x5s-build-guide/) de montaje y también fue útil [este otro](http://diy3dtech.com/gearbest-tronxy-x5s-3d-printer-manual/).

[image]

Manos a la obra!

daavm, 2018-04-09:
La impresora ha quedado montada entre el pasado jueves y la mañana de hoy. Hemos hecho la primera impresión (un calicat!) y no ha salido para nada mal :) Dejo algunas fotos del proceso y de dicha impresión:

[image]

[image]

[image]

Además, @JorFru ha hecho un cable management de los suyos :D 
## **Antes:**
[image]
## **Después:**
[image]

daavm, 2018-04-09:
**Problemas encontrados durante el montaje:**

- Uno de los perfiles no tenía la rosca bien hecha y los tornillos no apretaban. Tras intentar poner arandelas para que apretase sin éxito, optamos por darle la vuelta y usar el lado sin forma (?), ya que no era imprescindible en ese caso usarla con la orientación "correcta"
- Según comenta @JorFru, el cable de uno de los motores de Z no llega bien a la placa al hacer el buen cable management, le faltan unos 10cm, por lo que habría que hacer un empalme para que quede bien del todo. 
- La cadeneta horizontal que va por encima de la cama estaba algo defectuosa. Uno de los extremos no enganchaba bien con el anterior eslabón y por lo tanto quedaba un poco caída (con el tiempo podría llegar a caer más y chocar con la cama supongo). Hemos optado por ponerle una brida y ha quedado arreglado, al menos de forma temporal.
- La pieza donde se atornilla la cadeneta del eje Z tenía los agujeros muy pequeños y al principio no entraban los tornillos. Simplemente los hemos agrandado un poco con un destornillador y el problema ha quedado resuelto. 

**Otras observaciones:**
Falta por poner la tapa/ventilador de la placa. Apoyo una idea que ha dado @JorFru, y es que es una pena tapar por completo una placa que ha quedado tan bien. La sugerencia es imprimir un soporte para el ventilador que deje más a la vista la placa (como el que tiene Hija, por ejemplo). También cabe la posibilidad de cortar el acrílico que ya tenemos :) 
Por último, la cama nos ha parecido un poco cutre. De primeras, al calibrar la impresora por primera vez se ve que nos hemos acercado demasiado y ha quedado una pequeña marca, que no es muy notable pero ahí está. Hemos hablado de que quizá no sería mala idea cambiar la base de acrílico por un cristal de 330x330mm, que no sería muy caro.

Siotma, 2018-04-09:
El resultado es fantástico! 

Creo que puedo ayudar en el tema del espejo: [en IKEA los tienen de 30x30](https://www.ikea.com/es/es/catalog/products/39151700/) :+1:

GioSunrider, 2018-04-10:
Creo que si ponemos cristal la plancha magnetica va a servir de poco por el peso del cristal. Mejor imprimiria las piezas de las esquinas.

daavm, 2018-04-10:
> Creo que si ponemos cristal la plancha magnetica va a servir de poco por el peso del cristal. Mejor imprimiria las piezas de las esquinas.

No te sigo. Qué tiene que ver el peso del cristal?

GioSunrider, 2018-04-10:
que pude haber movimientos faciles de que se vaya mientras imprime, como en algun momento haya un minimo error y roce la punta puede desplazar con mayor facilidad el cristal, a eso me refiero.

daavm, 2018-04-10:
A mi me gusta también más la solución de las piezas impresas, que conste (por aquello de hacerlo nosotros), pero el cristal con el iman no se mueve con un toquecito.

GioSunrider, 2018-04-10:
Si has hecho pruebas con cristal votare a favor, pero creo que con la lamina de PEI hay muchas diferencias que no se pueden aplicar a un cristal.

David-Estevez, 2018-04-10:
Le pongas lo que le pongas no se va a mover porque la base sólo se desplaza en Z

GioSunrider, 2018-04-10:
Por moverse me refiero, que el extrusor choca con una pieza impresa, y esta al hacer la fuerza la cama se acabe desplazando, pero oye, si lo veis con fuerza adelante jajajajaja a mi me gustan mas las pinzas de extremos.

GioSunrider, 2018-04-12:
Hombre en cuanto a precio igual unas planchas de neodimio se nos va la olla ajajajajaja
Y en cuanto a eso como queráis, si queréis imanes adelante, no me opondré xF

David-Estevez, 2018-04-12:
Planchas no, con poner uno en cada esquina es suficiente. Un plancha yo creo que no la despegas ni de coña (igual tú sí, pero por tu fuerza sobrehumana :sweat_smile: ).

GioSunrider, 2018-04-12:
Jajajaajajaja es que creo que @daavm se refería por imanes a planchas que es lo que trajo un día, pero si, igual poniendo imanes en las direcciones de las camas sale un buen resultado.

daavm, 2018-04-12:
Yo hablaba de planchas de estas flexibles, pero lo de los imanes no es mala idea :+1:

Siotma, 2018-04-27:
@dmarcosm cierra esta issue como acordamos ayer.

closed 2018-04-27 by dmarcosm

JorFru, 2018-06-21:
https://youtu.be/4DUhI9M5A28